wireviz/WireViz #185: Create a "splice" type

MSBGit, 2020-10-22:
Sometimes I have one wire harness that is spliced to another wire harness (usually inside an enclosure for example). The splice is just wires soldered together and then heatshrinked, there is no physical connector at the splice. It would be useful to have a _splice_ type which functions like a connector but doesn't have pins or other connector attributes and would be rendered as one or more wires joining one or more other wires, but not in any type of "housing". Today, I create a _splice_ connector and use that to do what I want but it's confusing when someone looks at my wiring diagram and sees what looks like a connector but doesn't find a physical connector anywhere in the system (even though I comment the `yml` file when I define such a splice).
[image]

formatc1702, 2020-10-22:
Have you considered the `style: simple` and `autogenerate: true` attributes?

Would the following example work for you?

```yaml
connectors:
  S: &splice
    style: simple
    autogenerate: true
    type: splice
  S1:
    <<: *splice
    autogenerate: false
    show_name: false

cables:
  W1:
    wirecount: 3
    color_code: DIN
  W2:
    wirecount: 4
    color_code: IEC

connections:
  -
    - W1: [1-3]
    - [S, S, S1]  # wires 1+2 use autogenerated splice connectors
                  # wire 3 uses splice connector with user-specified designator
    - W2: [1-3]
  -
    - S1          # reuse S1 for the three-way splice
    - W2: 4
```

[image]

See #184 for ongoing work on improving auto-generated connectors.

formatc1702, 2020-10-22:
@MSBGit 
if my suggestion helps you, and you feel #184 might make this kind of work easier in the future, please close this issue.
Otherwise, let us know more details about your problem, or perhaps a sketch of how you would wish your kind of harness would be rendered.

closed 2021-03-20 by formatc1702