wireviz/WireViz #187: Issue building ex08.yml (image example)

MikeCrookes, 2020-10-23:
Having trouble building the ex08.yml which I'm sure is likely operator error but not sure what to try next.

Version info:
WireViz 0.2
graphviz version 2.38.0 (20140413.2041)
Python 3.8.0 (tags/v3.8.0:fa919fd, Oct 14 2019, 19:37:50) [MSC v.1916 64 bit (AMD64)]

So I have copied the yml file plus resources (image) directory into a local folder which I use to execute wireviz (another script runs wireviz on all yml files). I can attach if this is useful?

I get the following output;
```
Warning: No such file or directory while opening resources/stereo-phone-plug-TRS.png
Error: No or improper image file="resources/stereo-phone-plug-TRS.png"
in label of node Key
Warning: No such file or directory while opening resources/cable-WH+BN+GN+shield.png
Error: No or improper image file="resources/cable-WH+BN+GN+shield.png"
in label of node W1
aspect_ratio(): FileNotFoundError: [Errno 2] No such file or directory: 'C:\\bradwell\\_TEST\\WireViz\\ex08.yml.20201023-092259.out\\resources\\cable-WH+BN+GN+shield.png'
Traceback (most recent call last):

  File "<ipython-input-1-97b61b80a233>", line 1, in <module>
    runfile('C:/bradwell/_TEST/WireViz/wireviz_test01.py', wdir='C:/bradwell/_TEST/WireViz')

  File "C:\WPy64-3800\python-3.8.0.amd64\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 827, in runfile
    execfile(filename, namespace)

  File "C:\WPy64-3800\python-3.8.0.amd64\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 110, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "C:/bradwell/_TEST/WireViz/wireviz_test01.py", line 16, in <module>
    wireviz.parse_file(f,

  File "c:\bradwell\wireviz\src\wireviz\wireviz.py", line 212, in parse_file
    parse(yaml_input, file_out=file_out)

  File "c:\bradwell\wireviz\src\wireviz\wireviz.py", line 183, in parse
    harness.output(filename=file_out, fmt=('png', 'svg'), view=False)

  File "c:\bradwell\wireviz\src\wireviz\Harness.py", line 290, in output
    graph.render(filename=filename, view=view, cleanup=cleanup)

  File "C:\WPy64-3800\python-3.8.0.amd64\lib\site-packages\graphviz\files.py", line 207, in render
    rendered = backend.render(self._engine, format, filepath,

  File "C:\WPy64-3800\python-3.8.0.amd64\lib\site-packages\graphviz\backend.py", line 221, in render
    run(cmd, capture_output=True, cwd=cwd, check=True, quiet=quiet)

  File "C:\WPy64-3800\python-3.8.0.amd64\lib\site-packages\graphviz\backend.py", line 183, in run
    raise CalledProcessError(proc.returncode, cmd,

CalledProcessError: Command '['dot', '-Tpng', '-O', 'ex08.yml']' returned non-zero exit status 1. [stderr: b'Warning: No such file or directory while opening resources/stereo-phone-plug-TRS.png\r\nError: No or improper image file="resources/stereo-phone-plug-TRS.png"\r\nin label of node Key\r\nWarning: No such file or directory while opening resources/cable-WH+BN+GN+shield.png\r\nError: No or improper image file="resources/cable-WH+BN+GN+shield.png"\r\nin label of node W1\r\n']
```

The last line of output suggests to me that something is being appended to the image path?

I've tried various changes but no luck so far.

SnowMB, 2020-10-23:
What does your script do? It seems like it uses subdirectories (`ex08.yml.20201023-092259.out`) based on the yaml file(s)?

```
... FileNotFoundError: [Errno 2] No such file or directory: 'C:\\bradwell\\_TEST\\WireViz\\ex08.yml.20201023-092259.out\\resources\\cable-WH+BN+GN+shield.png'
```
Make sure to copy the `resources` into the same subdirectory.

MikeCrookes, 2020-10-23:
You are correct, I had missed this and removing this functionality in my script fixes the issue.

closed 2020-10-23 by MikeCrookes