wireviz/WireViz #191: cant generate BOM, unhashable type: 'list'`

Pherax, 2020-10-30:
images are created correctly but the BOM has some difficulties.

`
  File "C:\anaconda3\lib\site-packages\wireviz\Harness.py", line 339, in bom
    for group in Counter([connector_group(v) for v in self.connectors.values()]):

  File "C:\anaconda3\lib\collections\__init__.py", line 568, in __init__
    self.update(*args, **kwds)

  File "C:\anaconda3\lib\collections\__init__.py", line 655, in update
    _count_elements(self, iterable)

TypeError: unhashable type: 'list'`

I dont understand why I get this error. If I understand correct the keys (x1, x2,w1) are counted here. so maybe one of my keys is a list, but I checked them and they are all single strings.

formatc1702, 2020-10-30:
Are you using v0.2? Please share your input YAML, or at least a snippet that produces the same error.

kvid, 2020-10-30:
Thank you for reporting this possible bug. Can you please [describe a bit more information to enable another user to reproduce the error state described in your issue](https://github.com/formatc1702/WireViz/blob/master/docs/CONTRIBUTING.md#submitting-a-new-issue), typically the YAML input (and any image input files), and version number of WireViz, Python, Graphviz, and OS.

Pherax, 2020-10-30:
ha, while making the snippet I found i had one cable type wrong, it was a copy off the pinout list.

`  U16_X1:
    type: [WT,BN,GN]
    subtype: Cable
    pinlabels: [Supply +, Supply -, Shield]
    notes: pressure sensor U16_X1`

closed 2020-10-30 by Pherax