wireviz/WireViz #193: Allow referencing a cable's/bundle's wires by color or label, if unambiguous

formatc1702, 2020-11-11:
Similar to how a connector's pins can be referenced by its ID (as per its entry in `pins`), or by its label (as per its entry in `pinlabels`) in case the label is unambiguous, a cable's/bundle's wire needs the ability to be referenced by its color (if unambiguous) since that is often the most natural way for the user. Additionally, cables should have a `wirelabels` attribute analogous to a connector's `pinlabels`.

formatc1702, 2020-11-11:
I will attempt to implement this as part of #186 since the parsing of connection sets is being refactored there anyway.
The addition of the new `wirelabels` attribute should be trivial.

formatc1702, 2020-11-11:
This is an expansion of #169 and there was a suggestion of using `#`, `$` or similar to distinguish between sequential numbers, labels, etc.. but I can't seem to find it.

kvid, 2020-11-11:
> This is an expansion of #169 and there was a suggestion of using `#`, `$` or similar to distinguish between sequential numbers, labels, etc.. but I can't seem to find it.

I did suggest these prefixes in https://github.com/formatc1702/WireViz/pull/10#issuecomment-650581989 (almost 5 months ago).

formatc1702, 2020-11-11:
Thanks, searching for `#` and `$` within GitHub did not work, and it didn't help that the comment was in an already merged PR.

I'm happy to discuss this feature as a new issue/PR, feel free to open one; I already have some thoughts on the matter.

For the time being, #194 would close this issue ignoring the above mentioned feature.

PR #194 (2020-11-14): Reference wires by color or label
Closes #70.
Closes #169.
Closes #193.
closed 2020-11-14 by formatc1702