wireviz/WireViz #199: [feature]Add ability to embed the images into the HTML output

martonmiklos, 2020-11-19:
I think it would be useful to add an option to being able to embed images to HTML output in base64 encoded format. This would make the HTML output more "portable" or sharing friendly. i.e. one could share the plan as a single HTML file an call it a day. 

Syntax wise I would just add an 'embed' boolean under the image tags.

If no objections arose about this I could came up with a PR.

formatc1702, 2020-11-19:
Have you looked at #189?

The HTML file already has the SVG embedded, so it is "portable" right now. #189 embeds any referenced images in the SVG, and this carries over to the HTML output.

Or are you suggesting something else? If so, maybe you could elaborate, or present an example mockup of what you imagine.

martonmiklos, 2020-11-19:
> #189 will add the functionality of taking any images referenced in the YAML and embedding them into the SVG file. This, in turn, also carries over to the HTML output.

Ah I was not aware of this change or more exactly I thought that the images were part of the HTML output not the SVG. But I am glad to see that you found workarounds for the graphviz limitation.

closed 2020-11-19 by martonmiklos