wireviz/WireViz #200: [bug] Path to image error when running with -o pointing to a relative folder

ppskibinski, 2020-11-27:
## Observed behavior

When parsing the following `example.yml` file:
```
connectors:
    X1:
        type: my_connector
        pinlabels: [GND, VCC]
        show_pincount: false
        image: 
            src: my_connector.png
            width: 100
```

with `-o` option pointing to a relative path (containing a slash character), like here:
```
wireviz -o out/example example.yml
```
an error occurs with the following trace message:
```
aspect_ratio(): FileNotFoundError: [Errno 2] No such file or directory: '[my project folder here]/out/my_connector.png'
Warning: No such file or directory while opening my_connector.png
Error: No or improper image file="my_connector.png"
in label of node X1
Traceback (most recent call last):
  File "/usr/local/bin/wireviz", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.6/dist-packages/wireviz/wireviz.py", line 254, in main
    parse(yaml_input, file_out=file_out)
  File "/usr/local/lib/python3.6/dist-packages/wireviz/wireviz.py", line 183, in parse
    harness.output(filename=file_out, fmt=('png', 'svg'), view=False)
  File "/usr/local/lib/python3.6/dist-packages/wireviz/Harness.py", line 290, in output
    graph.render(filename=filename, view=view, cleanup=cleanup)
  File "/usr/local/lib/python3.6/dist-packages/graphviz/files.py", line 209, in render
    quiet=quiet)
  File "/usr/local/lib/python3.6/dist-packages/graphviz/backend.py", line 221, in render
    run(cmd, capture_output=True, cwd=cwd, check=True, quiet=quiet)
  File "/usr/local/lib/python3.6/dist-packages/graphviz/backend.py", line 184, in run
    output=out, stderr=err)
graphviz.backend.CalledProcessError: Command '['dot', '-Tpng', '-O', 'example']' returned non-zero exit status 1. [stderr: b'Warning: No such file or directory while opening my_connector.png\nError: No or improper image file="my_connector.png"\nin label of node X1\n']
```

## Platform 

- Ubuntu 18.04
- python 3.6.9
- dependances installed with pip3

## Affects
- stable 0.2 version
- master and dev branch as of 27 Nov 2020

## Workaround
It still works correctly when simply changing the target name (no folders) or not using the `-o` option at all.
```
wireviz -o my_output example.yml
```

This is particularly inconvenient when using WireViz in Makefile with separate source and build folders. The only solution I see now is to copy the source first to the build folder and then run wireviz or build in source folder and move the artefacts.

formatc1702, 2020-11-28:
Thanks for the detailed report!

Have a look at #189, which arose from exactly this behavior.
Feel free to try the `bugfix/image-src` branch and let us know if it fixes your problem :)

ppskibinski, 2020-12-05:
I just tested the code from this branch and it works great. What's more, I noticed that output svg-s have embedded images now which enables many further use cases for me, like including them in html documentation generated with Sphinx. Great job! I am looking forward to using future versions of this tool!

closed 2020-12-13 by formatc1702