wireviz/WireViz #201: Can't get Wireviz to generate anything

Rodrigoah, 2020-11-28:
Hi, i'm trying to get Wireviz to generate some wire schematics but I'm unable to do so. 

I'm running on python 3.9

I've installed Graph viz from the following link:

https://www2.graphviz.org/Packages/stable/windows/10/msbuild/Release/Win32/

And i've downloaded the latest package from PyPi:

When trying to run:

`wireviz C:\Users\Rodrigo\Desktop\Cableado\Centauro.yml`

I'm getting;

'Error: input file C:\Users\Rodrigo\Desktop\Cableado\Test.yml inaccessible or does not exist, check path
PS C:\Users\Rodrigo\Downloads\WireViz-master\src\wireviz> '

if I try to reach anocher location:

`wireviz C:\Users\Rodrigo\Downloads\WireViz-master\tutorial\tutorial01.yml`

I get:

`Format: "png" not recognized. Use one of:
Traceback (most recent call last):
  File "c:\python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\PYTHON39\Scripts\wireviz.exe\__main__.py", line 7, in <module>     
  File "c:\python39\lib\site-packages\wireviz\wireviz.py", line 254, in main  
    parse(yaml_input, file_out=file_out)
  File "c:\python39\lib\site-packages\wireviz\wireviz.py", line 183, in parse 
    harness.output(filename=file_out, fmt=('png', 'svg'), view=False)
  File "c:\python39\lib\site-packages\wireviz\Harness.py", line 290, in output
    graph.render(filename=filename, view=view, cleanup=cleanup)
  File "c:\python39\lib\site-packages\graphviz\files.py", line 207, in render 
    rendered = backend.render(self._engine, format, filepath,
  File "c:\python39\lib\site-packages\graphviz\backend.py", line 221, in render
    run(cmd, capture_output=True, cwd=cwd, check=True, quiet=quiet)
  File "c:\python39\lib\site-packages\graphviz\backend.py", line 183, in run
    raise CalledProcessError(proc.returncode, cmd,
graphviz.backend.CalledProcessError: Command '['dot', '-Tpng', '-O', 'tutorial01']' returned non-zero exit status 1. [stderr: b'Format: "png" not recognized. Use one of:\r\n']`

I've reached the limit of my knowledge

kvid, 2020-11-28:
- The first error is due to WireViz is unable to open the specified input file, but the input file in this error message is not identical as in your command line argument - try again and double check the spelling.
- The second error looks very similar to #170 - try the solution described there.

Rodrigoah, 2020-11-29:
Amazing! 

- First error turns out that it is not enough to rename a text file as .yml, had to use notepad++ to save as .yml
- Second error solved as #170 

However, I've had to avoid file paths containing spaces

Thanks a lot!

closed 2020-11-29 by Rodrigoah

kvid, 2020-11-29:
@Rodrigoah, I'm glad yor problems were solved.

> * First error turns out that it is not enough to rename a text file as .yml, had to use notepad++ to save as .yml

WireViz doesn't require `.yml` at the end of the input filename. You should be able to use any filename, but please make sure it's spelled correctly. I recommend using [TAB for filename completion](https://www.winhelponline.com/blog/how-to-enable-command-prompt-autocomplete-path-if-it-stops-working/) when typing in the command shell.

Be aware that renaming a file might fail if that file is kept open by a program (e.g. an editor) at the same time (it might depend on the program and the file system, I guess).

> * Second error solved as #170

Nice.

> 
> However, I've had to avoid file paths containing spaces

Please enclose the file path in double quotes when it might contain spaces, because all unquoted spaces are treated as separators between multiple command line arguments. This is common for most programs that takes command line arguments. Example usage:
  `wireviz "path with/spaces must/be quoted.yml"`