wireviz/WireViz #202: [feature] Add support for .webp image format

ppskibinski, 2020-12-05:
## Rationale

I am considering a use case where an internal (as in: not ever published on the internet or otherwise publicly accessible) documentation is created using small connector and wire images for clarity (`image:` directive in WireViz yaml files). These images may be readily found on electronics suppliers web sites next to specific connector components.(*) Most of the suppliers use modern `.webp` file format which is up to 3 time smaller than `.jpeg` and usually over 10 times smaller than `.png`. It would be beneficial if `.webp` image format could be recognized by WireViz. 

(*) I believe that such use is not violating any copyrights to these images. In fact it may be beneficial for the component suppliers as they are becoming more recognizable in a sense. Please correct me if I am wrong. It may also differ between countries.

## Feature proposition

1. Add `.webp` to the list of readable image formats for images defined by `image:` directive.
2. Add `.webp` file format to the list of targets generated by WireViz (replace `.png` ?).
3. Use images converted to `.webp` file format for embedding in `.svg` output files.

kvid, 2020-12-05:
- This [issue page](https://gitlab.com/graphviz/graphviz/-/issues/494) claims Graphviz does support `.webp`, but it doesn't say from which version. Personally, I'm running graphviz version 2.40.1 (20161225.0304) that doesn't support it.

- Maybe someone with a newer Graphviz version can try specifying a `.webp` image in a WireViz YAML input file, e.g. download from [Google](https://developers.google.com/speed/webp/gallery) and run it through WireViz to test the input capability. No changes to WireViz should be needed.

- To test if your Graphviz version can output `.webp`, please try executing e.g. "`dot -O -Twebp ex08.gv`". WireViz need some minor changes to support a new output format. A generic output format specifier that just forward it to Graphviz will hopefully be a future feature. WireViz should support a wide range of Graphviz versions, and let Graphviz raise an exception when e.g. an image format is unsupported.

- Execute "`dot -V`" to see the Graphviz version.

ppskibinski, 2020-12-06:
I confirm that 2.40.1 (20161225.0304) is exactly the version of Graphviz with which I failed to include .webp images in WireViz. This is the current version installed automatically as a dependency for WireViz when using pip3.

kvid, 2020-12-30:
> * Maybe someone with a newer Graphviz version can try specifying a `.webp` image in a WireViz YAML input file, e.g. download from [Google](https://developers.google.com/speed/webp/gallery) and run it through WireViz to test the input capability. No changes to WireViz should be needed.

I have now tried Graphviz version 2.44.1 (20200629.0846) for Windows with the two upper `.webp` images from the [Lossy Gallery](https://developers.google.com/speed/webp/gallery1) together with this YAML input:
```yaml
connectors:
  "1":
    style: simple
    image:
      src: resources/1.webp
  "2":
    style: simple
    image:
      src: resources/2.webp
```
Running it through WireViz from the current `dev` branch resulted in these warnings:
```
Warning: No loadimage plugin for "webp:cairo"
Warning: No loadimage plugin for "webp:cairo"
Warning: No loadimage plugin for "webp:svg"
Warning: No loadimage plugin for "webp:svg"
```
and both PNG and SVG output files included both nodes, each with an empty space sized to fit the images. It seems like Wireviz are able to read these image sizes, but not their contents, perhaps due to some missing expected plugin(s). Running the same input in Ubuntu with Graphviz version 2.40.1 (20161225.0304) gave the same warnings and similar results.

Modifying the YAML to use the two upper JPG files from the same gallery, resulted in these warnings:
```
Warning: No loadimage plugin for "jpeg:cairo"
Warning: No loadimage plugin for "jpeg:cairo"
```
and the SVG output file showed the images OK, but the PNG output file still had empty spaces as above. Running the same input in Ubuntu with Graphviz version 2.40.1 (20161225.0304) gave no warnings and both output files showing the images OK.

> * To test if your Graphviz version can output `.webp`, please try executing e.g. "`dot -O -Twebp ex08.gv`". WireViz need some minor changes to support a new output format. A generic output format [...]

This test resulted in this error message in Windows with Graphviz version 2.44.1 (20200629.0846):
`Format: "webp" not recognized. Use one of: bmp canon cmap cmapx cmapx_np dot dot_json emf emfplus eps fig gif gv imap imap_np ismap jpe jpeg jpg json json0 metafile mp pdf pic plain plain-ext png ps ps2 svg tif tiff tk vml xdot xdot1.2 xdot1.4 xdot_json`

And in Ubuntu with Graphviz version 2.40.1 (20161225.0304):
`Format: "webp" not recognized. Use one of: bmp canon cmap cmapx cmapx_np dot dot_json eps fig gd gd2 gif gtk gv ico imap imap_np ismap jpe jpeg jpg json json0 mp pdf pic plain plain-ext png pov ps ps2 svg svgz tif tiff tk vml vmlz vrml wbmp x11 xdot xdot1.2 xdot1.4 xdot_json xlib`

The question is: Is it possible to install some plugin(s) to support reading and/or writing the requested image format, and if yes, does it exist yet?
<details>
<summary>Click to expand my <code>"c:\Program Files\Graphviz 2.44.1\bin\config6"</code></summary>

```
# This file was generated by "dot -c" at time of install.

# You may temporarily disable a plugin by removing or commenting out
# a line in this file, or you can modify its "quality" value to affect
# default plugin selection.

# Manual edits to this file **will be lost** on upgrade.

gvplugin_core.dll core {
	device {
		dot:dot 1
		gv:dot 1
		canon:dot 1
		plain:dot 1
		plain-ext:dot 1
		xdot:xdot 1
		xdot1.2:xdot 1
		xdot1.4:xdot 1
	}
	device {
		fig:fig 1
	}
	device {
		ismap:map 1
		cmap:map 1
		imap:map 1
		cmapx:map 1
		imap_np:map 1
		cmapx_np:map 1
	}
	device {
		mp:mp -1
	}
	device {
		ps:ps 1
		ps2:ps 1
		eps:ps 1
	}
	device {
		svg:svg 1
	}
	device {
		json:json 1
		json0:json 1
		dot_json:json 1
		xdot_json:json 1
	}
	device {
		tk:tk 1
	}
	device {
		vml:vml 1
	}
	device {
		pic:pic -1
	}
	device {
	}
	render {
		dot 1
		xdot 1
	}
	render {
		fig 1
	}
	render {
		map 1
	}
	render {
		mp -1
	}
	render {
		ps 1
	}
	render {
		svg 1
	}
	render {
		json 1
		json0 1
		dot_json 1
		xdot_json 1
	}
	render {
		tk 1
	}
	render {
		vml 1
	}
	render {
		pic -1
	}
	render {
	}
	loadimage {
		png:svg 1
		gif:svg 1
		jpeg:svg 1
		jpe:svg 1
		jpg:svg 1
		png:fig 1
		gif:fig 1
		jpeg:fig 1
		jpe:fig 1
		jpg:fig 1
#FAILS		png:vrml 1
#FAILS		gif:vrml 1
#FAILS		jpeg:vrml 1
#FAILS		jpe:vrml 1
#FAILS		jpg:vrml 1
		eps:ps 1
		ps:ps 1
		(lib):ps 1
		png:map 1
		gif:map 1
		jpeg:map 1
		jpe:map 1
		jpg:map 1
		ps:map 1
		eps:map 1
		svg:map 1
		png:dot 1
		gif:dot 1
		jpeg:dot 1
		jpe:dot 1
		jpg:dot 1
		ps:dot 1
		eps:dot 1
		svg:dot 1
		png:xdot 1
		gif:xdot 1
		jpeg:xdot 1
		jpe:xdot 1
		jpg:xdot 1
		ps:xdot 1
		eps:xdot 1
		svg:xdot 1
		svg:svg 1
		png:vml 1
		gif:vml 1
		jpeg:vml 1
		jpe:vml 1
		jpg:vml 1
		gif:tk 1
	}
}
gvplugin_dot_layout.dll dot_layout {
	layout {
		dot 0
	}
}
gvplugin_gdiplus.dll gdiplus {
	render {
		gdiplus 1
	}
	textlayout {
		textlayout 8
	}
	loadimage {
		bmp:gdiplus 8
		gif:gdiplus 8
		jpe:gdiplus 8
		jpeg:gdiplus 8
		jpg:gdiplus 8
		png:gdiplus 8
	}
	device {
		metafile:gdiplus 8
		bmp:gdiplus 8
		emf:gdiplus 8
		emfplus:gdiplus 8
		gif:gdiplus 8
		jpe:gdiplus 8
		jpeg:gdiplus 8
		jpg:gdiplus 8
		png:gdiplus 8
		tif:gdiplus 8
		tiff:gdiplus 8
	}
	device {
		bmp:cairo 8
		gif:cairo 8
		jpe:cairo 8
		jpeg:cairo 8
		jpg:cairo 8
		png:cairo 8
		tif:cairo 8
		tiff:cairo 8
	}
}
gvplugin_neato_layout.dll neato_layout {
	layout {
		neato 0
		fdp 0
		sfdp 0
		twopi 0
		circo 0
		patchwork 0
		osage 0
		nop 0
		nop1 0
		nop2 0
	}
}
gvplugin_pango.dll cairo {
	render {
		cairo 10
	}
	textlayout {
		textlayout 10
	}
	loadimage {
		png:cairo 1
#FAILS		png:lasi 2
		png:ps 2
	}
	device {
		png:cairo 10
		ps:cairo -10
		eps:cairo -10
		pdf:cairo 10
		svg:cairo -10
	}
}
```
</details>
I have not yet found a similar configuration file in Ubuntu.