wireviz/WireViz #203: Custom Tables

ToniN9999, 2020-12-17:
I think it would be very beneficial if it would be possible to configure other custom tables similar to the BOM but to hold information such as drawing names, projects, authors, chnage information etc. Like below for example.

[image]

[image]

formatc1702, 2020-12-17:
See #32, this is a WIP that pretty much covers what you want :)

ToniN9999, 2020-12-17:
Thank you for that @formatc1702.
Apologies for not searching better i will close this request.

closed 2020-12-17 by ToniN9999