wireviz/WireViz #207: Error "first item not found anywhere" is unclear does not indicate the name of the item

hyphz, 2020-12-27:
It would be very helpful to list the name of the item that is not found, and the meaning of the error message is not clear at the moment (the first item of what?).

kvid, 2020-12-28:
Thank you for reporting this unclear error message. When you report something about what happens when running WireWiz, it will be much easier for others to reproduce the same state if you [include more information when submitting an issue](https://github.com/formatc1702/WireViz/blob/master/docs/CONTRIBUTING.md).

However, in your case I'm able to reproduce the same error message using e.g. this YAML input that contains a connection to non-existing connectors/cables:
```yaml
connections:
  - - X: 1
    - Y: 1
```
In a more realistic scenario, this error can easily happen when misspelling a connector/cable name. I agree that it will help the user to know the name of the item that is not found, and that this name is expected to be found in the set of connectors or cables. The exception is raised here: https://github.com/formatc1702/WireViz/blob/606ddbf9775da6300299640501083acfaef42077/src/wireviz/wireviz.py#L91

I suggest changing this to something like:
```py
            raise Exception(f'Connection item {first_item} not found in connectors or cables.')
```
There are also several other exceptions in WireViz that need improvements like this. Maybe we should raise a more general issue to identify more of them?

formatc1702, 2020-12-29:
> There are also several other exceptions in WireViz that need improvements like this. Maybe we should raise a more general issue to identify more of them?

**Yes.**

More user-friendly, helpful, and standardized (in terms of grammar) error messages have been on my mind for a while.

If @hyphz doesn't mind, I will hijack this issue and generalize as @kvid as suggested. Thanks!

martinrieder, 2024-05-16:
@nanangp
If there was a way to have a machine-readable link of the error output to the YAML input, this would highly improve the usability in nanangp/vscode-wireviz-preview

kvid, 2024-05-17:
In PR #251 I've replaced some print() warnings with warnings.warn(), and I think we could do that all over.

martinrieder, 2024-05-17:
Sure. That will definitely help, but still not refer to the actual line number from the YAML input. Therefore, I am linking this to the parser replacement issue: https://github.com/wireviz/WireViz/issues/223#issuecomment-2117019657

martinrieder, 2024-06-09:
@kvid wrote:
> In PR #251 I've replaced some `print()` warnings with `warnings.warn()`, and I suggest we could do that all over.

Please take a look at [Structlog](https://www.structlog.org/en/stable/). It can act as a drop-in replacement, while offering full configurability. See the demonstration of what an exception could look like: https://www.structlog.org/en/stable/_images/console_renderer.png

kvid, 2024-10-19:
As #426 shows, we should raise a `TypeError` if detecting any non-dict connector or cable entry, e.g. insert something like this in the existing [nested loop to parse YAML input file](https://github.com/wireviz/WireViz/blob/v0.4.1/src/wireviz/wireviz.py#L139):
```py
if not isinstance(attribs, dict):
    raise TypeError(f"Expecting a dict value for '{key}' in '{sec}', but got {type(attribs)}:\n{attribs}")
```