wireviz/WireViz #209: How to create an alternating sides daisy chain without templates?

hyphz, 2020-12-31:
The "alternating sides" daisy chain example also introduces a template syntax which is rather confusing as I'm not sure how to use one without the other. It seems to involve a "<<" member of the dict but it is not clear if this has to be on the connector, the wire, both, or what. Could the documentation on this be clarified?

kvid, 2020-12-31:
> The "alternating sides" daisy chain example also introduces a template syntax which is rather confusing as I'm not sure how to use one without the other. It seems to involve a "<<" member of the [...]

The [documentation](https://github.com/formatc1702/WireViz/blob/master/docs/syntax.md#inheritance) has a link to a page describing the anchor consept in general YAML files. I recommend reading that page. That consept is useful when several connectors or cables share a set of common attribute values.

hyphz, 2020-12-31:
That is very helpful for the template syntax, but it does not mention the details of the << signifier.

kvid, 2020-12-31:
> That is very helpful for the template syntax, but it does not mention the details of the << signifier.

Please search for "Extend" in that same page, and you'll find several examples using << to inherit from a template and then extend with other attributes.

hyphz, 2021-01-01:
Thanks. That is helpful, but in that case, what causes the zig-zagging in the zig zag example to happen? Is it triggered by the use of templates?

formatc1702, 2021-01-01:
Zig-zagging and templates are completely unrelated.

---

Regarding zig-zagging: Each individual connection set is interpreted as going from left to right.

Look at the [source code](https://github.com/formatc1702/WireViz/blob/master/examples/ex06.yml) for [Example 06](https://github.com/formatc1702/WireViz/blob/master/examples/readme.md#example-06):

Since `X1`, `X3` and `X5` are on the left, and `X2` and `X4` are on the right, the connection sets are defined as follows (ignoring the cables in between):

```
X1 -> X2
X3 -> X2  (going "backwards" in the daisy chain, but "forwards" as in left-to-right)
X3 -> X4
X5 -> X4  (same principle as above)
```

The actual arrangement of the nodes is handled by the GraphViz backend, so there might be cases where the visual output is not exactly what you expect, especially in more complex layouts. Unfortunately, currently there's not much WireViz can do about these cases.

---

Regarding templates, please note that these will not be required for most cases once #186 is finished and merged.

hyphz, 2021-01-01:
Thanks very much.

closed 2021-01-01 by hyphz