wireviz/WireViz #210: [feature] set Background color in Connector and Wire Box

designer2k2, 2021-01-09:
Hello,
let me first thank you for WireViz, it helps me keeping the wires in my car sorted and documented.

With larger projects the output gets a bit difficult to grasp, as everything is white.

How about an option to set the background-color from either the entire box, or the topmost box?

An example:
[image]

Manually edited in the svg:
[image]
If the entire box or the topmost im not shure, maybe the entire box can be a bit too much.

It would be nice if i can define this in the yml file, maybe as optional "bgcolor" that is passed to graphviz.

Anyway, thanks for WireViz!

kvid, 2021-02-20:
@designer2k2 and others that might need this feature are invited to clone my branch in PR #219, test this feature, and then make a review or some other comment about the suggested implementation.

designer2k2, 2021-03-09:
thanks kvid for #219, this solves my feature request 😀

I testes #219  and it works here for me as expected

closed 2021-09-28 by formatc1702