wireviz/WireViz #212: hide wiring information

wlgrd, 2021-01-11:
Is it possible to leave out the wire information, and just have the connections visible in the output?

kvid, 2021-01-11:
> Is it possible to leave out the wire information, and just have the connections visible in the output?

Currently, it's possible to make it a bundle and hide the cable name and wire count:
<details>
<summary>Click to expand YAML input</summary>

```yaml
connectors:
  X1:
    pincount: 4
  X2:
    pincount: 4

cables:
  W1:
    category: bundle
    wirecount: 4
    color_code: IEC
    show_name: False
    show_wirecount: False

connections:
  - - X1: [1-4]
    - W1: [1-4]
    - X2: [1-4]
```
</details>

[image]

1. Do you want to hide the whole Cable/bundle box including all contents, leaving just the wire splines between connectors?
2. Do you want to specify such a thing per cable, or for all cables in the harness?

If your answer to question 1 is yes, then I guess you might want something like this:
[image]
(I had to manually edit the `.gv` file and execute `dot -O -Tpng issue212.gv` to obtain the image above.)

wlgrd, 2021-01-11:
[image]
This is exactly what I'm thinking of. I haven't dived too deep into this module yet, could you elaborate on the .gv file?

formatc1702, 2021-01-14:
I never thought of this as a use-case, but I think it makes sense. An additional attribute for the cable would be necessary, and the GraphViz rendering would need to take this into account... I would welcome a PR for this matter!

kvid, 2021-01-16:
> This is exactly what I'm thinking of. I haven't dived too deep into this module yet, could you elaborate on the `.gv` file?

The `.gv` file is an intermediate file generated by WireWiz that is input to Graphviz for creating the diagram. Manually editing this file is time consuming for a large harness, and not recommended as a final solution.

When I get the time, I can try creating a PR supporting such a feature. It might fit well together with a couple of other changes I have in mind.

formatc1702, 2021-01-17:
Sounds good, I'm curious to hear about your other ideas.

kvid, 2022-02-04:
@wlgrd asked how to hide the wire information more than a year ago. If you (or other visitors) still need this feature, see my work-around suggestion below.

> When I get the time, I can try creating a PR supporting such a feature. It might fit well together with a couple of other changes I have in mind.

I'm sorry that I haven't had the time for this, but I still have my notes with ideas that could both resolve this and enhance the loops feature, but at the moment I should probably wait for the merging of PR #251 to avoid major conflicts in the code to generate the GV HTML.

> The `.gv` file is an intermediate file generated by WireWiz that is input to Graphviz for creating the diagram. Manually editing this file is time consuming for a large harness, and not recommended as a final solution.

The [tweak](https://github.com/formatc1702/WireViz/blob/master/docs/syntax.md#graphviz-tweaking-experimental) work-around suggested in https://github.com/formatc1702/WireViz/issues/174#issuecomment-707372543 was added in [v0.3](https://github.com/formatc1702/WireViz/blob/master/docs/CHANGELOG.md#03-2021-10-11) and that can help this old issue. Inspired by the https://github.com/formatc1702/WireViz/issues/270#issuecomment-1029535877, I found a work-around for this issue as well. Append the following to my one year old YAML input suggestion above:
```yaml
tweak:
  override:
    graph:
      ranksep: "0.5"  # Reduce the horizontal spacing (default 2)
    W1:
      shape: none  # Hide the surrounding box
      fontcolor: white  # Same as background to hide the text
```
This should then be the result, and is close to what you aimed for:
[image]

snelweg, 2025-03-11:
Using the workaround I notice that there is a white space going through the wires where normally the 'wire box' would be.
I hoped there would be a 'proper' way to not display the wire box, I noticed there were many people trying to get this working.

[image]

[powersuppliescabinet.yml.txt](https://github.com/user-attachments/files/19178089/powersuppliescabinet.yml.txt)

Crestless8, 2025-03-27:
I would also like to simply remove the box that contains the "1:RD" while leaving the wire visually intact. Thanks!

[image]