wireviz/WireViz #220: [feature] ability to import external yaml files for sharing connector and cable definitions

gmkado, 2021-03-04:
I would love to be able to have all my connector definitions in a single file and include it into my cable drawing yamls.  This would allow for things like a shared "library" of connectors that can be shared across a company.

From my searches online there are ways to do this with pyyaml:
- https://stackoverflow.com/questions/528281/how-can-i-include-a-yaml-file-inside-another
- https://gist.github.com/joshbode/569627ced3076931b02f
- https://pypi.org/project/pyyaml-include/

I tried adding [this code](https://gist.github.com/joshbode/569627ced3076931b02f) to the `wireviz.py` dev build but keep running into yaml syntax errors.  My knowledge of yaml parsing is pretty limited.

kvid, 2021-03-05:
Thank you for your effort, but please try including your library using the existing [command line option](https://github.com/formatc1702/WireViz/tree/master/docs#command-line-options) `--prepend-file`.

gmkado, 2021-03-09:
oops, should have read the docs better.  Thanks this does the trick!

gmkado, 2021-03-12:
Hmm I'm going to re-open this because the `--prepend-file` leaves something to be desired.  Ideally you would be able to keep common connector and cable definitions in a separate folder and point to it.  But doing this breaks paths to image files referenced by the definitions.

kvid, 2021-03-12:
> Hmm I'm going to re-open this because the `--prepend-file` leaves something to be desired. Ideally you would be able to keep common connector and cable definitions in a separate folder and point to it. But doing this breaks paths to image files referenced by the definitions.

The PR #189 already implements relative image paths differently, but your use case is still an issue and I describe the root cause in https://github.com/formatc1702/WireViz/pull/189#discussion_r593457127.

formatc1702, 2021-10-02:
There has been some progress in #189 regarding this issue, @gmkado please have a look and feel free to post feedback.

Liambeguin, 2021-11-17:
What about having something like this:
```yaml
templates:
  includes:
    - includes/DB9.yaml
    - includes/molex.yaml
    - include/inline_headers.yaml
```

This would make it explicit that this specific harness file depends on a list of "libraries"

Thanks for your work on this tool! It's really great at making harness drawings more efficient!

kvid, 2024-05-17:
The YAML parser library we currently use seems to need the full YAML contents as one long string before beginning the parsing. That's  why it's a bit hard to use YAML for specifying more YAML files to include. I don't know if the suggested #223 might help?

martinrieder, 2024-05-21:
@kvid take a look at https://github.com/adobe/himl and https://github.com/zerwes/hiyapyco

The hierarchical loading of YAML files could happen before the parsing in any other library, even the current one.

I still find the concept of using include statements quite promising. Since YAML uses # to start comments, you could theorerically employ any C code preprocessor with:

`#include "templates.yml"`

If it wasn't for the Windows compatibility, the beginning of a YAML file could even carry a shebang statement like:

`#!/usr/bin/wireviz --prepend-file xyz.yml`

This could turn your YAML file into an executable script!

tobiasfalk, 2024-05-21:
@martinrieder what is the Problem with windows, is this something purely becaus of the YAML library or what?

tobiasfalk, 2024-05-21:
One way could be to do it like/with a preprocessor, that goes through the file befor it is treated like a YAML file.

martinrieder, 2024-05-21:
@tobiasfalk see here: https://realpython.com/python-shebang/

The Windows CMD processor does not recognize this statement, but there are workarounds. This method would also rely on STDIN/STDOUT, and which is WIP.

Having the possibility to store CLI arguments in the YAML file would be a great way to generate reproducible output. We should place this header before an eventual `---` block start, so it could be processed before any YAML code.

tobiasfalk, 2024-05-21:
Then why not have the input file be a `.wz`(WireViz) file, this way one can add the association for Windows and the first line for Unix/Linux systems.
Then WiereViz takes this file let's a Preprocessor run over it, wich add all the includes and so, and exports a pure YAML file(`.wz.yalm`) wich then runs through the current code.

Similar to how a C compiler does it.

martinrieder, 2024-06-06:
@tobiasfalk I suggested Jinja in #377 for the HTML output. It could be applied to the input YAML as well!

tobiasfalk, 2024-06-06:
@martinrieder yes would also work, personally I would prefer C/C++ style, since I am familiar with it

martinrieder, 2024-06-09:
What do you think about the YAML addition that [LamaAni/bole](https://github.com/LamaAni/bole) suggests:
```yaml
settings:
    inherit: False # If true, allow inherit parent folders.
    inherit_siblings: True # If true allow inherit configuration files in the same source directory.
    allow_imports: True # If false dose not allow imports.
    use_deep_merge: True # Merge configurations via deep merge. If false, Only root keys are merged (and overwritten)
    concatenate_lists: True # When merging, append the merged list to the current one.
imports:
    - "**/*.config.yaml" # Recursively import all .config.yaml
    - path: my-config.yaml
      required: False # this item is not required.
      recursive: null # Applies if glob. Import recursively.
environments:
    [env name]:{ config overrides (any)}
```

It comes really close to what was suggested earlier.

liambeguin, 2024-06-09:
It's been a while since I posted that comment, and looking at it again, I think going with Jinja2 would be great! It provides a lot more than just includes and is also something pretty standard people are used to. 

I implemented a proof-of-concept here: #382

martinrieder, 2024-06-10:
Nice. Thanks for taking the initiative. I will give it a try ASAP...
I would strongly recommend using the `#` symbol instead of (or additionally) curly brackets. If Jinja statements look like a comment, you could still parse it as an ordinary YAML file. This would allow #348 to work without Jinja as well.

tobiasfalk, 2024-06-10:
Yes looks good.
As mentioned earlier, could we change the file ending in the same move?
Would not only the things mentioned above but programs like VScode could also detect dem as a wz file and have the right autocompletion and so on.

liambeguin, 2024-06-10:
> I would strongly recommend using the `#` symbol instead of (or additionally) curly brackets.

I'm not sure I understand what you mean by that. do you have an example? 

Jinja is a preprocessor, so if it sees it's keywords, it will interpret them otherwise the file is left untouched and can be parsed just like before.

> This would allow #348 to work without Jinja as well.

Schema validation is a great idea! have you considered using [pydantic](https://docs.pydantic.dev/latest/) for this?

martinrieder, 2024-06-10:
@tobiasfalk  please stick to `.yaml` or `.yml` as a file extenstion, so that text editors will properly recognize it. You can surely configure them to use any other extension, but it seems inconvenient. 
We might eventually upload a YAML schema to the [Schema Store](https://www.schemastore.org/json/), which could indicate a file name pattern. Common practice here is to use something like `.wireviz.yaml` as a file extension.

martinrieder, 2024-06-10:
> > I would strongly recommend using the `#` symbol instead of (or additionally) curly brackets.
> I'm not sure I understand what you mean by that. do you have an example?

https://jinja.palletsprojects.com/en/3.1.x/api/#jinja2.Environment

- `block_start_string` The string marking the beginning of a block. Defaults to '{%', replaced by '#%'.
- `block_end_string` The string marking the end of a block. Defaults to '%}', replaced by '%#'.
- `variable_start_string` The string marking the beginning of a print statement. Defaults to '{{', replaced by '#{'.
- `variable_end_string` The string marking the end of a print statement. Defaults to '}}', replaced by '}#'.
- `comment_start_string` The string marking the beginning of a comment. Defaults to '{#', replaced by '##'.
- `comment_end_string` The string marking the end of a comment. Defaults to '#}', replaced by '##'.

> Jinja is a preprocessor, so if it sees it's keywords, it will interpret them otherwise the file is left untouched and can be parsed just like before.

Understood, but I am saying that it would confuse syntax highlighting when editing these templates. Adding the hash symbol will make Jinja statements look like a YAML comment.

> > This would allow #348 to work without Jinja as well.
> 
> Schema validation is a great idea! have you considered using [pydantic](https://docs.pydantic.dev/latest/) for this?

Thanks for the hint on Pydantic. I should put this on the list, though maybe into #223 instead of #348.

liambeguin, 2024-06-10:
@martinrieder oh I see. I mean why not, but not all lines will be starting with a comment char. the content of blocks will not make a lot of sense if you don't consider the jinja code.

If you concern is just for code highlighting, parsers that understand jinja usually know that it's a preprocessor and allow for a 2 level highlighting.

> Thanks for the hint on Pydantic. I should put this on the list, though maybe into #223. Have another look at #348: I have just uploaded the YAML schema for Yamale.

I didn't know about yamale, interesting.. I usually just do `Ruamel` + `jsonschema`, but more recently pydantic has removed the need for manual schema handling.

liambeguin, 2024-06-29:
@martinrieder I like the approach of having a directory structure interpreted as YAML. I think it would be great to load a library of connectors and wires for example, but I'd keep it limited to that (that lib could also be a part of the repo, allowing to reuse and share many basic components). I think I mentioned it earlier, but I'd also recommend having a single way of including files in the yaml syntax. I'd have the yaml tree loader be a part of the code, with the option of passing in additional paths from the command line (to load a custom library for example).

snelweg, 2025-03-09:
Any news on this feature?

BugInACode, 2025-06-30:
I, too, have a need for this feature. Currently I only see my needs extending to having a single yaml file, or maybe a file for families of similar connectors/cables, which I can simply import into the new harness. After this  I think then the templates should be usable. 

Is it possible that there is a script which will prepend any templates to the file such that they can be used?

BugInACode, 2025-07-02:
I have somewhat of a workaround for the issue of breaking paths.

The workaround involves having the images in a high enough directory where the template file (in the same directory as the main yaml) can reference. 

This gave me the idea that maybe the templates and images can be in a system folder that is able to be "imported" similar to above.

synovalife, 2025-09-15:
Being able to include files inside the yml file is a feature that would be very useful. Perhaps using a very simple format like the image path would suffice, such as:

include:
  - /path/to/file1
  - /path/to/file2
  - /path/to/file3

I need to output each connector to its own file for drawings and manufacturing and need a master file to view all of the connectors together. Being able to simply include files allows easy updates to connectors where updating a single file would propagate into any other files that reference the base file. I think I may work on this code for myself to use and post if anyone else finds this useful/helpful.