wireviz/WireViz #222: [feature] Add labels on cable ends

andresR8, 2021-03-10:
Hi, great work. 
It would be very useful to have a way to describe labels on the cable ends 
[image]
[image]

formatc1702, 2021-03-20:
Some thoughts on this:

- Strictly speaking, this should be a property of the cable (`label` or similar), and it should allow for different labels on either end, something like `label_left` and `label_right`.
- It would probably be more intuitive to render this information closer to the connector, rather than inside the node for the cable. The question is how to implement this.
- In the simple case, where there is no ambiguity, this could be a property of the connector itself. This would only break when there is more than one cable branching off from a connector... Of course, the `label` attribute can be allowed for both connectors and cables and leave it up to the user.

Extrapolating from this idea, it might make sense to allow general-purpose user-defined attributes for connectors and cables, to allow for labels, but also any other information that is not covered by the standard attributes.

Example proposal:
```yaml
  X1:
    type: Plug
    pincount: 4
    pn: 123
    manufacturer: ACME
    mpn: ABC
    additional_attributes:
      - Sleeving removal: 15 mm
      - Insulation removal: 2 mm
      - Label: A01-5-10
      - Tag: ABCDEFG
    additional_components:
      -
        type: Crimp
        pn: 987
        manufacturer: ACME
        mpn: ZYX
        qty_multiplier: populated
      -
        type: Housing
        pn: 345
        manufacturer: OTHER
```

Result:
[image]

<details>
<summary>GraphViz snippet</summary>

```
	X1 [label=<
<table border="0" cellspacing="0" cellpadding="0">
 <tr><td>
  <table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr>
   <td balign="left">
     X1
   </td>
  </tr></table>
 </td></tr>
 <tr><td>
  <table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr>
   <td>
   <table border="0"><tr><td border="1" style="rounded">1</td></tr></table>
   </td>
   <td balign="left">P/N: 123</td>
   <td balign="left">ACME: ABC</td>
  </tr></table>
 </td></tr>
 <tr><td>
  <table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr>
   <td balign="left">Plug</td>
   <td balign="left">4-pin</td>
  </tr></table>
 </td></tr>
 <tr><td>
  <table border="0" cellspacing="0" cellpadding="3" cellborder="1">
   <tr>
    <td align="left">Sleeving removal</td>
    <td align="left">15 mm</td>
   </tr>
   <tr>
    <td align="left">Insulation removal</td>
    <td align="left">2 mm</td>
   </tr>
   <tr>
    <td align="left">Label</td>
    <td align="left">A01-5-10</td>
   </tr>
   <tr>
    <td align="left">Tag</td>
    <td align="left">ABCDEFG</td>
   </tr>
  </table>
 </td></tr>
 <tr><td>
  <table border="0" cellspacing="0" cellpadding="3" cellborder="1">
   <tr>
    <td port="p1r">1</td>
   </tr>
   <tr>
    <td port="p2r">2</td>
   </tr>
   <tr>
    <td port="p3r">3</td>
   </tr>
   <tr>
    <td port="p4r">4</td>
   </tr>
  </table>
 </td></tr>
 <tr><td>
  <table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr>
   <td balign="left">Additional components</td>
  </tr></table>
 </td></tr>
 <tr><td>
  <table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr>
   <td align="left" balign="left" sides="tbl">
    <table border="0"><tr><td border="1" style="rounded">4</td></tr></table>
   </td>
   <td align="left" balign="left" sides="tb">4x</td>
   <td align="left" balign="left" sides="tb">Crimp</td>
   <td align="left" balign="left" sides="tb">P/N: 987</td>
   <td align="left" balign="left" sides="tbr">ACME: ZYX</td>
  </tr>
  <tr>
   <td align="left" balign="left" sides="tbl">
    <table border="0"><tr><td border="1" style="rounded">6</td></tr></table>
   </td>
   <td align="left" balign="left" sides="tb">1x</td>
   <td align="left" balign="left" sides="tb">Housing</td>
   <td align="left" balign="left" sides="tb">P/N: 345</td>
   <td align="left" balign="left" sides="tbr">OTHER</td>
  </tr></table>
 </td></tr>
</table>
> fillcolor=white margin=0 shape=none style=filled]
	X2 [label=<
<table border="0" cellspacing="0" cellpadding="0">
 <tr><td>
  <table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr>
   <td balign="left">X2</td>
  </tr></table>
 </td></tr>
 <tr><td>
  <table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr>
   <td balign="left">P/N: 234</td>
   <td balign="left">ACME: DEF</td>
  </tr></table>
 </td></tr>
 <tr><td>
  <table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr>
   <td balign="left">Receptacle</td>
   <td balign="left">4-pin</td>
  </tr></table>
 </td></tr>
 <tr><td>
  <table border="0" cellspacing="0" cellpadding="3" cellborder="1">
   <tr>
    <td port="p1l">1</td>
   </tr>
   <tr>
    <td port="p2l">2</td>
   </tr>
   <tr>
    <td port="p3l">3</td>
   </tr>
   <tr>
    <td port="p4l">4</td>
   </tr>
  </table>
 </td></tr>
 <tr><td>
  <table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr>
   <td balign="left">Additional components</td>
  </tr></table>
 </td></tr>
 <tr><td>
  <table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr>
   <td align="left" balign="left">1 x #5 (Crimp)</td>
  </tr></table>
 </td></tr>
</table>
> fillcolor=white margin=0 shape=none style=filled]
```

</details>

formatc1702, 2021-03-20:
Alternative implementation idea (separate and independent from the `additional_attributes` proposal:

- You need the label to appear in the BOM
- If multiple cables use the same type of label (but different text on it), they should count as the same BOM item
  - Solution: Add a `note` field for additional BOM items that is ignored when checking for uniqueness

```yaml
  X1:
    type: Plug
    pincount: 4
    pn: 123
    manufacturer: ACME
    mpn: ABC
    additional_components:
      -
        type: Crimp
        pn: 987
        manufacturer: ACME
        mpn: ZYX
        qty_multiplier: populated
      -
        type: Housing
        pn: 345
        manufacturer: OTHER
      -
        type: Label
        pn: 789
        note: '"A01-5-10"'

```

[image]