wireviz/WireViz #223: [internal] Switch from PyYAML to Ruamel?

formatc1702, 2021-03-20:
This would [add support for YAML 1.2](https://yaml.readthedocs.io/en/latest/pyyaml.html) and solve #213.
Meanwhile, YAML 1.2 support seems to still be an [open issue for PyYAML](https://github.com/yaml/pyyaml/issues/116).

Any opinions on this matter?

martinrieder, 2024-05-17:
Referring to the discussion in #305:

> It seems to be how the [current YAML parser](https://pyyaml.org/wiki/PyYAML) is intended to work. It might change in the future, e.g. if deciding to use a parser supporting [YAML version 1.2](https://yaml.org/spec/1.2/spec.html) - where the unquoted words Yes/No and On/Off are no longer interpreted as aliases for the boolean values true/false. See also the related issues #213 and #223.
> 
> To [override how to interpret the type of these unquoted words](https://stackoverflow.com/questions/36463531/pyyaml-automatically-converting-certain-keys-to-boolean-values), will have these effects:
> - It's a global change and thus applies to all YAML input.
> - The WireViz input language no longer conforms to the YAML 1.1 specification.

[...] 

> @EvilMustacheTwirl is making a good point. Interpreting everything as a string could help fix some issues. Numbers and Booleans should be evaluated only where the WireViz syntax requires this. Would a simple regex replacement hack possibly also do the job? 

I realize that the above will not bring the benefit of linking the YAML input to line numbers. The YAML parser replacement seems essential for linking the error output to the input file. This would be required for https://github.com/nanangp/vscode-wireviz-preview/issues/7

martinrieder, 2024-05-17:
I did a bit of research and found that https://github.com/nexB/saneyaml might be lower hanging fruit than Ruamel.

martinrieder, 2024-05-21:
For anyone who wants to dive a bit deeper into the topic, I recommend reading the following document:

https://eemeli.org/yaml/#parsing-yaml

This comes from the library that is used to provide syntax highlighting and error checking in https://github.com/redhat-developer/YAML-language-server for VScode. It is not programmed in Python, but we might take advantage of the CLI that it provides:
It relies on parsing YAML into a CST as an intermediate step to provide an AST. This is a unique approach that other libraries do not take, but seems to be just what we need here. 

Side note: Graphviz dot is also supported in VScode through https://github.com/nikeee/dot-language-server

kvid, 2024-05-21:
To enable comparing the alternatives, we should list the goals we want to achieve by switching library. Unless all goals can be achieved, then I guess we also must rank the goals by importance. A few goal candidates:
- Support all YAML features used by WireViz
- Resolve issues #213, #220, #297, #305, #307
- Avoid too much code redesign

martinrieder, 2024-05-30:
Please extend the list of criteria with:
- easy dependency management
- matching schema validation capabilities

martinrieder, 2024-06-06:
The parser might need to provide default values other than `none` if some attributes are missing. See the proposed use case here: https://github.com/wireviz/WireViz/issues/257#issuecomment-2151346345

martinrieder, 2024-06-14:
I would like to add another library with some interesting features to the list that should be evaluated: [fabiocaccamo/python-benedict](https://github.com/fabiocaccamo/python-benedict)

DeflateAwning, 2025-09-21:
+1 for this. It seems like it should be an easy switch to make (swapping out the library). It's been years and PyYAML still hasn't made the switch.