wireviz/WireViz #224: Improving additional component display, BOM item numbers

formatc1702, 2021-03-20:
I have been using WireViz at work quite a bit (work being the main reason I haven't had time to contribute to the project 😕) and I have noticed the treatment of "main" components and additional components is not as consistent as it maybe should be.

*Note: The following thoughts might partially contradict previous opinions and decisions... this is what happens when theory meets practice.*

Take the following sample file:

[image]

<details>
<summary>Source</summary>

```yaml
connectors:
  X1:
    type: Plug
    pincount: 4
    pn: 123
    manufacturer: ACME
    mpn: ABC
    additional_components:
      -
        type: Crimp
        pn: 987
        manufacturer: ACME
        mpn: ZYX
        qty_multiplier: populated
      -
        type: Housing
        pn: 345
        manufacturer: OTHER

  X2:
    type: Receptacle
    pincount: 4
    pn: 234
    manufacturer: ACME
    mpn: DEF
    additional_components:
      -
        type: Crimp
        pn: 876
        manufacturer: ACME
        mpn: WVU

cables:
  W1:
    wirecount: 4
    color_code: DIN

connections:
  -
    - X1: [1-4]
    - W1: [1-4]
    - X2: [1-4]

```

</details>

Issues (easily solvable)

- When using the image file on its own (without the exported BOM), part number info is not visible for additional components. This might be desirable when, e.g. pasting the image file into a larger documentation document. It is probably more consistent to show info for additional components as well.
- Additional components show their assigned item number in the image (`#4` etc.), where as main components (connectors and cables) do not. It should be either both, or none.
  - How about a global option to `show_bom_item_numbers` that then applies to both?
- Personally, I find the `#` symbol visually jarring.. perhaps there is a more elegant solution
  - Maybe [circled numbers](http://xahlee.info/comp/unicode_circled_numbers.html)? ⓪ ① ② ③
    They appear to go at least up to ㊿.
  - Another idea (though potentially cumbersome) would be to use a 1x1 table to make BOM numbers appear in a box.
    This is similar to "balloons" used in mechanical technical drawings
- The additional components section within the component node could be a table with invisible column separators, to get a  more even look when it starts getting more populated.
[... 215 lines cut ...]
   <tr>
    <td>X1:4</td>
    <td>
     4:YE
    </td>
    <td>X2:4</td>
   </tr>
   <tr>
    <td colspan="3" border="0" cellspacing="0" cellpadding="0" port="w4" height="6">
     <table cellspacing="0" cellborder="0" border="0">
      <tr><td colspan="3" cellpadding="0" height="2" bgcolor="#000000" border="0"></td></tr>
      <tr><td colspan="3" cellpadding="0" height="2" bgcolor="#ffff00" border="0"></td></tr>
      <tr><td colspan="3" cellpadding="0" height="2" bgcolor="#000000" border="0"></td></tr>
     </table>
    </td>
   </tr>
   <tr><td>&nbsp;</td></tr>
  </table>
 </td></tr>
</table>
> fillcolor=white margin=0 shape=box style=""]
}

```
</details>

martinrieder, 2024-06-04:
@formatc1702 wrote:
> I have been using WireViz at work quite a bit (work being the main reason I haven't had time to contribute to the project 😕) and I have noticed the treatment of "main" components and additional [...]

I was going to argue that the `additional_bom_items` in the syntax are inconsistent with`additional_components`, but then I found your thoughts here. I need to take a closer look at the suggestions, yet. Since some time has passed, you might have gained some more 'practical' insights... Any updates on your opinions and decisions? Are you still open for discussion on the items below? 

> Other thoughts (discussion required)
> 
> - Should user-defined BOM item numbers be allowed?
>   E.g. someone might want connectors numbered from 1, cables numbered from 50, use letters for crimps, etc...
> - Is it cleaner to use some indented auto-numbering scheme for additional components?
>   E.g. A `X1` is item `1`, then the crimp could be `1a`, the housing `1b`.
>   - Obviously this is problematic when using the same crimp for various connectors with different pincounts, etc. I thought I would bring it up anyways

I would say that the most consistent implementation is this:

- Define a global section for `additional_components`
- Use designators for each item, making sure they are unique and not in conflict with connectors and cables 
- References to these designators from both cables and connectors (or any other items) 

> The only thing getting lost in this case would be any `additional_bom_items` definde outside of connectors and cables, but I see that as a separate issue.
> 
> Personally, I would vote for having `show_bom_item_numbers = False` and  `show_part_numbers = True` by default. This way, no information is lost when using only the graphics output, where as BOM cross-referencing is left as an "advanced" feature.

Keep in mind that there are two common use cases:

- Connector crimps or single wire seals that need to be multiplied by pin count
- Wire tubing or sleeves that need to be multiplied with cable length

The latter might cause confusion because wire count is another possible multiplication factor.

kvid, 2024-06-12:
I suggest keeping:
- The BOM item numbering strictly numeric, and auto-assigned after BOM de-duplication.
- A global `additional_bom_items` and local `additional_components` for each connector/cable.

However, I suggest we add a local designator suffix for each `additional_components` item, e.g. auto-assigned numeric (with some separator character between the global connector/cable designator and the local designator suffix) unless perhaps optionally specified differently for each item. We must require such local designators to be unique within the local scope (connector/cable) and the combined designators (e.g. `X1-1`, `X1-2`...) must be globally unique so the user cannot specify e.g. a global designator `X1-2` that conflicts with the local designator suffix `-2` to a global connector designator `X1`.

In addition, all items in the `designators` attribute of each `additional_bom_items` entry should also be included in the global scope together with connector/cable designators in the uniqueness checks described above.

The short designators suggested in https://github.com/wireviz/WireViz/issues/350#issuecomment-2150763321 should then also be unique within the same local scope I suggest above, so such short items are automatically included among the `additional_components` of the same connector, both in the local list inside the connector node, and in the global BOM, as I very briefly suggested in https://github.com/wireviz/WireViz/pull/369#issuecomment-2156432482

tobiasfalk, 2024-06-13:
Loops and shorts are often done with wires and therefore can not only have a color(for the isolation) bit also a gauge and so on, that also means that someone may wants to have the short in the list with all the wires.

kvid, 2024-06-13:
@tobiasfalk wrote:
> @kvid Loops and shorts are often done with wires

I agree to that. 😄 

> and therefore can not only have a color(for the isolation) bit also a gauge and so on,

I'm not sure I interprete this part correctly. Do you mean "and therefore always need to specify more attributes than only color and gauge"?

If so, then I agree in most cases, but I've also seen very simple use cases where the diagram only is used to show the electrical connections, and has no needs for more details than perhaps different colors for readability.

> that also means that someone may wants to have the short in the list with all the wires.

What "list with all the wires" do you mean?

Is it the global BOM? If so, then I agree in most cases, except in very simple use cases like I describe above where a BOM is not needed at all.

Is it the list of wires in a bundle? If so, then I wonder how such a bundle should be shown in relation to the connector having the loops?

Do you mean a bundle node close to the connector with the loops drawn between these two nodes? If so, then I'm worried it will most often look a bit ugly because `dot` is not good at drawing a circular graphs.

Do you mean the wire list of a bundle embedded into the connector node, e.g. just below the pin list, and loops at both sides of the connector to connect the pins to each end of the wire entry in such a wire list? If so, then it might work out well, except I'm a bit worried about readability when the number of loops increases.

tobiasfalk, 2024-06-13:
@kvid 
What I mean with the same list as the wires is that when you define a W1 with the gauge of .5mm^2 and also have a short with the same gauge, then one may want it in the same BOM entry for total wire length

tobiasfalk, 2024-06-13:
Also the color can either be only for the purpose of differentiation bur could also be used for defining the color of the wire isolation

martinrieder, 2024-06-13:
> # Local Designators
> I suggest keeping:
> 
> * The BOM item numbering strictly numeric, and auto-assigned after BOM de-duplication.
> 
> However, I suggest we add a local designator suffix for each `additional_components` item, e.g. auto-assigned numeric (with some separator character between the global connector/cable designator and the local designator suffix) unless perhaps optionally specified differently for each item. We must require such local designators to be unique within the local scope (for each connector/cable), and the combined designators (e.g. `X1-1`, `X1-2`, ...) must be globally unique so the user cannot specify e.g. a global designator `X1-2` that conflicts with the local designator suffix `-2` to a global connector designator `X1`.

Going by the [Graphviz dot language specification](https://www.graphviz.org/doc/info/lang.html), the ID could contain any character when it is a quoted string. The exception is:

> In addition, the content must be legal XML, so that the special XML escape sequences for ", &, <, and > may be necessary in order to embed these characters in attribute values or raw text. 

Therefore, I suggest to put the auto-assigned numbering into brackets like: `X1<1>, X1<2>, X1<3>, ...`

tobiasfalk, 2024-06-13:
Why not just add one ore multiple shorts or loops to a additional component, and I we should not put the pin definition in to the additional component but keep it separate.
```
additional_components:
    - Sh1, Sh2, L1
      color: PK
      mpn: 42A5Z42
      type: 3 pin Short ov er 5 pins, name of the part
```

Also for shorts and loops wit wire/cable I think somthing like `additional_cable` would be good, this is practically `additional_components` but with the type `Cable`

martinrieder, 2024-06-13:
@tobiasfalk wrote:
> Why not just add one ore multiple shorts or loops to a additional component, and I we should not put the pin definition in to the additional component but keep it separate.
> 

Also possible, but please add some `[]` brackets to indicate an array:
> ```yaml
> additional_components:
>     - [Sh1, Sh2]
>       color: PK
>       mpn: 42A5Z42
>       type: 3 pin Short over 5 pins
> ```

kvid, 2024-06-13:
@tobiasfalk wrote:
> @kvid What I mean with the same list as the wires is that when you define a W1 with the gauge of .5mm^2 and also have a short with the same gauge, then one may want it in the same BOM entry for total wire length

Yes, that is normal behavior. I called it "BOM de-duplication" in my last comment. To obtain a match between a wire in `additional_components` and the same wire in a bundle, we need all these properties of the BOM entries to be equal: ("description", "unit", "pn", "manufacturer", "mpn", "supplier", "spn")

"description" of a bundle wire = Wire, type, gauge, color
"description" of an additional component = type, subtype  (I also suggested above to append color here)

That means if "type, subtype" of an additional component == "Wire, type, gauge" of a connector, then they match and they will be de-duplicated if also all remaining properties are equal. In my example above, I used `type: Wire, 24 AWG` to obtain this.

@tobiasfalk wrote:
> Also the color can either be only for the purpose of differentiation bur could also be used for defining the color of the wire isolation

Yes, I don't see any conflict between this and my suggestions above.

kvid, 2024-06-13:
@tobiasfalk and @martinrieder - All your YAML input suggestions contain syntax errors. This is an example of an online YAML parser where you can test your syntax:  https://codebeautify.org/yaml-parser-online

Each `additional_components` must either have only list items, or only key/value items. Trying to mix will result in syntax error.

Why do you want a list of local designators for an entry? If you want some entries to be a copy of another entry, you can use inheritance.

Please explain your motivation for suggesting these strange non-YAML syntax entries, and we might find alternatives that are allowed as YAML input.

martinrieder, 2024-06-13:
Sorry, I have put the `:` back to where they were in the example above and now it passes the checks. I used https://www.yamllint.com/ instead, which seems to use a different parser than the one you suggested...
The suggestion to use an array of designators is indeed a bit odd. It is inspired by the way that multiple arrows can be defined in the connections list. I guess we do not really need this special case.

kvid, 2024-06-13:
Be aware that https://www.yamllint.com/ seems to use YAML v1.2 which is different from v1.1 used by WireViz.

tobiasfalk, 2024-06-14:
```
additional_components:
    - [Sh1, Sh2, L1]:
      color: PK
      mpn: 42A5Z42
      type: 3 pin Short ov er 5 pins, name
```
This past the test, the idea behind it is to not have to type the same thing multiple times

martinrieder, 2024-06-14:
> Be aware that https://www.yamllint.com/ seems to use YAML v1.2 which is different from v1.1 used by WireViz.

Thanks. I could not find that info directly on the website. This is another reason for #223 and #348.
The discussion is getting off topic if we are to discuss missing characters. It is the idea of (auto-)assigning designators to specific features that should be discussed. 

I prefer the original idea with ballooned numbers, so I strongly recommend using lists (arrays) to make that possible.

ferdnyc, 2024-06-16:
@martinrieder 

> Going by the [Graphviz dot language specification](https://www.graphviz.org/doc/info/lang.html), the ID could contain any character when it is a quoted string. The exception is:
> 
> > In addition, the content must be legal XML, so that the special XML escape sequences for ", &, <, and > may be necessary in order to embed these characters in attribute values or raw text.
> 
> Therefore, I suggest to put the auto-assigned numbering into brackets, which would also closely resemble the balloon numbers discussed above: `X1<1>, X1<2>, X1<3>, ...` The syntax would assure uniqueness, because users would need to explcitly use escape sequences if they wanted a designator like this.

Oh, no, that's a misinterpretation. The restrictions you quoted were for _HTML-like strings_, which have to be surrounded by `<` and `>` in their entirety, _instead of_ being quoted.

```dot
graph G {
  // In other words, both of these are invalid graphviz:
  X1<1> [color=red];        // Creates 2 nodes, "X1" and "1"
  X1&lt;1&gt; [color=red];  // Syntax error
  // But this is perfectly legal:
  "X1<1>" [color=green shape=box];
  // (And this, which is the form that note was describing):
  <X1&lt;1&gt;> [color=blue shape=box];
}
```

So, if you're using _quoted_ strings, there's no restriction on including `<` or `>` in the contents, and therefore no restrictions on users entering names like `X1<1>`.

If you're using _HTML-like strings_ for node names... well, I wouldn't recommend that, because it gets ugly fast. (And wouldn't really solve the problem, because you'd end up with a user who wanted to use a reserved character like `&` or `>` in their labels, so you'd end up having to HTML-escape their input anyway.)

Better, I think, to just surround _the user-provided content_ with some characters — in _all_ uses — to ensure uniqueness.

For example:
* Global `X1` combined with local designator `2`:
   * Store global ID as `[X1]` (with `X1` in the `label`)
   * Store local designator as `[X1].2`
   * A global `X1.2` would become `[X1.2]` and `[X1.2].2`

Even if the user creates a global ID `[X1].2`, it'll be _stored_ as `[[X1].2]`, and the local designator will be `[[X1].2].2`, so there are no collisions possible. (It just requires care when _parsing_ those names, if it ever comes to that.)

But unless you're _modifying_ the user's content in some way when storing it, there's **always** a chance they can enter a string that, unmodified, collides with IDs generated from other unmodified names.