wireviz/WireViz #225: [feature] Define page background color, line color, text color

formatc1702, 2021-03-20:
This is the next step after #219.

Any area that accepts a background color should have the option to set a foreground/text color, e.g. for good contrast against a dark background.

Additionally, the possibility of a "dark theme", with light text and lines on a dark background, would be very appealing.

[image]

kvid, 2021-05-05:
I'm currently testing some draft code that will be able to read YAML input like this:
```yaml
X1:
  pincount: 4
  box:
    color: WH
    bgcolor: BK
    fontcolor: IV
    fontname: times
    fontsize: 12
  title:
    bgcolor: RD
    fontsize: 20
```
I guess specifying all this for each connector and cable is not wanted by anyone, but it might be useful to have the option. By grouping these attributes under `box` and `title` it is easy to use common templates.

If #214 is accepted, I also have a draft code for harness options that will be able to read YAML input like this:
```yaml
options:
  base:
    color: WH
    bgcolor: BK
    fontcolor: IV
    fontname: times
    fontsize: 16
  connector:
    bgcolor: RD
  cable:
    bgcolor: BU
  bundle:
    bgcolor: GN
```
These will be used for the whole harness unless overridden for some connectors/cables as described in the upper part of my comment.

kvid, 2021-05-05:
By moving the `title` dict into `box` and likewise into all the option dicts with color/font, it becomes easier to specify common `title` attributes.

kvid, 2022-01-30:
@formatc1702 wrote above:
> 
> I propose `Style`/`NodeStyle` instead of `Look`; this would be consistent with the nomenclature in other programming/markup languages.
> 
> I am aware that `style` is currently an attribute for connectors (with `style="single"` the only currently supported value besides `None`), but I am strongly considering to consolidate this with the `category` attribute already present in cables.
> 

If the `style` attribute gets available, then I adjust my suggested syntax to accept an input like this:
```yaml
options:
  style:
    bordercolor: WH
    bgcolor: BK
    fontcolor: IV
    fontname: times
    fontsize: 16
    nodes:
      title:
        color: YE
    connectors:
      bgcolor: RD
      title:
        fontsize: 18
    cables:
      bgcolor: BU
      title:
        fontsize: 20
    bundles:
      bgcolor: GN
connectors:
  X1:
    pincount: 4
    style:
      bordercolor: WH
      bgcolor: BK
      fontcolor: IV
      fontname: times
      fontsize: 12
      title:
        bgcolor: RD
        fontsize: 20
```
This way, it becomes easy to define and use style templates and even theme templates for the options.style attribute. It also becomes easier to implement the style dataclasses independently of any inheritance between the dataclasses containing style attributes:
```py
@dataclass
class Style:
    """Colors and font that defines how an element should look like."""
    bgcolor: Optional[Color] = None
    bordercolor: Optional[Color] = None
    fontcolor: Optional[Color] = None
    fontname: Optional[PlainText] = None
    fontsize: Optional[Points] = None

@dataclass
class NodeStyle(Style):
    """How a node look like possibly with a different looking title."""
    title: Style = field(default_factory=dict)

@dataclass
class DiagramStyle(Style):
    """How a diagram look like possibly with different looking node classes."""
    nodes: NodeStyle = field(default_factory=dict)
    connectors: NodeStyle = field(default_factory=dict)
    cables: NodeStyle = field(default_factory=dict)
    bundles: NodeStyle = field(default_factory=dict)

@dataclass
class Options:
    color_mode: ColorMode = 'SHORT'
    mini_bom_mode: bool = True
    style: DiagramStyle = field(default_factory=dict)

@dataclass
class Image:
    ...
    style: Optional[Style] = None
    ...

@dataclass
class Connector:
    ...
    style: Optional[NodeStyle] = None
    ...

@dataclass
class Cable:
    ...
    style: Optional[NodeStyle] = None
    ...
```