wireviz/WireViz #232: [Suggestion] Possible improvement on multi row connectors

ulek, 2021-04-17:
Hey, 
I decided to try and capture something in my projects.
This particular thing has many twisted shielded pairs (I know that is being worked on)
The other thing that it utilizes multi row connectors (either 2 or 3). 
Very often on those type of connectors the signals are placed on multiple rows next to each other. I.e. in 16 pin 2 row connector very often pins 1 and 9 carry + and - of the same signal. 
Now if you just lay out the pins numerically it, it creates a mess as attached
[image]

Maybe there could some property of a connector that would group the pins either "logically" or "per row basis", 1 right next to 9, 2 right next to 10 etc. Placing pins in that order on the graph would allow to deal better with the resulting rats nest.
I'm not sure what would be the best way of dealing with bit. Maybe staggering/offsetting the second row pins with respect to the first row, be in line but have some different coloring.

formatc1702, 2021-04-18:
Have you tried using the `pins` attribute of the connector? Here's a minimal example.

```yaml
X1:
  pins: [1,9,2,10]
  pinlabels: [sig1+, sig1-, sig2+, sig2-]
```

This way, you can force the pin order to whatever you need.

ulek, 2021-04-18:
Actually just this morning noticed that example. So my bad on that end. 

However it still would be nice if we could somehow specify a connector property that would automatically for you.
like a maybe a multirow property that you would say how many rows, how pins set up etc.

I have untangled one of the connectors but the other one is a weird 3 row (8-7-8) no fighting with that one at the moment
Once I have that I'll do my best to update here

kvid, 2021-04-19:
> However it still would be nice if we could somehow specify a connector property that would automatically for you.
> like a maybe a multirow property that you would say how many rows, how pins set up etc.

How about a new alternative to specify `pincount` as a list of pincounts in each pin group (for each row in your case)? In your case, it could look something like this:
```yaml
connectors:
  J1-M:
    type: Samtec
    subtype: female
    pincount: [8, 8]
    pinlabels: [sig1+,sig2+,sig3-,NC,SHIELD,NC,NC,sig4+,sig1-,sig2-,sig3+,NC,NC,NC,NC,sig4-]
  J2-I:
    type: Samtec
    subtype: female
    pincount: [8, 7, 8]
    pinlabels: [a,b,sig1+,sig2-,NC,d,e,f,g,h,sig3+,sig2+,NC,i,j,k,sig1-,sig3-,GND,NC,m,n,o]
```
I suggest rendering such pin groups like this:
[image]
If combined with `pins` with custom pin names and/or ordering, the custom pins can also be split into groups similar to the default numeric pin names as shown above. Is this flexible enough, or do we also sometimes need to specify how to interleave the pin groups?
 - In the example above, the first pin from each group are strictly ordered by group index.
 - One alternative could be to order the first pin from large groups above the first pin from smaller groups, e.g. pin 16 above pin 9 for `J2-I` in your case.
 - Maybe even a custom group order might be needed sometimes?

ulek, 2021-04-25:
I think that they way you showed with (i.e. specify row types and use functional group) is definitely workable  
I think that it should cover most of the use cases.

catherinepapad, 2024-11-26:
> > However it still would be nice if we could somehow specify a connector property that would automatically for you.
> > like a maybe a multirow property that you would say how many rows, how pins set up etc.
> 
> How about a new optional feature to allow the `pincount` value to be a list of pincounts in each pin group (for each row in your case)? Another alternative is to use a new `pingroups` attribute for such a list value, and this latter alternative might even be easier to implement. In your case, it could look something like this:
> 
> ```yaml
> connectors:
>   J1-M:
>     type: Samtec
>     subtype: female
>     pincount: [8, 8]
>     pinlabels: [sig1+, sig2+, sig3-, NC, SHIELD, NC, NC, sig4+,
>                 sig1-, sig2-, sig3+, NC, NC,     NC, NC, sig4-]
>   J2-I:
>     type: Samtec
>     subtype: female
>     pincount: [8, 7, 8]
>     pinlabels: [a, b,     sig1+, sig2-, NC, d, e, f,
>                 g, h,     sig3+, sig2+, NC, i, j,
>                 k, sig1-, sig3-, GND,   NC, m, n, o]
> ```
> 
> I suggest rendering such pin groups like this: [image] If combined with `pins` with custom pin names and/or ordering, the custom pins can also be split into groups similar to the default numeric pin names as shown above. Is this flexible enough, or do we also sometimes need to specify how to interleave the pin groups?
> 
> * In the example above, the first pin from each group are strictly ordered by group index.
> * One alternative could be to order the first pin from large groups above the first pin from smaller groups, e.g. pin 16 above pin 9 for `J2-I` in your case.
> * Maybe even a custom group order might be needed sometimes?

Hello @kvid 

Could you provide the code you use to connect the two mentioned connectors? I am trying to implement something similar, but I cannot connect the connectors, I get an error when building.

Thank you in advance!

kvid, 2024-11-27:
@catherinepapad wrote:
> [...]
> Hello @kvid
> 
> Could you provide the code you use to connect the two mentioned connectors? I am trying to implement something similar, but I cannot connect the connectors, I get an error when building.
> 
> Thank you in advance!

Unfortunately, I never implemented any code to support this suggested functionality. To create the suggested diagram, I manually modified the `.gv` file produced by WireViz to add extra pin table columns and distributed the pin names into these new columns

<details>
  <summary>Expand for YAML input</summary>

  ```yaml
graph {
// Graph generated by WireViz 0.3-dev
// https://github.com/formatc1702/WireViz
	graph [bgcolor=white fontname=arial nodesep=0.33 rankdir=LR ranksep=2]
	node [fillcolor=white fontname=arial shape=record style=filled]
	edge [fontname=arial style=bold]
	"J1-M" [label=<
<table border="0" cellspacing="0" cellpadding="0">
 <tr><td>
  <table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr>
   <td balign="left">J1-M</td>
  </tr></table>
 </td></tr>
 <tr><td>
  <table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr>
   <td balign="left">Samtec</td>
   <td balign="left">female</td>
   <td balign="left">16-pin</td>
  </tr></table>
 </td></tr>
 <tr><td>
  <table border="0" cellspacing="0" cellpadding="3" cellborder="1">
   <tr>
    <td>sig1+</td>
    <td sides="BLT">1</td>
    <td sides="BRT" port="p1r"></td>
   </tr>
   <tr>
    <td>sig1-</td>
    <td sides="BLT"></td>
    <td sides="BRT" port="p9r">9</td>
   </tr>
   <tr>
    <td>sig2+</td>
    <td sides="BLT">2</td>
    <td sides="BRT" port="p2r"></td>
   </tr>
   <tr>
    <td>sig2-</td>
    <td sides="BLT"></td>
    <td sides="BRT" port="p10r">10</td>
   </tr>
   <tr>
    <td>sig3-</td>
    <td sides="BLT">3</td>
    <td sides="BRT" port="p3r"></td>
   </tr>
   <tr>
    <td>sig3+</td>
    <td sides="BLT"></td>
    <td sides="BRT" port="p11r">11</td>
   </tr>
   <tr>
    <td>NC</td>
    <td sides="BLT">4</td>
    <td sides="BRT" port="p4r"></td>
[... 399 lines cut ...]
   </tr>
   <tr>
    <td colspan="3" border="0" cellspacing="0" cellpadding="0" port="w2" height="6">
     <table cellspacing="0" cellborder="0" border="0">
      <tr><td colspan="3" cellpadding="0" height="2" bgcolor="#000000" border="0"></td></tr>
      <tr><td colspan="3" cellpadding="0" height="2" bgcolor="#ffffff" border="0"></td></tr>
      <tr><td colspan="3" cellpadding="0" height="2" bgcolor="#000000" border="0"></td></tr>
     </table>
    </td>
   </tr>
   <tr><td>&nbsp;</td></tr>
   <tr>
    <td>J1-M:5:SHIELD</td>
    <td>Shield</td>
    <td>J2-I:19:GND</td>
   </tr>
   <tr><td colspan="3" cellpadding="0" height="2" bgcolor="#000000" border="0" port="ws"></td></tr>
   <tr><td>&nbsp;</td></tr>
  </table>
 </td></tr>
</table>
> fillcolor=white margin=0 shape=box style="filled,dashed"]
}
```
</details>

catherinepapad, 2024-12-13:
> @catherinepapad wrote:
> 
> > [...]
> > Hello @kvid
> > Could you provide the code you use to connect the two mentioned connectors? I am trying to implement something similar, but I cannot connect the connectors, I get an error when building.
> > Thank you in advance!
> 
> Unfortunately, I've not (yet) implemented any code to support this suggested functionality. To create the suggested diagram, I manually modified the `.gv` file produced by WireViz to add extra pin table columns and distributed the pin names into these new columns. I then executed the Graphviz dot tool to produce the diagram image from the modified `.gv` input.
> 
> Expand to see my manually modified Graphviz dot input

Thank you very much for providing your code!