wireviz/WireViz #235: [License] Multi License Binaries

chreg, 2021-07-20:
Hi all,

As mentioned in a previous license issue, I would propose to offer multi license binaries to be able to use this application also in proprietary products without breaking copyleft philosophy to the source code. See the PlantUML project https://plantuml.com/download

Regards
Chreg

chreg, 2021-08-20:
Can nobody answer this?

kvid, 2021-08-20:
I think such questions can only be answered by the owner, @formatc1702, but he has not been very active here the last months. I guess he might be quite busy.

chreg, 2021-08-20:
@kvid thanks for reply.

formatc1702, 2021-08-22:
Hi,
indeed I haven't had time to tend to the project recently; but I don't want to leave you hanging.

Personally, I see no issue in offering multiple binary versions. The more people are able to use this tool, the better!
However, I don't have much knowledge on the details... what versions might be needed, how to easily generate them, and even how compiled binaries would be then used by others. FYI: I am a hardware guy doing a bit of Python, not a software engineer ;-)

I will welcome any proposal to easily generate different binaries depending on your needs, but I cannot currently put time and effort into implementing anything like this. So feel free to write a more detailed description of what you envision, a PR that automates the release process, etc.

chrisalbertson, 2021-08-22:
This is GPL'd, it is already possible to use this as part of a
proprietary system if you (1) state that it is covered by GLP and (2) point
the user to this Github repository where he can get the source code.   As
long as the executable remains separately executable GPL allows it.