wireviz/WireViz #237: Feature - Page Frame with details

leoheck, 2021-08-20:
It it possible to have a frame around the graph with some basic information that we use to have on technical drawings?

For instance, version, date,. company name, and other generic details? 

My main interest is to have a version/date to have an easy and fast way to find the newest diagram.

kvid, 2021-08-20:
See issue #32 and WIP PR #74.

leoheck, 2021-08-20:
This is really good. Do you have any about the status of this feature?

leoheck, 2021-08-20:
I meant. Do you think it will be merged soon, or it is better to use that branch instead, I want to have these features today?

formatc1702, 2021-08-22:
If you want the features today, please use the branch, or integrate it into `master` or `dev` yourself (depending on which one you are normally using).

I have been extensively using a custom branch at work, which implements the frame+title block feature (with a few tiny improvements), and I am very happy with it. So expect this to be one of the next branches to be merged into `dev` officially. I cannot make any promises on when this will happen, though.

closed 2021-08-22 by formatc1702

leoheck, 2021-08-22:
Alright. Makes sense. So, if I got it right, you are saying the discussion/planing phase is over. It is just a matter of integrating the feature. This is good.