wireviz/WireViz #243: Defined in both pins and pinlabels for different pins, specify using pins

Azimath, 2021-09-25:
Hello,
I have some battery management connectors like this:
[image]
Where the pins are named by cell number with some others mixed in. I defined these as pin names, and would like to connect them to harnesses, but am stopped by a "CG13:1 is defined both in pinlabels and pins, for different pins." exception. I'd like to specify using the pin numbers since this simplifies to `CG13: [1-26, 28-40]`, but there doesn't seem to be a way to do this. Am I too sleepy to read the docs properly or is this not supported?

Azimath, 2021-09-25:
I'll just tweak the names and work around this, but would still like to know if it's possible to force the use of names or numbers.

closed 2021-09-25 by Azimath

formatc1702, 2021-09-27:
What version/branch of WireViz are you using?
Can you share a YAML input that exhibits the error?

It sounds like you could use the `pins` attribute to actually change the numbering, instead of using `pinlabels`, which I suspect you are using now.

kvid, 2021-09-27:
> I'm using 0.2 from pip. The txt attached has a stripped down YAML that exhibits the problem.

I have verified that both v0.2 and the current v0.3-dev (`dev` branch) accept the following input, where the only difference from your input, is using `pins` instead of `pinlabels` in `CG13`, as suggested by @formatc1702:
```yaml
connectors:
    CG13:
        pins: [1-, 1, 2]
    RNB1:
        pinlabels: [C10, C8, C6, C4, C2, C0, X, X]
cables:
    W1:
        wirecount: 39
connections:
  -
    - CG13: [1-,2]
    - W1: [1,3]
    - RNB1: [C0, C2]
```
This way, all pin names are unique within the connector, and not ambiguous when considering pin labels too.