wireviz/WireViz #258: Strange output (input, cable and output on separate "lines")

TorbjornHouge, 2021-11-15:
[image]

I am a very new user, but trying to run tutorial examples (and example yml-files), all end up the same, with the input X1, wire W1 and output X2 on separate lines, making them very strange and hard to read compared to the nice versions on Github tutorial.

This happens on all yml-files I have tried, also the last one, but it seems that it consistently places the input on the first row, the wire on second and output on third even when there are several inputs, wires and outputs like it is in the final tutorial example.

formatc1702, 2021-11-15:
That is quite weird! I've never seen something like this myself.

Could you please post

- The WireViz version / GitHub branch you are using
- Your installed GraphViz version
- The contents of the `.gv` file that gets generated along with the image you have posted above?

You can also paste the `.gv` output on https://edotor.net/ yourself to see if it gets rendered correctly there... If it appears fine, it seems like your local GraphViz installation could be to blame...

TorbjornHouge, 2021-11-15:
Thank you for the reply.

WireViz version 0.3.1 (Installed using pip3 today, running on Python 3.10 installed from Windows 10 store).
GraphViz version 2.49.3 (20211023.0002)

.gv file (renamed to .txt to allow it to be uploaded):
[test1.gv.txt](https://github.com/formatc1702/WireViz/files/7540987/test1.gv.txt)
It does, however, look like the png file shown on top when pasting to https://edotor.net/ 

I am, as you see above, running this on Windows, and I installed today, so I am not surprised if that has something to do with it. Also, Windows store installs python into a somewhat odd location I think, but it doesn't give any errors when running it, so I don't really know.

formatc1702, 2021-11-15:
It looks like a newline issue. Look at the first two/three lines of the `.gv` file:
```
graph {
// Graph generated by WireViz 0.3.1// https://github.com/formatc1702/WireViz	graph [bgcolor="#FFFFFF" fontname=arial nodesep=0.33 rankdir=LR ranksep=2]
```
There is a missing newline between the end of the GithUb link, and the `graph []` keyword.

The code should show:
```
graph {
// Graph generated by WireViz 0.3.1// https://github.com/formatc1702/WireViz
	graph [bgcolor="#FFFFFF" fontname=arial nodesep=0.33 rankdir=LR ranksep=2]
```

and editing that line in edotor.net fixes the issue.

formatc1702, 2021-11-15:
I generally develop and use WireViz on macOS, but I have successfully been using a custom branch, very similar to v0.3 on Windows 10 using Python 3.8.10 and GraphViz 2.38.0, with no issues.

Please have a look at `Harness.py`, lines 94 to 97:
```
    def create_graph(self) -> Graph:
        dot = Graph()
        dot.body.append(f'// Graph generated by {APP_NAME} {__version__}')
        dot.body.append(f'// {APP_URL}')
```

What happens when you add `\n` to the end of the two appended strings?
What happens when you add `\r\n` to simulate a Windows-style CR+LF?

TorbjornHouge, 2021-11-15:
Thanks, that fixes it.
Looking a bit more, it appears that the crucial part is that "rankdir=LR" gets commented away and without that it looks strange.

formatc1702, 2021-11-15:
Looks like my last comment beat yours by two seconds, so I'm writing again in case you missed it.

TorbjornHouge, 2021-11-15:
Both \n and \r\n gives good results with the same output to png, the only difference I spot is an extra line in the gv file.
Thanks a lot for the help.

formatc1702, 2021-11-15:
No problem.
We'll have to look into who is introducing the issue:
- Python 3.10 vs 3.8
- GraphViz 2.49 vs 2.38
- The graphviz Python module 0.18.2* vs 0.17

\* 0.18.2 is the current version on PyPI so I assume this is what you have installed.

TorbjornHouge, 2021-11-15:
Correct, 0.18.2

laurierloi, 2021-11-15:
Had similar issue running in WSL

jlecoeur, 2021-11-26:
I encountered the same issue, solved it in the PR linked above.

formatc1702, 2021-11-27:
I can confirm that this behavior is introduced by v0.18 of the `graphviz` Python package; where as v0.17 did include the newline automatically.

I considered filing a bug report, but the current [documentation](https://graphviz.readthedocs.io/en/stable/manual.html?highlight=append#custom-dot-statements) seems to indicate that the final newline should be included in the string:

> [the `body` attribute of the `Graph` or `Digraph` object] holds the verbatim list of lines to be written to the source file (including their newline).

Thanks @jlecoeur for the corresponging PR.

kvid, 2023-08-14:
I know this issue was fixed in the main `branch` and therefore closed in 2021. However, multiple duplicate issues has been reported because the fix is still not included in the `dev` branch or any of the PR branches. By reopening and pinning this issue (until including the fix in the major development branches), I hope it will be easier to discover, and thereby reduce the amount of duplicate issues.

kvid, 2023-08-28:
I cherry-picked the fix for this issue into `dev` yesterday (see https://github.com/wireviz/WireViz/issues/316#issuecomment-1694685896 for details) and when the PR branches get rebased on top of `dev` again, then they also will contain it.

PR #261 (2021-11-27): Fix graphviz file header
The two header comments were missing an endline.

Closes #258
commit 9af0cb8a (2021-11-27): Fix graphviz file header

The two header comments were missing an endline.

Closes #258
commit 9dbf4538 (2023-08-27): Fix graphviz file header (#261)

The two header comments were missing trailing newlines.

This behavior is introduced by v0.18 of the graphviz Python package;
where as v0.17 did include the newline automatically.

Closes #258
closed 2024-05-12 by formatc1702