wireviz/WireViz #259: inline connectors or cables into connections?

egradman, 2021-11-15:
It would be very helpful to be able to inline a connector or a cable into a connection!

```
connectors:
  X1:
    type: D-Sub
    subtype: female
    pinlabels: [DCD, RX, TX, DTR, GND, DSR, RTS, CTS, RI]
  X2:
    type: Molex KK 254
    subtype: female
    pinlabels: [GND, RX, TX]

connections:
  -
    - X1: [5,2,3]
    - W1:
         gauge: 0.25 mm2
         length: 0.2
         color_code: DIN
         wirecount: 3
         shield: true
    - X2: [1,3,2]
```

(this implies an ordering on the cable of `[1-3]`)

martinrieder, 2024-05-30:
Browsing through the issues list, I noticed that this might be a duplicate of #25 and #134.

kvid, 2024-05-30:
@martinrieder wrote:
> Browsing through the issues list, I noticed that this might be a duplicate of #25 and #134. Are these resolved by PR #186?

Thank's for trying to reduce the number of open issues by identifying duplicates. However, this issue seems, IMHO, to be quite independent from the others you mention.

@egradman seems only to request a syntax that allows connector/cable entries embedded in connection entries to enable a slightly more compact YAML input, without any enhanced functionally besides that. Maybe the title is a bit misleading?

I believe other changes of the `connections` syntax that enhances functionality has been given higher priority, but later, when the syntax has matured a bit, this issue might be considered.

martinrieder, 2024-06-04:
Thanks for the clarification. I really did not see that this issue is about the syntax. The suggested change only seems reasonable in such small examples. I do not see how this would improve either file size or readability.

Please see https://github.com/wireviz/WireViz/issues/224#issuecomment-2146398473 about some more detailed reasons. IMHO the YAML list should be used only for referencing on predefined designators.