wireviz/WireViz #260: Loops function

TTROD2MLE, 2021-11-23:
Hi all,
I'm trying to use the Loops function but I can't find any example.
What I want to do is to connect pins on a same connector and I guess this is the right function to use.
Can anybody help me?
Thanks!

formatc1702, 2021-11-23:
Hi,

- The actual `loops` attribute of a connector is very primitive. It is a list, where every entry needs to be a list with exactly two pin numbers. You cannot control the apppearance (color) or any parameters of the loop "wire"
- You can also create a loop using a real `cable` inbetween, but the rendering admittedly leaves something to be desired.

See the following code which implements both variants:
```yaml
connectors:
  X1:
    pincount: 4
    loops:
      - [1, 2]  # simple loop

cables:
  W1:
    wirecount: 1
    colors: [GN]

connections:
  -  # loop using a "real" wire
    - X1: [3]
    - W1: [1]
    - X1: [4]
```

Result:
[image]

formatc1702, 2021-11-23:
See also #48 for discussion on some possible future implementations.

TTROD2MLE, 2021-11-24:
Thank you so much for the quick answer, it will be very useful.

closed 2021-11-24 by formatc1702