wireviz/WireViz #262: Feature request: inporting templates

TorbjornHouge, 2021-12-17:
I have been using this for a few months, and I am finding myself copying and pasting a lot between files in order to keep up to date a complete cable harness for my system. I found myself thinking that much of that could be saved if I could keep a template file where I document connectors (with pinout, image etc) and cables that I use most. That way, if I change the connector or pinout for CAN BUS, it propagetes through the system with less chance of errors.

kvid, 2021-12-17:
Thank you for the feed-back. Have you tried the [--prepend-file](https://github.com/formatc1702/WireViz/blob/master/docs/README.md#command-line-options) command line option with your common file containing [templates](https://github.com/formatc1702/WireViz/blob/master/docs/syntax.md#inheritance) that can be used in several harnesses?

kvid, 2021-12-18:
See also #220 that seems related. If your use case is fully covered by #220, then I suggest you join that discussion and mark this issue as a duplicate of that one.

TorbjornHouge, 2021-12-20:
Thanks. For me the --prepend-file option works, and for now I am quite ok with using a local template for each project so I'll close this. I should have seen this myself, but somehow I didn't, so thanks a lot for pointing it out. I will follow #220 as well since it looks interesting and goes beyond my immediate need asked in this request.

closed 2021-12-20 by TorbjornHouge