wireviz/WireViz #265: [bug] TypeError: 'in <string>' requires string as left operand, not int (connect in Harness.py)

scandey, 2022-01-10:
Hello! I discovered WireViz today while looking for options to improve on my current harnessing system of Excel spreadsheets.
I've been attempting to use it, but I keep coming across exceptions without any clear way to debug them.

The issue I keep hitting my head against at the moment is a TypeError: 
```
Traceback (most recent call last):
  File "/usr/local/bin/wireviz", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/site-packages/wireviz/wireviz.py", line 266, in main
    parse(yaml_input, file_out=file_out)
  File "/usr/local/lib/python3.9/site-packages/wireviz/wireviz.py", line 188, in parse
    harness.connect(from_name, from_pin, via_name, via_pin, to_name, to_pin)
  File "/usr/local/lib/python3.9/site-packages/wireviz/Harness.py", line 72, in connect
    if via_wire in cable.colors and via_wire in cable.wirelabels:
TypeError: 'in <string>' requires string as left operand, not int
```

I've included my .yml file as a .txt so that Github would let me upload it. I'm sure I'm doing something silly, but for the life of me I can't figure it out. 
[dsub-airborn.txt](https://github.com/formatc1702/WireViz/files/7841519/dsub-airborn.txt)

Versions:
wireviz 0.3.2
dot - graphviz version 2.50.0 (20211204.2007)
Python 3.9.9
Mac OS 12.1

scandey, 2022-01-10:
Ah! I just figured it out: I used colors instead of color_code for a cable instance. 

I've edited the title to change from bug report to feature request: this seems like the kind of syntax mistake that should throw an error with line number in the YML source, rather than falling through to a TypeError.

kvid, 2022-02-12:
This is related to the general error message improvements discussed in #207.