wireviz/WireViz #267: BOM From Multiple Cables

leoheck, 2022-01-15:
I have a project where I have some cables and wires described with wireviz.

I am looking for a way to have a unified BOM that I can use to buy parts.

For instance
- cable_a.yml
- cable_b.yml
- cable_c.yml

Cables A and B use the same wires to make the cables.
B and C have a connector in common.

I would like to have BOM that has all these on the same table so the one buying parts could buy all togueter.
It could be generated like this (if this feature does not exist yet)

```
wireviz --bom cable_a.yml cable_b.yml cable_c.yml
```

Do we have such a feature?

kvid, 2022-01-28:
Thank you for this new use case. There is no current feature supporting this, but I have created a simple Python script `tsv2yaml.py` that might help you:
```py
# Convert a set of WireViz BOM TSV output files into WireViz YAML input for a total BOM
# Created by @kvid for issue #267
from pathlib import Path
from sys import argv

print('additional_bom_items:')
for a in argv[1:]:
  prefix = Path(a).with_suffix('').stem.replace(' ', '_')
  print(f'  # From {a}')
  with Path(a).open() as tsv:
    keys = [k.strip().lower().replace('/', '') for k in next(tsv).split('\t')]
    for row in tsv:
      print('  -')
      for k, v in zip(keys, row.split('\t')):
        if k != 'id':
          v = v.strip()
          if k == 'designators':
            v = ', '.join([f'{prefix}:{d}' for d in v.split(', ') if d != ''])
          print(f'    {k}: {v}')
```
Execute the commands e.g. like this on the WireViz tutorials:
```sh
python3 tsv2yaml.py ../../tutorial/*.tsv > _test/tutorials.yaml
python3 wireviz.py _test/tutorials.yaml
```
I got this result from the commands above:
[image]
Note that I have not verified the correctness of this total BOM. Please try it out on your use case, and tell me if this might do the job good enough.

kvid, 2022-01-29:
A minor variation of my Python script above that format the designator prefix differently, and also include the prefix as designator in entries without any other designator:
```py
#!/usr/bin/env python3
# -*- coding: utf-8 -*-
# Convert a set of WireViz BOM TSV output files into WireViz YAML input for a total BOM
# Created by @kvid for issue #267
from pathlib import Path
from sys import argv

print('additional_bom_items:')
for a in argv[1:]:
  prefix = Path(a).with_suffix('').stem.replace(' ', '_')
  print(f'  # From {a}')
  with Path(a).open() as tsv:
    keys = [k.strip().lower().replace('/', '') for k in next(tsv).split('\t')]
    for row in tsv:
      print('  -')
      for k, v in zip(keys, row.split('\t')):
        if k != 'id':
          v = v.strip()
          if k == 'designators':
            v = f'({prefix})' if v == '' else \
              ', '.join([f'({prefix}){d}' for d in v.split(', ') if d != ''])
          print(f'    {k}: {v}')
```
Excerpt of the result when running the same commands as in my previous message:
[image]
Please comment the prefix+designator formatting. What is more readable?

leoheck, 2022-01-29:
This is awesome. I don't have a computer with me for some days. But I will definitely test thus asap. Thanks for this.