wireviz/WireViz #269: Add option for reversing connector output

martonmiklos, 2022-01-21:
Hi folks,

I have made a cable recently with WireViz, which looks the following:
[image]

To ease the understanding when manufacturing I would have rather drawn it the following way (with strait cable):
[image]
but I have not found a way of reversing the connector pins.

I was thinking about adding a simple reverse_pins optional boolean attribute under the connector to being able to achieve this.
If you think it would be useful I can sort out the implementation in a PR.

kvid, 2022-01-21:
Thank you for sharing your use case. With the current WireViz version (v0.3.2), this is one way to obtain the output I guess you might want:
```yaml
connectors:
  P1:
    pincount: 10
  P2:
    pins: [10,9,8,7,6,5,4,3,2,1]
cables:
  W1:
    wirecount: 10
    color_code: DIN
connections:
  - - P1: [1-10]
    - W1: [1-10]
    - P2: [10-1]
```
[image]
When describing a use case like this, please include your input YAML (if possible) and version numbers as described in [CONTRIBUTING.md](https://github.com/formatc1702/WireViz/blob/master/docs/CONTRIBUTING.md) because that often helps to reproduce your use case much quicker.

martonmiklos, 2022-01-21:
> When describing a use case like this, please include your input YAML (if possible) and version numbers as described in CONTRIBUTING.md because that often helps to reproduce your use case much quicker.

Mea culpa!

I had:
```
connectors:
  P1_VEZERLO:
    type: '15EDGK3.81-10P vagy kompatibilis'
    pinlabels: [OSZLOP1,OSZLOP2,OSZLOP3,OSZLOP4,OSZLOP5,OSZLOP6,OSZLOP7,OSZLOP8,OSZLOP9,OSZLOP10]
    image: 
        src: '10p_degson.png'
  P2_ELOSZTO:
    type: '15EDGK3.81-10P vagy kompatibilis'
    pinlabels: [OSZLOP10,OSZLOP9,OSZLOP8,OSZLOP7,OSZLOP6,OSZLOP5,OSZLOP4,OSZLOP3,OSZLOP2,OSZLOP1]
    image: 
        src: '10p_degson_reverse.png'
      
cables:
  W1:
    color_code: DIN
    shield: false
    wirecount: 10
        
connections:
  -
    - P1_VEZERLO: [1,2,3,4,5,6,7,8,9,10]
    - W1: [1,2,3,4,5,6,7,8,9,10]
    - P2_ELOSZTO: [10,9,8,7,6,5,4,3,2,1]
```

closed 2022-01-21 by martonmiklos

kvid, 2022-01-21:
> Mea culpa!

No bad feelings, I hope? As you have contributed before, my comment was only meant as a friendly reminder about factors that can help us resolving issues.

> 
> Many thanks for the hint, it solved my issue, my drawing now looks clean and straightforward

I'm glad this solved your issue. Don't hesitate raising more issues when you come across new use cases with e.g. unexpected output or missing features.

martonmiklos, 2022-01-22:
> No bad feelings, I hope?
Absolutely not :)

> I'm glad this solved your issue. Don't hesitate raising more issues when you come across new use cases with e.g. unexpected output or missing features.

Sure, I was thinking about adding this or something similar as an example just to make it easier for the following users.