wireviz/WireViz #272: [feature] pincolors specification for connector pins does not accept striped/banded combo code

Halfwalker, 2022-02-06:
Looks like specifying pincolors in a connector will not accept (silently ignores) combination color codes for striped/banded colors.
Using the following:
```
connectors:
  Elite-A36:
    type: Haltech 34
    pins:
      - A01
      - A02
      - A03
      - A04
      - A05
      - A25

  EA1:
    pins:
      - EA1-1
      - EA1-2
      - EA1-3
      - EA1-4
      - EA1-5
      - EA1-6
      - EA1-7
      - EA1-8
    pincolors: [BU, WHGN, BK, BKYE, IV, BKRD, WH, GY]

  EA2:
    pins:
      - EA2-1
      - EA2-2
      - EA2-3
      - EA2-4
      - EA2-5
    pincolors: [BK, BKRD, IV, WH, BKRD]

cables:
  EA1-Elite:
    category: bundle
    gauge: 20 AWG
    length: 0.2
    wirecount: 2
    colors: [GN, RD]

connections:
  -
    - Elite-A36: [A01, A25]
    - EA1-Elite: [1-2]
    - EA1: [EA1-2, EA1-8]
```
results in this
[image]

Everything looks fine except the combo colors for pins are missing.  Given that it's just a small square of color, this is likely just the way it is.  Any chance of expanding the color square to a small rectangle to incorporate striped wires ?  For example, **RDYE** could produce 
[image]
This would be extremely helpful when building a harness to connect into existing connectors.  You tag the existing connectors with the wire colors they have, and that provides a visual check when building the harness.

kvid, 2022-02-12:
You are right, it's currently designed for single colors only, and I admit this should have been stated more clearly in the [documentation](https://github.com/formatc1702/WireViz/blob/master/docs/syntax.md#connector-attributes). It probably is possible to expand this feature to support color combinations like for wire colors, but in my quick and dirty trials today I got some ugly results, so more work is needed to find a simple way to produce a readable result within the limitations of [Graphviz HTML](https://graphviz.org/doc/info/shapes.html#html).

If your motivation for suggesting this is to enable pin color marks that matches wire colors, then I would guess horizontal color stripes in a rectangle are wanted because that is the only multi-colored wires currently supported. The rectangle you suggest looks more like a wire with color bands that might be supported in the future, but #120 seems not close to finished yet. However, there might be other considerations to take to obtain readable pin color marks. What are your thoughts?

Halfwalker, 2022-02-12:
That sample rectangle was just a crop from a commercial wiring diagram (Haltech).  It doesn't need to be any specific style, just indicate that the wire has two colors in it.  Though it should be clear and easy to see the two colors - sometimes the horizontal color stripes are hard to determine.

For example, B-R Black-Red is easy to see here
[image]

P-G Pink-Green is a little harder to make out here
[image]

Perhaps a simple rectangle with two color blocks in it - by definition 1st is the main color and 2nd is the stripe color.