wireviz/WireViz #274: [feature] Allow include files in yml

Halfwalker, 2022-02-09:
I'm playing with WireViz to see if I can use it to document an automotive wiring harness.  This is a fairly complex beast, with several large-pin-count connectors - 26, 34, 38, 20, etc. plus a ton of smaller ones.  I think trying to do them all, the whole harness, in a single image will just get too unwieldy.  So perhaps doing sub-portions of the harness ?

For example, the in-cabin portion of the harness will have ECU_A (24pins), ECU_B (36 pins), II1 (38 pins), IJ1 (20 pins) connectors.  There are about 20 wires that go between these 4, so do a run that just diagrams those connections and associated pins, hiding unused pins.  Another would be between II1, IJ1, EA1, EA2, so another run for just those connections and the associated pins.  I can see doing 5 or 6 sub-portions like that.

The problem is that each yml file will need to contain the same connectors, and keeping them in sync is an easily overlooked place for errors.  It would be nice to define the connectors each in their own file, then include them in the `connectors:` section as needed for each sub-portion yml file.

```
connectors:
  ECU_A:
    src: ECU_A.yml
  ECU_B:
    src: ECU_B.yml
  :
```
I know we can simulate this using a Makefile to assemble the sub-portion run yml files, but it would be cleaner this way I think.

kvid, 2022-02-09:
This seems related to #220. Please try including your library of connector and cable templates as a YAML input file using the existing [command line option](https://github.com/formatc1702/WireViz/tree/master/docs#command-line-options) `--prepend-file`. See also [syntax](https://github.com/formatc1702/WireViz/blob/master/docs/syntax.md#inheritance) and [tutorial](https://github.com/formatc1702/WireViz/tree/master/tutorial#07---daisy-chaining-ii) for using such templates. Does that solve your issue? If not, please describe use cases where more or different features are required, but please consider doing that by joining the discussion at #220 unless you feel that issue differs too much from your issue.

closed 2022-02-09 by Halfwalker

snelweg, 2025-03-11:
For other people who also had trouble getting the prepend thing to work, here is how I do it.

> One .yml file with all my connectors and cables
> Another .yml file with just connections and meta data

`wireviz --prepend  connectors.yml 7i75.yml `
Will result in a 7i75.html file with all the connections defined in the 7i75.yml file using all the connectors from the connectors.yml file.

So keep the connectors and the cables in one file, then you can use another file to make specific connections for your application.

It would have been nice to be able to merge numerous files, that way the files do not turn out too big and messy.
Would also have been nice if we could alternate between connectors and cables and back, that way there is still some logic in the cinnectors file as to what belongs to what.