wireviz/WireViz #276: Specify use of pre-built assemblies?

jamesfowkes, 2022-02-15:
I'm specifying a cable bundle that will make use of a pre-built [DC cable assembly](https://www.digikey.co.uk/en/products/detail/tensility-international-corp/CA-2189/568580) (a 2-core cable terminated at one end with a moulded DC jack).

What is the best way of specifying this kind of assembly? I can obviously model it as a cable+connector, but then it looks like I'm specifying that they are separate parts rather than a complete assembly.

I can add notes to the cable+connector to make it clearer, but I'm wondering if there's a more elegant way.

kvid, 2022-02-16:
I don't know any more elegant way to model your use case, but I suggest ignoring the jack in the BOM like this to avoid separate BOM entries for the assembly parts:
```yaml
connectors:
  Jack:
    pins: [Center, Sleeve]
    color: BK
    ignore_in_bom: true
cables:
  "DC cable assembly":
    colors: [BKWH, BK]
    gauge: 24 AWG
    length: 1.83 m
    supplier: Digi-Key
    spn: CP-2189-ND
    manufacturer: Tensility International Corp
    mpn: CA-2189
connections:
  - - Jack: [Center, Sleeve]
    - "DC cable assembly": [BKWH, BK]```

kvid, 2022-02-16:
The disadvantage with using the cable as the main part, as I suggested above, is that several instances of it will be shown in the BOM as a common entry with the sum of their lengths as quantity, and that doesn't make sense with such an assembly that has a fixed length. It probably is better to use the connector as the main part, because then several instances of it will be shown in the BOM as a common entry with the total count as quantity.

It is also possible to group them together visually using [the experimental feature `tweak`](https://github.com/formatc1702/WireViz/blob/master/docs/syntax.md#graphviz-tweaking-experimental):
```yaml
connectors:
  A1:
    type: DC jack 5.5 x 2.1 x 12 mm
    subtype: cable assembly
    color: BK
    pins: [Center, Sleeve]
    pinlabels: [+12 V, Ground]
    manufacturer: Tensility International Corp
    mpn: CA-2189
    supplier: Digi-Key
    spn: CP-2189-ND
  X1:
    type: Your circuit
    pincount: 2
cables:
  A1 cable:
    type: DC assembly
    gauge: 24 AWG
    length: 1.83 m
    colors: [BKWH, BK]
    notes: Fixed part of A1
    ignore_in_bom: true
connections:
  - - A1: [Center, Sleeve]
    - A1 cable: [BKWH, BK]
    - X1: [1-2]
tweak:
  append: |
    subgraph cluster1 {
      label="DC cable assembly"
      style=dashed
      A1
      "A1 cable"
    }
```
[image]

jamesfowkes, 2022-02-21:
Good advice, thanks!

closed 2022-02-21 by jamesfowkes