wireviz/WireViz #277: Blocks are squeezed together

AshleyAtVivalyte, 2022-02-16:
[image]
Hi there,

I seem to have an issue (even with demo code) gives this squeezed image. Is this a setting from Graphviz that's being reflected here. Or am I just looking over something obvious that i'm missing?  

I tried some of the overrides I found in here [Issue 174](https://github.com/formatc1702/WireViz/issues/174) But that didn't change much.

I hope

`
connectors:
  X1:
    type: D-Sub
    subtype: male
    pincount: 25
    pins: [1,14,3,16,5,18,7,20,9,22,11,24,13]
    pinlabels: [ SENSE_P_1, SENSE_N_1, SENSE_P_2, SENSE_N_2, SENSE_P_3, SENSE_N_3, SENSE_P_4,SENSE_N_4, SENSE_P_5, SENSE_N_5, SENSE_P_6, SENSE_N_6, GND ]
  X2:
    type: F48
    subtype: female
    pincount: 48
    pins: [ z2,b2,d2,z4,b4,d4,z6,b6,d6,z8,b8,d8,z10,b10,d10,z12,b12,d12,z14,b14,d14,z16,b16,d16,z18,b18,d18,z20,b20,d20,z22,b22,d22,z24,b24,d24,z26,b26,d26,z28,b28,d28,z30,b30,d30,z32,b32,d32 ]

cables:
  W1:
    gauge: 0.25 mm2
    length: 0.2
    color_code: DIN
    wirecount: 12
    shield: true

connections:
  -
    - X1: [1,14,3,16,5,18,7,20,9,22,11,24]
    - W1: [2,1,4,3,6,5,8,7,10,9,12,11]
    - X2: [d4,z2,d10,z8,d16,z14,d20,z18,d26,z24,d32,z30]
  -
    - X1: 13
    - W1: s
`

kvid, 2022-02-16:
This seems to be related to #258. Please inform us about your version numbers, as described in [CONTRIBUTING.md](https://github.com/formatc1702/WireViz/blob/master/docs/CONTRIBUTING.md).

AshleyAtVivalyte, 2022-02-16:
@kvid Thank you! modifying the file like described fixed the issue.

closed 2022-02-16 by AshleyAtVivalyte

kvid, 2022-02-16:
Please inform us about your version numbers, as described in [CONTRIBUTING.md](https://github.com/formatc1702/WireViz/blob/master/docs/CONTRIBUTING.md#contribution-guidelines) so others might learn from your experience, and all of us will benefit from a reduced number of repeated issues!!!

I suspect you don't use the latest version because PR #261 that fixes #258 is both merged and released.