wireviz/WireViz #280: Displaying row as column

andreastoellefsen, 2022-04-19:
I'm experiencing som strange behavior when using the feature/technical-drw branch.
WireViz seems to displayed the row as a column. 
[image]
 
When switching to the master branch everything is fine:
[image]

The only thing I'm changing is the branch

Got any ideas?

I really appreciate the effort you have put into this tool!

Andreas

andreastoellefsen, 2022-04-19:
I should have noted that I'm using the examples/demo02.yml as a test

kvid, 2022-04-19:
This seems to be a duplicate of #258 that was fixed by PR #261 and then merged into master from [v0.3.2](https://github.com/formatc1702/WireViz/releases/tag/v0.3.2), but not currently included in `dev`, `latest`, or any of the PR branches. If you need this fix together with `feature/technical-drw`, then I suggest you create your own local branch from this PR branch and then either merge master into it, or cherry-pick only the fix 9af0cb8ab0314bc6754b153769f7562b5b38ec10.

andreastoellefsen, 2022-04-19:
Thanks, that did the trick!

Andreas

closed 2022-08-05 by formatc1702