wireviz/WireViz #281: how to add image

picli3, 2022-04-30:
I have been trying to add an image of the connector, but I have not been successful nor do I see anything related in the tutorial.

[image]

I'm putting it as HTML, I don't know.
[image]
I discovered this tool recently, and I find it very good.

kvid, 2022-04-30:
Thank you for reporting this strange error message. I admit it doesn't tell you much about what really is the problem. We are discussing how to improve error messages in #207.

See [example-08](https://github.com/formatc1702/WireViz/tree/master/examples#example-08) and the [syntax doc](https://github.com/formatc1702/WireViz/blob/master/docs/syntax.md#images) for proper input syntax.

picli3, 2022-04-30:
Thank you very much for answering, I'm doing it this way and it still doesn't work.
[image]
i get the following error
[image]

the image is in the same file path, use an absolute path and a relative path and the result is the same.

kvid, 2022-04-30:
To be able to reproduce your use case, please provide your version numbers and full input YAML as described in the [guidelines](https://github.com/formatc1702/WireViz/blob/master/docs/CONTRIBUTING.md). The error might be triggered by something in your input YAML that is not shown in your excerpt above. Try also the original example-08 as input to see if that also triggers the same error. Both absolute and relative paths should work.

picli3, 2022-05-01:
It already worked, I had a previous syntax problem.

thanks for your help.

<pre>
  X2:
    type: DT04-3P
    subtype: male
    pinlabels: [BAT+, BAT-, IGN]
    image:
      src: "/home/maykol/Documentos/cables/TPS/wifi.png"        # path to the image file
      # optional parameters:
      #caption: <str>     # text to display below the image
      #bgcolor: <color>   # Background color of entry in diagram component box
      width: 100      # Expand image cell to this minimum height
      #fixedsize: false # Avoid scaling to a fixed size
</pre>

[image]

kvid, 2023-04-10:
It seems this issue has been solved, and #292 might help to avoid similar mistakes in the future. Feel free to re-open this issue again @picli3 if you don't agree.

closed 2023-04-10 by kvid