wireviz/WireViz #282: [bug] mm² to AWG tables are inaccurate

brianjk66, 2022-05-12:
After designing a harness with WireViz and setting `show_equiv` to `true` for my wires, I noticed the conversion from 20 AWG to mm² was not correct. 20 AWG should be 0.5mm², but WireViz converted it to 0.75mm².

I took a look at `wv_helper.py` and noticed that awg_equiv_table seems to have incorrect values for 21AWG, 20AWG, & 18AWG. That's according to this table I found after a quick Google search: https://lexproducts.com/technical/wire-conversion. Other tables I looked at seem to agree as well though.

kvid, 2022-05-15:
You are right. Thank you for reporting this bug that has been present since [v0.1](https://github.com/formatc1702/WireViz/releases/tag/v0.1). Based on the definitions described at [Wikipedia](https://en.wikipedia.org/wiki/American_wire_gauge), this code prints correct conversions with 3 digits precision:

```py
  from math import pi
  for a in range(-3, 41):
    print(f'{a if a > 0 else str(1-a)+"/0"} AWG = {pi * (0.005 * 25.4/2)**2 * 92**((36-a)/19.5):.3g} mm²')
```
The code above prints this output that shows deviations from all values in the current table (some more than others, though):
```
4/0 AWG = 107 mm²
3/0 AWG = 85 mm²
2/0 AWG = 67.4 mm²
1/0 AWG = 53.5 mm²
1 AWG = 42.4 mm²
2 AWG = 33.6 mm²
3 AWG = 26.7 mm²
4 AWG = 21.2 mm²
5 AWG = 16.8 mm²
6 AWG = 13.3 mm²
7 AWG = 10.5 mm²
8 AWG = 8.37 mm²
9 AWG = 6.63 mm²
10 AWG = 5.26 mm²
11 AWG = 4.17 mm²
12 AWG = 3.31 mm²
13 AWG = 2.62 mm²
14 AWG = 2.08 mm²
15 AWG = 1.65 mm²
16 AWG = 1.31 mm²
17 AWG = 1.04 mm²
18 AWG = 0.823 mm²
19 AWG = 0.653 mm²
20 AWG = 0.518 mm²
21 AWG = 0.41 mm²
22 AWG = 0.326 mm²
23 AWG = 0.258 mm²
24 AWG = 0.205 mm²
25 AWG = 0.162 mm²
26 AWG = 0.129 mm²
27 AWG = 0.102 mm²
28 AWG = 0.081 mm²
29 AWG = 0.0642 mm²
30 AWG = 0.0509 mm²
31 AWG = 0.0404 mm²
32 AWG = 0.032 mm²
33 AWG = 0.0254 mm²
34 AWG = 0.0201 mm²
35 AWG = 0.016 mm²
36 AWG = 0.0127 mm²
37 AWG = 0.01 mm²
38 AWG = 0.00797 mm²
39 AWG = 0.00632 mm²
40 AWG = 0.00501 mm²
```
A table with these numbers will only help converting from AWG to mm² because the mm² values don't match common wire dimensions exactly. There are several options when converting from mm² to AWG:
- Compute a decimal AWG value by rounding to 3 digits precision, but decimal AWG numbers are highly unusual and might confuse the user - especially for AWG values < 1.
- Show the closest integer AWG value above or below. This might be unsafe because if a user actually use an AWG wire that is thinner than specified by mm², then the wire might get too hot.
- Show the closest integer AWG value below or equal (i.e. not thinner).

thestumbler, 2023-07-03:
I just ran into this today with some wire which appears to be uniquely specified in mm2 and there is no exact AWG equivalent. It is 0.12 mm2, closest to 26 AWG. I like using the formula instead of a table, and would add my preference to round to the nearest standard AWG vs always going up to the next largest size. I understand the issue, and that makes sense in many applications, but when making small cables, sometimes such rounding up could kick you out of the range of the crimp connectors, for example.

I would also like to see the equivalent value so-indicated, perhaps by parentheses. In my example, it would be `0.12 mm2 (26 AWG)`, or if you want to indicated a poor rounding match, something like  `0.12 mm2 (~26 AWG)`

kvid, 2023-07-04:
When such a preference might differ between applications, maybe we should add some options to let the user decide, e.g. something like this:
```yaml
options:
  gauge_equiv:
    awg: nearest/safe
    mm2_digits: 3
    poor_match: (some tolerance limit for showing a leading '~')
```
Maybe the `show_equiv` attribute of cables optionally also could contain similar attributes instead of just true/false to .

thestumbler, 2023-07-04:
I thought of a parallel example with machine screws. If I have some mounting holes to attach feet to a prototype board, people can safely use either M3 or #4-40 screws as if they're equivalent. But if someone were designing a back porch or a pedestrian foot bridge, the screw sizes would be absolutely critical (not to say there couldn't be equivalents, but each option would have to be checked by the mechanical engineer). 

I think this issue of equivalent wire gauges is quite similar. Heck, if Wikipedia can be believed, there are some "standard" metric wire sizes for which there is no AWG equivalent (interesting that my 0.12 mm2 wire doesn't seem to exactly belong in either category).

thestumbler, 2023-07-10:
Oh. I should've been clearer. I do like your proposed solution very much.

I had suddenly thought of the machine screw analogy shortly after sending my original comment, and posted it as a second comment in case it helped anyone understand the logic behind this. 

In my own recent design searches, I think there's a standard or convention for purely metric series of wire sizes -- wire whose diameters / cross sectional areas aren't intentionally aligned with the AWG series, but are themselves "normal" metric values. Kind of like a 2 mm pitch connector vs a 2.54 mm connector. The latter is metric, but clearly it's just the equivalent of part designed in inches. The former is a part made originally using the metric system. 

I mention this because it would be nice to include such a table in the program alongside the list of AWG sizes.

kvid, 2023-07-12:
> Oh. I should've been clearer. I do like your proposed solution very much.
> 

Thank's, but please also add variations that you find useful.

> I had suddenly thought of the machine screw analogy shortly after sending my original comment, and posted it as a second comment in case it helped anyone understand the logic behind this.
> 

Yes, I now understand it was not meant as an argument after reading my suggestion.

> In my own recent design searches, I think there's a standard or convention for purely metric series of wire sizes -- wire whose diameters / cross sectional areas aren't intentionally aligned with [...]

[IEC 60228](https://www.vde-verlag.de/iec-standards/211610/iec-60228-2004.html)  specifies the nominal cross-sectional areas, in the range 0.5 mm² to 2500 mm², for conductors in electric power cables, but the standard does not apply to conductors for telecommunication purposes.

> I mention this because it would be nice to include such a table in the program alongside the list of AWG sizes.

In what way would you expect such a list to be used? Maybe an optional conversion preference to select the nearest equivalent in such a table, but then we ideally should also cover small dimensions, and I don't know a standard for that. Maybe something like the list attribute `mm2_available` that I suggested above is the best we can do if the actual list of dimensions depends heavily on your cable suppliers.

thestumbler, 2023-07-12:
> In what way would you expect such a list to be used? 

Maybe "list" is the wrong term. I meant that the program should be aware of these wire sizes, too, in as much as it needs to be aware of any standard wire gauge information. Today that takes the form of a list. And upon further thought, I think that list is only used for conversion between AWG and metric. I assume any revised algorithm as outlined above would probably use a similar table to "know" the standard sizes. I'm just guessing that a purely mathematical approach to deriving the standard sizes is probably more hassle than using a tabulated approach (just think about the headaches of using equations to generate the E96 number series, for example). The total number of different (and commonly used in modern designs) standard wire sizes seems manageable and shouldn't be so large as to exclude the tabulated data approach, IMHO. 

I'll dig around and see if I can figure out where those "pure metric" wire sizes come from that I found on Wikipedia last week -- if they are a standard or just made up for illustrative purposes on that chart. My first reaction to the notion of digging through wire manufacturers' data books is "oh no, that doesn't seem like a good idea". But on further thought, if indeed there is no standard but it's obvious that a major wire companies are using a single de-facto "standard" set of sizes, then it's not a bad idea. But if there are like four different private sets of wire sizes, I think that would become unwieldy. 

Wild thought, how difficult would it be for an (advanced?) user to provide their own table of wire information to the program's algorithm? I'm not sure of the usefulness of such a facility, but if it turns out there are really a bunch of conflicting manufacturers' private wire standards when it comes to pure metric sizes in the small signal low voltage space, it might make sense. 

As to your request for additional parameters to the prosed new way of specifying wire size, I'm not thinking of any at present. However, a kind of related size topic would be the wire's insulation size. I suspect that's more unclear than wire gauge sizes, but if there are some reasonable standards it might be helpful to include those in a manner similar to gauge? I'm just thinking out loud here, having recently worked on a design where the insulation diameter was also important for PCB strain relief purposes.

kvid, 2023-07-13:
Variation 1:
```yaml
options:
  gauge_equiv:
    show: true/false  # Default 'show_equiv' value for cables in this harness
    awg:
      rounding: nearest  # Or other rounding strategies, e.g. named 'thicker', 'thinner', ...
      available: [13, 15, 17, 19, 21, 23]  # Subset to use
    kcmil:  # Unit to use when thicker than 0000 AWG
      rounding: nearest  # Or other rounding strategies, e.g. named 'thicker', 'thinner', ...
      available: [250, 300, 350, 400, 500]  # Subset to use
    mm2:
      digits: 3  # Compute 3-digit equivalent
      available: [0.5, 0.75, 1, 1.5, 2.5, 4, 6, 10]  # Subset to use
      rounding: nearest  # Or other rounding strategies, e.g. named 'thicker', 'thinner', ...
    match:
      rounding: nearest  # Common strategy unless specified for each unit
      poor: 10 %  # Show a leading '~' unless rounding within 10 %
      invalid: 50 %  # Don't show equivalent unless rounding within 50 %
```
- Specifying both `digits` and `available` for `mm2` should probably be illegal.
- Perhaps is a common rounding strategy always enough,  and we can skip supporting rounding for each unit. If so, then the remaining attribute value for each unit could be specifyed for the unit directly to simplify the syntax.

kvid, 2023-07-13:
> Maybe "list" is the wrong term. I meant that the program should be aware of these wire sizes, too, in as much as it needs to be aware of any standard wire gauge information. Today that takes the [...]

A full list of standard sizes could be the default values of the `available` list attributes in my variation 1 and 2 above.

> I'll dig around and see if I can figure out where those "pure metric" wire sizes come from that I found on Wikipedia last week -- if they are a standard or just made up for illustrative purposes on [...]

Yes, references to relevant standards (or de-facto "standards") are welcome.

> Wild thought, how difficult would it be for an (advanced?) user to provide their own table of wire information to the program's algorithm? I'm not sure of the usefulness of such a facility, but if [...]

Se my variations 1 and 2 above.
 
> As to your request for additional parameters to the prosed new way of specifying wire size, I'm not thinking of any at present. However, a kind of related size topic would be the wire's insulation [...]

This last suggestion about extra parameters might be resolved by https://github.com/wireviz/WireViz/issues/222#issuecomment-803313632

thestumbler, 2023-07-13:
I'd say variation 2 is best, but then again I'm not a power user.

kvid, 2023-07-14:
> I'd say variation 2 is best, but then again I'm not a power user.

I agree it looks better than variation 1, but don't stop looking for even better variations.

> 
> Not sure of the usefulness of specifying the closest match in such a fine-grained approach.

Is it the `rounding` attribute you find less useful? My idea of different rounding values are:
- `nearest` to round up or down to the nearest value.
- `thicker` or `safe` to always round towards a thicker gauge value for safety.
- `thinner` to always round towards a thinner gauge value to stay within size restrictions of connector terminals or ferrules.

> 
> These proposed yaml directives are on a "per-invocation of the program" basis? Or could they be overridden on a per-cable basis? If the latter, that seems like it would take care of any situation where the user wanted to tweak the nearest-match tolerance.

I suggest accepting the `gauge_equiv` dict attribute in the `options` section - i.e. "per-invocation of the program", but I also mentioned early that the `show_equiv` attribute for each cable could optinally (as an alternative to the simple `true`/`false` value) accept the same dict entries to override the common options when needed.

thestumbler, 2023-07-14:
> > Not sure of the usefulness of specifying the closest match in such a fine-grained approach.
> Is it the rounding attribute you find less useful? My idea of different rounding values are:

By fine-grained, I meant specifying rounding separately for each kind of wire gauge standard, not the kinds of rounding options. I think those are good.

kvid, 2023-07-14:
> By fine-grained, I meant specifying rounding separately for each kind of wire gauge standard, not the kinds of rounding options. I think those are good.

I agree - that was also my motivation for variation 2. Originally, I was thinking separate values for each unit to allow a computed value for mm² and rounding to list entries for AWG, but then I realized only one value is needed for each unit - either the number of digits in a computed value, or a list of available values for matching.