wireviz/WireViz #283: Software not working

AstroKAL, 2022-06-26:
Hello

I downloaded and installed all the required files for the software, but it doesn't work!

Can someone please assist me in making the software work for me?

Thanks

kvid, 2022-07-02:
Thamk you for reporting your problem. However, more information is needed before anybody can help you. Please describe or refer to the (installation and usage procedures)[https://github.com/formatc1702/WireViz#usage] 
 you have tried. Please also describe your environment and version numbers as described (here)[https://github.com/formatc1702/WireViz/blob/master/docs/CONTRIBUTING.md#submitting-a-new-issue].

kvid, 2023-04-10:
This issue doesn't describe any information about what has been done, nor any error message/unexpected result. I therefore close this issue as unsolvable due to lack of information. Feel free to re-open it again @AstroKAL if you can provide the information we need to help you.

closed 2023-04-10 by kvid