wireviz/WireViz #284: [bug] Image src path is relative to output location

matthewspydell, 2022-06-29:
wireviz version 0.3.2
python version 3.9.13

I discovered that the image path placed in the YAML file, e.g.
```yaml
image:
  src: resources/connector.png
```

is relative to the output directory because of this line:https://github.com/formatc1702/WireViz/blob/b0d0070f08247e132b11bf45e617da9e8e1881f1/src/wireviz/wireviz.py#L62

As you can see the `fileout` variable that is created based on the `output` argument is used to prepend to the image src parameter. This means if you provide an output location the image path in the YAML file needs to point to the image as if it is in the output directory.

I believe this is a bug and the fix is to instead pass the `filein` variable to create the context for finding the image since it makes logical sense that you will put the path to an image relative to where the YAML file lives.

For context I have a repo containing multiple subdirectories all with various YAML files and a single `output/` directory that mirrors the source directory hierarchy but only contains the files generated by wireviz. To use the images I would have to add a relative path to the yaml file that would make no sense from a human-readable standpoint.

```
.
├── README.md
├── connector_pinout_images
│   └── scsi68_pinout.png
├── generate.sh
├── mhr
│   ├── acropack_bob
│   │   ├── acropack_in.yaml -> ../tx_asb_v1/acropack_in.yaml
│   │   └── hd15_txch01.yaml -> ../tgen_v1/hd15_txch01.yaml
│   ├── if_agc
│   │   ├── j10.yaml
│   │   ├── j11.yaml
│   │   └── j12.yaml
│   ├── tgen_v1
│   │   ├── hd15_rxch01.yaml
│   │   ├── hd15_txch01.yaml
│   │   ├── scsi_gpio.yaml
│   │   └── scsi_timing.yaml
│   └── tx_asb_v1
│       └── acropack_in.yaml
├── output
│   └── mhr
│       ├── acropack_bob
│       │   ├── acropack_in.bom.tsv
│       │   ├── acropack_in.gv
│       │   ├── acropack_in.html
│       │   ├── acropack_in.png
│       │   ├── acropack_in.svg
│       │   ├── hd15_txch01.bom.tsv
│       │   ├── hd15_txch01.gv
│       │   ├── hd15_txch01.html
│       │   ├── hd15_txch01.png
│       │   └── hd15_txch01.svg
│       ├── if_agc
│       │   ├── j10.bom.tsv
│       │   ├── j10.gv
│       │   ├── j10.html
│       │   ├── j10.png
│       │   ├── j10.svg
│       │   ├── j11.bom.tsv
│       │   ├── j11.gv
│       │   ├── j11.html
│       │   ├── j11.png
│       │   ├── j11.svg
│       │   ├── j12.bom.tsv
│       │   ├── j12.gv
│       │   ├── j12.html
│       │   ├── j12.png
│       │   └── j12.svg
│       ├── tgen_v1
│       │   ├── hd15_rxch01.bom.tsv
│       │   ├── hd15_rxch01.gv
│       │   ├── hd15_rxch01.html
│       │   ├── hd15_rxch01.png
│       │   ├── hd15_rxch01.svg
│       │   ├── hd15_txch01.bom.tsv
│       │   ├── hd15_txch01.gv
│       │   ├── hd15_txch01.html
│       │   ├── hd15_txch01.png
│       │   ├── hd15_txch01.svg
│       │   ├── scsi_gpio.bom.tsv
│       │   ├── scsi_gpio.gv
│       │   ├── scsi_gpio.html
│       │   ├── scsi_gpio.png
│       │   ├── scsi_gpio.svg
│       │   ├── scsi_timing.bom.tsv
│       │   ├── scsi_timing.gv
│       │   ├── scsi_timing.html
│       │   ├── scsi_timing.png
│       │   └── scsi_timing.svg
│       └── tx_asb_v1
│           ├── acropack_in.bom.tsv
│           ├── acropack_in.gv
│           ├── acropack_in.html
│           ├── acropack_in.png
│           └── acropack_in.svg
└── requirements.txt
```

kvid, 2022-07-02:
Thank you for the detailed bug report. It seems to be very similar to #200. Take a look at the suggestion for that issue.

matthewspydell, 2022-07-06:
The better issue to look at is #189 as it seems to be the first issue addressing the image path problem. That issue notes that the branch [bugfix/image-src](https://github.com/formatc1702/WireViz/tree/bugfix/image-src) contains the fix to this problem as well as additional features (notably that it embeds images within the SVG and HTML output files).

Hopefully this branch gets merged at some point and makes it into a release version.

closed 2022-07-06 by matthewspydell