wireviz/WireViz #285: [feature] shield: true for connectors

RobEnduro, 2022-07-13:
At this time, it doesn't seem like its possible to define a shield for connectors, only for cables.
Take Demo 01 for an example, where X1 is a 9-pin D-sub. It has pin 5 connected to the cable shield.
Generally, D-Sub connectors have a metal housing which could of course form electrical connection.
Outside of defining a "9-pin" connector with an explicit 10th pin for this purpose, is this something that can be done, or a possible feature that could be added?

kvid, 2022-07-14:
Thank you for your suggestion. What would you expect the connector should look like in the output diagram if implementing such a feature? Should it just be added as pin with a special name, or is that not enough?

Currently, there is no shield specifier for connectors, but you can specify custom pin names, e.g. like this if you want a pin named `s`:
```yaml
connectors:
  X1:
    type: D-Sub
    subtype: female
    pinlabels: [DCD, RX, TX, DTR, GND, DSR, RTS, CTS, RI, SHIELD]
    pins: [1, 2, 3, 4, 5, 6, 7, 8, 9, s]
```
Does this create the diagram you want, or would you like something different?

[image]

RobEnduro, 2022-07-16:
As a workaround, I probably would do something just like this. The only thing that seems off in this case is that the info given in the header box is "10-pin" on what is known as a 9-pin connector. I feel like what happens in the wire bundles would work fine in this kind of case, where only the explicit wires are counted and +S gets appended.

formatc1702, 2022-08-05:
> The only thing that seems off in this case is that the info given in the header box is "10-pin" on what is known as a 9-pin connector.

You can specify `pincount: 9` in addition to the `pins` and `pinlabels` lists. This way, the explicit pincout will drive what is shown in the top row of the connector node.