wireviz/WireViz #286: Ability to Specify Jumpers

AlanLivingston, 2022-07-19:
Currently, if I want to jump from a connector pin 3 to pin 4, I have to create a wire set, and in the connections, specify the connection like:
`connectors:`
`  X6:`
`    pinlabels: [+28VDC, +28VDC RTN, Token-In REF, Token-In, NC, NC, RS-485+, RS-485-, RS-485 Shield]`
`    type: P1`
``
`cables:`
`  W5:`
`    colors: [BU]`
``
`connections:`
`  -`
`    - X6: 3`
`    - W5: 1`
`    - X6: 4`

But this creates a very bulky diagram, when what I really want is a jumper from pin 3 to pin 4 on the left side of connector X6.

kvid, 2022-07-19:
I agree the current version (v0.3.2) doesn't handle such loops in the diagram very well, but there is an optional `loops` attribute you can use for connectors that should fit well in your use case:
```yaml
connectors:
  X6:
    pinlabels: [+28VDC, +28VDC RTN, Token-In REF, Token-In, NC, NC, RS-485+, RS-485-, RS-485 Shield]
    type: P1
    loops:
      - [3, 4]
```
[image]

See [the syntax doc](https://github.com/formatc1702/WireViz/blob/master/docs/syntax.md#connector-attributes), and also #48 where this feature and some possible alternatives are discussed.

kvid, 2022-08-23:
> I missed the loops attribute. That is, indeed, what I wanted.

👍

> 
> As mentioned in [#48](https://github.com/formatc1702/WireViz/issues/48), the length of the shorting wire needs some control. It overlapped the wire description box:

Suggestions about how to solve this are welcome!

> 
> 
> I was able to work around this by reordering the pins:
> 
> ```
>   X2:   # D369-P99-NS1
>     pins: [1, 2, 5, 6, 7, 8, 9, 3, 4]
>     pinlabels: [+28VDC, +28VDC RTN, NC, NC, RS-485+, RS-485-, RS-485 Shield, Token-In REF, Token-In]
>     type: P1
>     subtype: Deutsch D369-P99-NS1 9-Pin Receptacle Connector
>     loops:
>       - [8, 9]
> ```
> 
> [image]
> 
> I had to refer to the loop by the pin index within pins, and not its value. I don't know if this is intended or a bug.
> 

I would say it sounds like an unwanted feature.

rr-david, 2023-03-01:
A quick hacky solution would be to increase or give control of the spacing for tables such that the loops generated wouldn't overlap them.

moving X2 table 30mm to the right would make this useable

[image]

kvid, 2023-03-25:
Increasing the horizontal spacing, as suggested by @rr-david, can be done like this:
```yaml
connectors:
  X2:
    pinlabels: [+28VDC, +28VDC RTN, Token-In REF, Token-In, NC, NC, RS-485+, RS-485-, RS-485 Shield]
    type: P1
    loops:
      - [3, 4]
cables:
  W1:
    colors: [RD, BK, IV, BU, GN]
connections:
  - - W1: [1-5]
    - X2: [1-2, 7-9]
tweak:
  override:
    graph:
      ranksep: "3"  # Increase the horizontal spacing (default 2)
```
However, be aware that this is a global setting that affects spacing in the whole diagram.
[image]

blackbit42, 2023-04-05:
Increasing the horizontal spacing is useful for the current implementation of loops as it helps with the wire table not overlapping the loop(s). Though, doing so is only necessary because loops are currently very pronounced... well, loops.

Is there perhaps a clever way to use [GraphViz tweaking](https://github.com/formatc1702/WireViz/blob/master/docs/syntax.md#graphviz-tweaking-experimental) to influence the appearance of loops?

In case that cannot be done just for loops, I could imagine that a hint to GraphViz not to generally make bends less pronounced on a global level would potentially not hurt regular wires much but improve the appearance of loops.

kvid, 2023-04-05:
@blackbit42 wrote:
> Increasing the horizontal spacing is useful for the current implementation of loops as it helps with the wire table not overlapping the loop(s). Though, doing so is only necessary because loops are currently very pronounced... well, loops.

How wide the loops become seems to be dependent on other elements in the diagram,  their spacing, or some other relations. Compare the small loop in my single connector diagram above with the wider loops in the larger diagrams.

The reason behind such a difference has to be connected to the internal algorithms of Graphviz - that I don't know much abour. Please search/ask at Graphviz user phorums to investigate further.

> 
> Is there perhaps a clever way to use [GraphViz tweaking](https://github.com/formatc1702/WireViz/blob/master/docs/syntax.md#graphviz-tweaking-experimental) to influence the appearance of loops?

I recommend editing the `.gv` file to try out possible changes and run it through the Graphviz `dot` tool to see the result. When you know **what** to change, then we can see if it can be done with the `tweak` functionality.

> 
> In case that cannot be done just for loops, I could imagine that a hint to GraphViz not to generally make bends less pronounced on a global level would potentially not hurt regular wires much but improve the appearance of loops.

This is a question you should ask at some Graphviz phorum. In #3 you can find some examples on advanced Graphviz attributes for controlling the wire curves in a twisted wire usecase. Maybe that can be of any help.

kvid, 2024-05-04:
I stumbled on a possible solution to this issue today when reading this: https://stackoverflow.com/questions/70996779/graphviz-edges-between-cells-in-the-same-html-table-are-too-long

@AlanLivingston, @rr-david, @blackbit42, and others with similar use cases - please try this work-around (adding an empty label to all edges) for your use cases, and report here if it works or if you find any unexpected side effects:
```yaml
connectors:
  X2:
    pinlabels: [+28VDC, +28VDC RTN, Token-In REF, Token-In, NC, NC, RS-485+, RS-485-, RS-485 Shield]
    type: P1
    loops:
      - [3, 4]
cables:
  W1:
    colors: [RD, BK, IV, BU, GN]
connections:
  - - W1: [1-5]
    - X2: [1-2, 7-9]
tweak:
  override:
    edge:
      label: " "
```
[image]

martinrieder, 2024-05-16:
Instead of defining the loop by a single edge, I would simply split it into two edges, running through a node (either point or none). This should allow for a higher degree of freedom for Graphviz. Another side effect is that it will avoid edges overlapping the nodes.

tobiasfalk, 2024-05-17:
Wouldn't be a solution similar to this betrer:
[image]

This way the Jumper can not interfier with any wires and it is clear that Jumpers are meant.

kvid, 2024-05-17:
@tobiasfalk - your suggestion is very similar to the 4 year old suggestion in https://github.com/wireviz/WireViz/issues/48#issuecomment-656802355.

If you find a way to implement this in the Graphviz HTML syntax (in a `.gv` file), please share that - preferably in a new issue (or PR if you create code for it too). Such a visualization might also be useful for other features.

tobiasfalk, 2024-05-17:
@kvid I have no idea about GraphViz, it was only a idea that I thought I share

tobiasfalk, 2024-05-17:
I have tried this in the table:
```
   <tr>
    <td port="p1l">1</td>
    <td>+28VDC</td>
    <td port="p1J">&#11044;</td>
   </tr>
   <tr>
    <td port="p2l">2</td>
    <td>+28VDC RTN</td>0x00002B24
    <td port="p2J">⬤</td>
   </tr>
```
and:
```
	X2:p1J:c -- X2:p2J:c
```

And the SVG displays the Circle quite nicely, but the PNG is the Problem:
SVG(converted with Inkscape:
[image]
PNG:
[image]

I also do not know how to make the strait lines between the centers of the circles but one step at a time.

tobiasfalk, 2024-05-17:
The Problem seams to be the font(arial) what is a bit confusing. By setting it to Adobe Kaiti Std it works but I think this is not the rigth way to fix it.
https://stackoverflow.com/questions/6743382/graphviz-not-supporting-utf-8-encoding

martinrieder, 2024-05-17:
@tobiasfalk The PNG rendering is a Unicode issue that can probably be solved. See here: https://graphviz.org/doc/char.html
I am not sure how you would get the edge rendered in a straight dashed line though....

Note that I have updated https://github.com/wireviz/WireViz/issues/286#issuecomment-2116112620 with some pictures above.

tobiasfalk, 2024-05-17:
For the beginngin something without the doted line between the circle could be a start:
[image]

tobiasfalk, 2024-05-17:
[Here](https://stackoverflow.com/questions/13898151/drawing-outgoing-edges-starting-from-the-center-of-record-elements-in-graphviz) they talk about starting the line in the center of the node, with this it should be posible to do the dotet lines between the Point/Circles.
I am currently not at my PC so I am not able to try it.

tobiasfalk, 2024-05-17:
It seams with when the attributes `[headclip=false, tailclip=false]` are set this way than it begins and ends in the center but also is straight.

tobiasfalk, 2024-05-18:
Since doing the dashed line between the point may be not possible or at least quite hard, I though that coloring the point could improve readability.
I also asked for help on the WireViz forum: https://forum.graphviz.org/t/straitening-one-line-throu-a-table/2196

tobiasfalk, 2024-05-29:
https://gitlab.com/graphviz/graphviz/-/issues/2547

AlanLivingston, 2024-06-17:
From the diagrams, PR #369 is preferable to me.  I'll have to figure out how to build the development branch to test.

AlanLivingston, 2024-06-17:
I performed a quick test with the code that's in branch pr/369, adding an example ajl.yml and letting build_examples build it.

The resulting .png file didn't render the black dot, instead a rectangle with numbering is rendered.  The svg has the black circle, but the connecting lines are not centered.

I believe both of these issues are mentioned in #350 but maybe the changes didn't make it into the PR?

AlanLivingston, 2024-06-17:
Here's the yaml:

    connectors: # This is based on ex1 and adds some Shorts to the connectors
    X1: # An example of Shorts with more information
      pinlabels: [GND, VCC, RX, TX, GND, VCC, GND]
    X2: # An example of a simple short definition
      pinlabels: [GND, VCC, RX, TX, GND, VCC, GND]
      shorts: # The definition is the same as above but there is no additional_components for more information
        - SH1: [1, 5, 7]
        - SH2: [2, 6]
  
    cables:
      W1:
        wirecount: 4
        shield: true
    
    connections:
      -
        - X1: [1-4]
        - W1: [1-4]
        - X2: [1,2,4,3]
      -
        - X1: 1
        - W1: s

graphviz v0.20.3
Python v3.12.3
Windows 11 23H2 OS Build 22631.3737

AlanLivingston, 2024-06-17:
> Your connectors need some indentation, otherwise they will not be recognized at all.

I think that's because I suck at GitHub markdown!  X1 and X2 are indented in the .yml file.

AlanLivingston, 2024-06-17:
Here's the file, renamed as .log so GitHub will link it.  It's really ajl.yml.

[ajl.log](https://github.com/user-attachments/files/15877295/ajl.log)

tobiasfalk, 2024-06-17:
I am using Win10 with `dot - graphviz version 11.0.0 (20240428.1522)` and the PNG output works fine:
[image]

I have created a new [branch (Jumpes_sync)](https://github.com/tobiasfalk/WireViz/tree/Jumpes_sync)where I removed the Face attribute for the Black Circle, this breaks it on my system, but could you try it?
My result(PNG):
[image]

tobiasfalk, 2024-06-17:
Your YAML code is ex17

tobiasfalk, 2024-06-17:
As for the offset Circle in SVG, I have no Idea how to solve this, since this is a GraphViz Problem, mentioned here https://github.com/wireviz/WireViz/issues/350#issuecomment-2119239205
or [scnorth](https://forum.graphviz.org/u/scnorth)@https://forum.graphviz.org/t/text-in-table-in-svg-output-not-completly-centered/2199

> That’s curious. The native SVG driver is capable of generating text using text-anchor="middle", but the HTML table rows use text-anchor="start". Because the downstream SVG renderer cannot be relied on to use the same fonts or to render them in the same way as when Graphviz computes text box sizes, this can happen. Maybe the problem would be solved by using dot -Tsvg:cairo because cairo embeds fonts in the output (which is more precise, but at the cost of larger SVG files that are more difficult to process downstream and for humans to read and understand.)
>
> Someone could also look into whether lib/common/htmltable.c could be enhanced to call gvrender_textspan() with centered instead of left-justified text as needed. Offhand my hunch is this may not be as obviously easy as one might hope.

This would require a separate Cairo installation that currently can not be achieved with pip (to my knowledge) and I tried many things and was not able to install it in a way that I could test `dot -Tsvg:cairo`. Side note, if one is able to enable this with a pip install of GraphViz and Cairo it would also allow many other cool SVG inserten and manupulation stuff, but to be honest it was not on my top priorety list.

tobiasfalk, 2024-06-18:
@AlanLivingston 
Could you also try https://github.com/tobiasfalk/WireViz/tree/Jumpers_Test_2 with some other changes, in this branch ex17(your code) does not exist, but you can look at ex15.

AlanLivingston, 2024-06-18:
Here's ex17.png from branch Jumpes_sync:

[image]

And here's ex15.png from Jumpers_Test_2:

[image]

AlanLivingston, 2024-06-18:
I realized that this morning after trying to match Tobias' graphviz version.  The version I have installed is 11.0.0:

>C:\Users\alivingston>dot -v
dot - graphviz version 11.0.0 (20240428.1522)
libdir = "C:\Program Files\Graphviz\bin"
Activated plugin library: gvplugin_dot_layout.dll
Using layout: dot:dot_layout
Activated plugin library: gvplugin_core.dll
Using render: dot:core
Using device: dot:dot:core
The plugin configuration file:
        C:\Program Files\Graphviz\bin\config6
                was successfully loaded.
    render      :  cairo dot dot_json fig gdiplus json json0 map mp pic pov ps svg svg_inline tk xdot xdot_json
    layout      :  circo dot fdp neato nop nop1 nop2 osage patchwork sfdp twopi
    textlayout  :  textlayout
    device      :  bmp canon cmap cmapx cmapx_np dot dot_json emf emfplus eps fig gif gv imap imap_np ismap jpe jpeg jpg json json0 metafile mp pdf pic plain plain-ext png pov ps ps2 svg svg_inline svgz tif tiff tk xdot xdot1.2 xdot1.4 xdot_json
    loadimage   :  (lib) bmp eps gif jpe jpeg jpg png ps svg

tobiasfalk, 2024-06-18:
I am working on a solution with a image of a Black circle instead. This should sollen the Charakter debacle and the maybe the SVG offset problem, but I am not sure about the later one.

tobiasfalk, 2024-06-18:
@AlanLivingston could you test the latest Version of my Jumpers Branch, it now no longer uses characters but some GraphViz magic to draw dots at the ends of lines

tobiasfalk, 2024-06-18:
This not only eliminates potential Character Problems but also solves the SVG offset problem. It also does not requere any Pictures or so to be inserted, which may bloade files or make processing longer
My ex15:
[image]

AlanLivingston, 2024-06-18:
I concur.  I tested on my system with both ex15.yml and my funky test, ajl.yml, and both .png and .svg files render properly!

formatc1702, 2024-07-12:
Those latest renders look excellent!

My main thought is whether the individual short groups need to have names (SH1, SH2), or if in most cases, it would be obvious by seeing which pins are shorted (all GNDs, all VCCs, ...) and having duplicate info just clutters the table.
There's a point to be made for both variants, and the header row could be omitted if no named short lists exist (see image below).
```yaml
shorts:
- [1,2,3]  # if its obvious
- NAME: [4,5,6]  # if it's not
```

The other question is whether "shorts" is the proper word, since to me, a short circuit sounds like a fault. 
Some words that come to mind: jumpers, joints, common, shared, bridge(d)

Final point: having colorful short dots could be helpful, but I would lean towards black lines+dots, with optional colorful filling, to preserve full legibility even when printing in B&W or using very light colors (yellow, white).

[image]

tobiasfalk, 2024-07-12:
> My main thought is whether the individual short groups need to have names (SH1, SH2), or if in most cases, it would be obvious by seeing which pins are shorted (all GNDs, all VCCs, ...) and having [...]

You are right about both, for not showing the short name in the diagram, I would add the option to hide them. For not having names, I can currently not test how to do this but it is highly possible that it may not possible to Mix both. Partially because not having names I directly Sets the hide names to true.

> The other question is whether "shorts" is the proper word, since to me, a short circuit sounds like a fault. 
> Some words that come to mind: jumpers, joints, common, shared, bridge(d)

I see your point, and yes Jumpers or Bridges Sound good. 

> Final point: having colorful short dots could be helpful, but I would lean towards black lines+dots, with optional colorful filling, to preserve full legibility even when printing in B&W or using very light colors (yellow, white).

Not sure if I currently can color the dot like it is in your example, but yes makes sense.

> Bonus points for adding some white outline to avoid collisions between the "short lines" and the table borders!

So you want the line color String something like this: `<bgk_color>::BLACK::<bgk_color>`

bgk_color: Background Color, since this may not be white.

formatc1702, 2024-07-12:
> Should the pin names and labels really be separated from their numbers?

I've split off this discussion into #404

formatc1702, 2024-07-12:
> > My ex15:
> > ![ex15](https://private-user-images.githubusercontent.com/56540480/340836056-7e378e7f-2fd7-4b83-a5e0-f1b80ac2c8c7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA4MDM4NjQsIm5iZiI6MTcyMDgwMzU2NCwicGF0aCI6Ii81NjU0MDQ4MC8zNDA4MzYwNTYtN2UzNzhlN2YtMmZkNy00YjgzLWE1ZTAtZjFiODBhYzJjOGM3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzEyVDE2NTkyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk3YjZmOTkwYTdlNzIwZDVlY2Q5NDBhZDhhZmI0YjliOTA4YzBiOTg1NGY2ODI4YWRlYmZjNjY5NDYwNjA0OWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.aVW5FI_Iz9dIYT1GWxeIfF0xZjEMbX6akY9BekacSCg)
> 
> Comparing the left and right connector, I came to realize, a systematic behavior should define which columns the Jumpers are drawn into. Pin numbers may appear on the left or right side, sometimes even both. Should the pin names and labels really be separated from their numbers (as seen in the left connector)?
> 
> EDIT: To give you some background, I want to share some thoughts that I have been considering to post as a feature request recently: Currently, pin activation takes effect on both sides of the connector, causing the entire pin list to be shown on whichever side there is any connection. Should it also be possible to show individual pin listings on the left and right side of connectors? Or would it just overcomplicate things, especially in respect to the jumper implementation?

Regarding the order in which individual jumpers should appear (left to right), the simplest solution would be, the topmost pin that is part of a jumper set gets the leftmost column, and so on successively.

Theoretically, it could be possible to stack jumper sets vertically if there is no collision (e.g. pins 1+2 shorted, pins 3+4 shorted, pins 5+6 shorted, could all share one column), but this breaks down if jumper sets are meant to be named, and the algorithm will be more complex ;-)

formatc1702, 2024-07-12:
After some thinking, my preference would be, no smart nesting, and use the order in which they are defined, to preserve the intention of the harness designer. E.g. they could want a pin-wise systematic ordering (as I sketched above), alphabetical based on the name of the jumper set, by groups (GND first, then voltages ascending, ...).

> ```yaml
> shorts:
> - [ [1,2], [3,4] ]  # 1st column
> - NAME: [5,6] # 2nd column named
> ```

I like the approach! Code would need to handle lists (unnamed sets), dicts (named sets) and lists of lists (multiple sets in one column).

tobiasfalk, 2024-07-12:
> How about an explicit definition of the compact jumper set?
> 
> ```yaml
> shorts:
> - [ [1,2], [3,4] ]  # 1st column
> - NAME: [5,6] # 2nd column named
> ```
> 
> I would vote against some magical algorithm that reorders things on its own. 

Yes, it could be done and looks good, but I would for now implement the "basic" jumpers, maybe without naming, but I would make naming mandatory at the beginning and then add the option to not display the name and after thet that the name is not needed in the syntax.
When these are done, I would add the adition of having a list of jumpers in one jumper instance(your #1 colomm), maybe with colistition detection.
Another option would be to automatically change the color of the line(maybe with a black border) when a collisions occurs.

kvid, 2024-07-12:
As an additional usage of such optional labels in both `loops` and `shorts`, I suggested in my https://github.com/wireviz/WireViz/issues/224#issuecomment-2163805131 to optionally allow specifying more details of each entry as equally labeled entries in `additional_components` in the same connector, but still allowing labeled and unlabeled loops/shorts entries without matching entries in `additional_components`, and also still allowing other entries in `additional_components` that don't match any loops/shorts.

My main motivation was to avoid separate lists with very similar entries of loop wires, short components, and the general `additional_components`. A future benefit might be to also allow the dot syntax for such labels (I also called them _Local designators_) to support both named and unnamed entries, etc.

tobiasfalk, 2024-07-20:
> Those latest renders look excellent!
> 
> My main thought is whether the individual short groups need to have names (SH1, SH2), or if in most cases, it would be obvious by seeing which pins are shorted (all GNDs, all VCCs, ...) and having duplicate info just clutters the table. There's a point to be made for both variants, and the header row could be omitted if no named short lists exist (see image below).
> 
> ```yaml
> shorts:
> - [1,2,3]  # if its obvious
> - NAME: [4,5,6]  # if it's not
> ```
> 
> The other question is whether "shorts" is the proper word, since to me, a short circuit sounds like a fault. Some words that come to mind: jumpers, joints, common, shared, bridge(d)
> 
> Final point: having colorful short dots could be helpful, but I would lean towards black lines+dots, with optional colorful filling, to preserve full legibility even when printing in B&W or using very light colors (yellow, white). Bonus points for adding some white outline to avoid collisions between the "short lines" and the table borders!
> 
> ![image](https://private-user-images.githubusercontent.com/2447251/348274049-4bf143f0-2926-44b6-8bbc-57f0c6b1b064.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE1MTE5MjksIm5iZiI6MTcyMTUxMTYyOSwicGF0aCI6Ii8yNDQ3MjUxLzM0ODI3NDA0OS00YmYxNDNmMC0yOTI2LTQ0YjYtOGJiYy01N2YwYzZiMWIwNjQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcyMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MjBUMjE0MDI5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZWI1ZjY2YTMwNjc5YzE4ZjJlMWNkODcxNGM1MzdjNzE5NjJjMjk4ZmFhZTAwY2YwMDBlZmQ4NzkxN2IyYTJhZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.zfgBskiLDqnQl7zvzw-S6H75ZDq9asESwvYctef80gg)

 @formatc1702 I have added your syntax suggestion in #412 (but you can only have either of the two and not combine them in one connector), I now added a white border to the vertical line but Multicolor dots a bit of a problem and need more thinking of how to implement them, maybe a more advanced gvpr script,

kvid, 2024-12-29:
@formatc1702 wrote:
> The other question is whether "shorts" is the proper word, since to me, a short circuit sounds like a fault.
> Some words that come to mind: jumpers, joints, common, shared, bridge(d)

As far as I remember, the term started out as "Internal shorts", and I have probably contributed to the transition to "shorts", but today, I agree with @formatc1702 that we should discuss alternatives.

For terminal block usage, I've seen several terms in use:
 - Short circuit piece/bar/jumper
 - Shorting piece/bar/jumper/link
 - Short connector/piece
 - Connecting/connection strip/bar
 - Jumper (bar/busbar)
 - Jumper terminal strip
 - (Plug-in) bridge

In what other contexts than terminal blocks might this functionality be useful?

Personally, I agree with @AlanLivingston (in the original issue description) and with @formatc1702 that the term "jumpers" is a good alternative, unless someone see a possible conflict with some other connector feature (existing in the wild and/or to be added to WireViz in the future). I know the term "jumper" is also widely used for shorting two neighbor pins of a connector at a PCB, but I see no conflict - it's the same concept.

kvid, 2024-12-29:
I also got an idea while re-reading old comments in this issue, but I'm not sure if it fits here, or should be moved to another issue: I assume the list of pins for shorts/jumpers will accept unique pinlabels (and maybe colors) in addition to pin names. In addition, we could also choose to accept non-unique pinlabels and add all pins matching the pinlabel in such cases, e.g. GND for all pins with this label. Where does this idea belong?