wireviz/WireViz #287: [install] Development installation fails

kvid, 2022-08-08:
I try to [install the development version](https://github.com/formatc1702/WireViz/tree/refactor/big-refactor#installing-the-development-version) to try out PR #251, but the installation seems to fail.

Commands I try from Ubuntu bash in WSL1/Windows 10:
```bash
lsb_release -a
echo $SHELL
dot -V
git clone https://github.com/formatc1702/WireViz.git clean-WireViz
cd clean-WireViz/
git checkout refactor/big-refactor
python3.7 -m venv venv
. venv/bin/activate
python -V
pip3 install -e .
```

Output:
```bash
kvid@UBUNTU18.04.6-WSL1:~/src/formatc1702$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.6 LTS
Release:        18.04
Codename:       bionic
kvid@UBUNTU18.04.6-WSL1:~/src/formatc1702$ echo $SHELL
/bin/bash
kvid@UBUNTU18.04.6-WSL1:~/src/formatc1702$ dot -V
dot - graphviz version 2.40.1 (20161225.0304)
kvid@UBUNTU18.04.6-WSL1:~/src/formatc1702$ git clone https://github.com/formatc1702/WireViz.git clean-WireViz
Cloning into 'clean-WireViz'...
remote: Enumerating objects: 4150, done.
remote: Counting objects: 100% (741/741), done.
remote: Compressing objects: 100% (156/156), done.
remote: Total 4150 (delta 630), reused 604 (delta 569), pack-reused 3409
Receiving objects: 100% (4150/4150), 12.86 MiB | 1.20 MiB/s, done.
Resolving deltas: 100% (2671/2671), done.
kvid@UBUNTU18.04.6-WSL1:~/src/formatc1702$ cd clean-WireViz/
kvid@UBUNTU18.04.6-WSL1:~/src/formatc1702/clean-WireViz$ git checkout refactor/big-refactor
Branch 'refactor/big-refactor' set up to track remote branch 'refactor/big-refactor' from 'origin'.
Switched to a new branch 'refactor/big-refactor'
kvid@UBUNTU18.04.6-WSL1:~/src/formatc1702/clean-WireViz$ python3.7 -m venv venv
kvid@UBUNTU18.04.6-WSL1:~/src/formatc1702/clean-WireViz$ . venv/bin/activate
(venv) kvid@UBUNTU18.04.6-WSL1:~/src/formatc1702/clean-WireViz$ python -V
Python 3.7.5
(venv) kvid@UBUNTU18.04.6-WSL1:~/src/formatc1702/clean-WireViz$ pip3 install -e .
Obtaining file:///home/kvid/src/formatc1702/clean-WireViz
Collecting click (from wireviz==0.4.dev0)
  Using cached https://files.pythonhosted.org/packages/c2/f1/df59e28c642d583f7dacffb1e0965d0e00b218e0186d7858ac5233dce840/click-8.1.3-py3-none-any.whl
Collecting graphviz (from wireviz==0.4.dev0)
  Using cached https://files.pythonhosted.org/packages/de/5e/fcbb22c68208d39edff467809d06c9d81d7d27426460ebc598e55130c1aa/graphviz-0.20.1-py3-none-any.whl
Collecting pillow (from wireviz==0.4.dev0)
  Using cached https://files.pythonhosted.org/packages/8c/92/2975b464d9926dc667020ed1abfa6276e68c3571dcb77e43347e15ee9eed/Pillow-9.2.0.tar.gz
Collecting pyyaml (from wireviz==0.4.dev0)
  Using cached https://files.pythonhosted.org/packages/eb/5f/6e6fe6904e1a9c67bc2ca5629a69e7a5a0b17f079da838bab98a1e548b25/PyYAML-6.0-cp37-cp37m-manylinux_2_5_x86_64.manylinux1_x86_64.manylinux_2_12_x86_64.manylinux2010_x86_64.whl
Collecting tabulate (from wireviz==0.4.dev0)
  Using cached https://files.pythonhosted.org/packages/92/4e/e5a13fdb3e6f81ce11893523ff289870c87c8f1f289a7369fb0e9840c3bb/tabulate-0.8.10-py3-none-any.whl
Collecting importlib-metadata; python_version < "3.8" (from click->wireviz==0.4.dev0)
  Using cached https://files.pythonhosted.org/packages/d2/a2/8c239dc898138f208dd14b441b196e7b3032b94d3137d9d8453e186967fc/importlib_metadata-4.12.0-py3-none-any.whl
Collecting zipp>=0.5 (from importlib-metadata; python_version < "3.8"->click->wireviz==0.4.dev0)
  Using cached https://files.pythonhosted.org/packages/f0/36/639d6742bcc3ffdce8b85c31d79fcfae7bb04b95f0e5c4c6f8b206a038cc/zipp-3.8.1-py3-none-any.whl
Collecting typing-extensions>=3.6.4; python_version < "3.8" (from importlib-metadata; python_version < "3.8"->click->wireviz==0.4.dev0)
  Using cached https://files.pythonhosted.org/packages/ed/d6/2afc375a8d55b8be879d6b4986d4f69f01115e795e36827fd3a40166028b/typing_extensions-4.3.0-py3-none-any.whl
Building wheels for collected packages: pillow
  Running setup.py bdist_wheel for pillow ... error
  Complete output from command /home/kvid/src/formatc1702/clean-WireViz/venv/bin/python3.7 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-gle1tnef/pillow/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpkvl7roiypip-wheel- --python-tag cp37:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help

[... 153 lines cut ...]
     #include <Python.h>
              ^~~~~~~~~~
    compilation terminated.
    src/_imaging.c:75:10: fatal error: Python.h: No such file or directory
     #include "Python.h"
              ^~~~~~~~~~
    compilation terminated.
    building 'PIL._imagingmath' extension
    x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/freetype2 -I/usr/include/openjpeg-2.3 -I/tmp/pip-build-gle1tnef/pillow -I/usr/include/x86_64-linux-gnu -I/home/kvid/src/formatc1702/clean-WireViz/venv/include -I/usr/local/include -I/usr/include -I/home/kvid/src/formatc1702/clean-WireViz/venv/include -I/usr/include/python3.7m -c src/_imagingmath.c -o build/temp.linux-x86_64-3.7/src/_imagingmath.o
    building 'PIL._imagingmorph' extension
    x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/freetype2 -I/usr/include/openjpeg-2.3 -I/tmp/pip-build-gle1tnef/pillow -I/usr/include/x86_64-linux-gnu -I/home/kvid/src/formatc1702/clean-WireViz/venv/include -I/usr/local/include -I/usr/include -I/home/kvid/src/formatc1702/clean-WireViz/venv/include -I/usr/include/python3.7m -c src/_imagingmorph.c -o build/temp.linux-x86_64-3.7/src/_imagingmorph.o
    src/_imagingmath.c:16:10: fatal error: Python.h: No such file or directory
     #include "Python.h"
              ^~~~~~~~~~
    compilation terminated.
    src/_imagingmorph.c:14:10: fatal error: Python.h: No such file or directory
     #include "Python.h"
              ^~~~~~~~~~
    compilation terminated.
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1

    ----------------------------------------
Command "/home/kvid/src/formatc1702/clean-WireViz/venv/bin/python3.7 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-gle1tnef/pillow/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-exh6ikk8-record/install-record.txt --single-version-externally-managed --compile --install-headers /home/kvid/src/formatc1702/clean-WireViz/venv/include/site/python3.7/pillow" failed with error code 1 in /tmp/pip-build-gle1tnef/pillow/
(venv) kvid@UBUNTU18.04.6-WSL1:~/src/formatc1702/clean-WireViz$
```

formatc1702, 2022-08-08:
That's strange...

I tried reproducing it here, and was able to install it without problems.
I am on MacOS, however, nothing about the setup procedure has really changed (except for the entry point which now goes to `vw_cli.py`), and especially not with the dependencies...

I checked out `refactor/big-refactor` branch via GUI, then:

```
% python3 -m venv venv-new/
% source venv-new/bin/activate
(venv-new) ~/Projects/WireViz % pip install -e .
Obtaining file:///Users/daniel/Projects/WireViz
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Collecting pillow
  Using cached Pillow-9.2.0-cp39-cp39-macosx_10_10_x86_64.whl (3.1 MB)
Collecting graphviz
  Using cached graphviz-0.20.1-py3-none-any.whl (47 kB)
Collecting tabulate
  Using cached tabulate-0.8.10-py3-none-any.whl (29 kB)
Collecting click
  Using cached click-8.1.3-py3-none-any.whl (96 kB)
Collecting pyyaml
  Using cached PyYAML-6.0-cp39-cp39-macosx_10_9_x86_64.whl (197 kB)
Installing collected packages: tabulate, pyyaml, pillow, graphviz, click, wireviz
  Running setup.py develop for wireviz
Successfully installed click-8.1.3 graphviz-0.20.1 pillow-9.2.0 pyyaml-6.0 tabulate-0.8.10 wireviz-0.4.dev0
WARNING: You are using pip version 21.2.4; however, version 22.2.2 is available.
You should consider upgrading via the '/Users/daniel/Projects/WireViz/venv-new/bin/python3.9 -m pip install --upgrade pip' command.
(venv-new) % pip list
Package                  Version  Location
------------------------ -------- ----------------------------------
click                    8.1.3
graphviz                 0.20.1
Pillow                   9.2.0
pip                      21.2.4
PyYAML                   6.0
setuptools               57.4.0
tabulate                 0.8.10
wireviz                  0.4.dev0 /Users/daniel/Projects/WireViz/src
wireviz-o9vk7u9n9yc5n42l 0.3
```

All good.

formatc1702, 2022-08-08:
Have you tried with Python >= 3.8? I have 3.9 installed.

formatc1702, 2022-08-10:
Tried WSL at work. It had Python 3.8 installed and WireViz installation worked smoothly with the steps you describe, except for using `python3` instead of `python3.7`.

kvid, 2022-08-12:
Now, I've tried in Windows 10 with Python 3.9.13, and this similar installation procedure seems to work there:
```bash
ver
echo %COMSPEC%
dot -V
git clone https://github.com/formatc1702/WireViz.git clean-WireViz
cd clean-WireViz/
git checkout refactor/big-refactor
python3 -m venv venv
venv\Scripts\activate.bat
python -V
pip3 install -e .
```
I still don't understand why the installation fails in my WSL1 environment. For many years I've done most of my Python coding (including WireViz contributions) in this WSL1 environment. One difference I can see, is that the global Python environment in WSL1 has more than 200 packages installed. A lot of these packages are due to my ROS installation that I need for another project, and I therefore cannot just uninstall them for testing the WireViz installation. However, I still have a virtual environment in my original WSL1 WireViz workspace that works together with the current `master` and `dev` branches.

Running `python3` in my WSL1 with Ubuntu 18.04.6 LTS will start Python 3.6.9, and I therefore in 2020 installed Python 3.7 in addition to comply with WireViz requirements.

formatc1702, 2022-08-13:
Yes, I highly recommend using `venv` and/or [`pipx`](https://github.com/pypa/pipx) to keep the project's workspace clean :)

Also, I am planning to drop Python 3.7 since that version will [stop receiving security fixes next year](https://en.wikipedia.org/wiki/History_of_Python#Table_of_versions) and the new version of `setup.py` reflects this. I guess I should add 3.11 soon as well.

closed 2023-04-10 by kvid