wireviz/WireViz #291: [feature] connector pin attributes

T0jan, 2022-08-26:
Hello,

I usually use additional parameters in my harness plans like the signal direction or the voltage levels as additional layers of security. One could add them to the pinlabels ofc but I would like to see them in an additional column for each connector. If it is not already implemented and I missed it somewhere I would suggest a solution with optional attributes for the pinout definition:

```
connectors:
  J101:
    pinlabels: [+5V, +5V, GND, GND, RX+, RX-, TX+, TX-, SDA, SCL]
    pinattributes1: [PWR, PWR, PWR-RET, PWR-RET, IN, IN, OUT, OUT, IO, OUT]
    pinattributes2: [5V, 5V, GND, GND, RS422, RS422, RS422, RS422, 3V3, 3V3]
```

I think giving the option to add two or three individual attribute fields should cover the most cases in which this could be useful.

kvid, 2023-04-06:
Thank you @T0jan for this feature suggestion. I'm sorry you had to wait so long without any feed-back. I'm not against your suggestion, but I just want to present a possible/optional alternative syntax to specify the same attributes:
```yaml
connectors:
  J101:
    pinlabels:
      - [+5V, PWR,     5V]
      - [+5V, PWR,     5V]
      - [GND, PWR-RET, GND]
      - [GND, PWR-RET, GND]
      - [RX+, IN,      RS422]
      - [RX-, IN,      RS422]
      - [TX+, OUT,     RS422]
      - [TX-, OUT,     RS422]
      - [SDA, IO,      3V3]
      - [SCL, OUT,     3V3]
```

T0jan, 2023-04-11:
@kvid All good, I am glad too see this is not dead. I like your suggestion much more then my initial idea as it is much better readable. I think my idea was based on how it was done for the other properties like colors or pin numbers so I wanted to keep it similar but I think your solution is better to work with for the user.

Would there be a limit on how many custom columns a user can add like this?

kvid, 2023-05-24:
@T0jan wrote:
> @kvid All good, I am glad too see this is not dead.

Don't worry! Open issues are not dead, but sometimes it might take a while before someone reply, and even after a reply with suggestions, it doesn't always mean that someone will implement such suggestions soon. However, if you (or someone else) are motivated to try some Python coding yourself, feel free to make a note of it here and then [submit a PR](https://github.com/formatc1702/WireViz/blob/master/docs/CONTRIBUTING.md#submitting-a-new-pull-request). It's often useful to discuss alternative functionalities and visual appearances in this open issue first, before diving into the coding job, but problems or design choices that appear during coding might also be discussed here (or in the PR when that is created).

> I like your suggestion much more then my initial idea as it is much better readable. I think my idea was based on how it was done for the other properties like colors or pin numbers so I wanted to keep it similar but I think your solution is better to work with for the user.

The same attribute can be able to take simple values for simple use cases, and more complex values when needed for complex use cases. I think this is a convenient way to expand the functionality without breaking compatibility with the original simple use case.

> 
> Would there be a limit on how many custom columns a user can add like this?

I haven't tried coding any of this, so I don't know for sure, but in principle, it doesn't need to be such a limit.

kvid, 2024-06-29:
@martinrieder - I'm not against your approach. The syntax is very similar to my suggestion from last year above, but the functionality of your approach is perhaps closer to the suggestion in #279 that also is an optional alternative syntax for existing list attributes. The different approaches don't exclude each other, though. Any conflicts between specifying different values for the same attribute via the different approaches should raise an exception.

Comments about your syntax:
- To support the original request of this issue, would you suggest `pinattributes1` and `pinattributes2` as headers, or would you like to support any custom header (or any with e.g. a `userattr.` prefix to avoid conflicts with future attribute names) as user defined labels?
- Instead of `pintable` and `wiretable`, I suggest  a common name, e.g. `attr_table` or `attributes_table`.
- Do you also consider optionally accepting something like `attr_table: X1_attributes.tsv` (or from a CSV file) as a future expansion of your approach?

kvid, 2024-06-29:
@martinrieder wrote:
> [...]
> > * Instead of `pintable` and `wiretable`, I suggest  a common name, e.g. `attr_table` or `attributes_table`.
> 
> Maybe simply name it `attributes` or `table`. I am open for suggestions, though I prefer short names without abbreviation.

Of these two alternatives, I prefer `attributes` because `table` doesn't say anything about what to specify.

> 
> > * Do you also consider optionally accepting something like `attr_table: X1_attributes.tsv` (or from a CSV file) as a future expansion of your approach?
> 
> Not sure about this. It seems to be a separate issue to me. This file would definitely require the same header in the first row... It could possibly be included through Jinja as an advanced feature:

Then we forget about this. It's not a feature I _want_. I just asked if you considered something like this because you wrote "_Being able to load these as a table i.e. from CSV provides a big usability improvement._"

> 
> > Any conflicts between specifying different values for the same attribute via the different approaches should raise an exception.
> 
> I prefer warnings or error messages over exceptions.

I agree about preferring warnings when it makes sense to ignore the unexpected event and continue, but what do you mean by preferring "_error messages over exceptions_"?

All exceptions provide an error message describing the problem. In some cases it might make sense to catch some low level exceptions and raise a higher level exception with more context information, but in most such cases it's useful to also include the low level error message as the root cause.

> The implementation of additional components needs to be discussed with respect to #224.

How does this issue affects anything about additional components?

kvid, 2024-10-10:
@Jachimo wrote:
> Not sure what the current status of this feature is,

It's one of many feature requests that are still open for discussions, but not yet have any suggested implementation.

> but I'd just like to note that the ability to explicitly specify the name/label for each pin explicitly (as a dictionary or some other similar structure) would be great when working on large connectors such as MIL-38999s.

I agree, and personally, I argue for accepting alternative ways to specify such information, because different use cases might have different needs.

> [...]
> It would be nice to be able to write something like:
> 
> ```
> connectors:
>   J24:
>     pins:
>       - name: A
>         label: GPS
>         type: M39029/103-559
>       - name: Z
>         label: GPS_VCC
>         type: M39029/1-100
>       - name: a
>         label: GPS_GND
>         type: M39029/1-100
> ```

I agree this might be useful, but it seems like a duplicate of what is already suggested in #279, so please add your comments to that issue. The only detail that seems to differ, is your `name` attribute instead of `pin` - and that can be your first contribution/suggestion to #279.