wireviz/WireViz #293: Force connector to wire connections to the same side

martonmiklos, 2022-09-15:
Hello all,

I am trying to document a H topology cable:
[image]
From the following source:

https://gist.github.com/martonmiklos/8603dcaaebea8eea07143e39b3de4fa0

It renders what I want to make, but the arranging is not quite as descriptive as it should. 

Is there any way to force the +12V out 2 and GND out 2 connctors:
[image]

To force the wire entry to one side like happens on the +12V out 1 and GND out 1?
[image]

kvid, 2022-09-15:
Do you perhaps want a layout similar to [Tutorial 7](https://github.com/formatc1702/WireViz/tree/master/tutorial#07---daisy-chaining-ii) with all connectors along the left and right sides with zig-zag cables daisy-chained between them? If so, choose alternating sides for each connector in the daisy-chain, and specify all connections from the left connector via the cable to the right connector.

In your use case, you might try the "in 2" connectors left, then the "out 2" and "in 1" connectors right, and finally "out 1" connectors left, but swapping left and right sides should also work.

kvid, 2022-09-16:
My suggestion above would look something like this:
[image]
The main difference from the original source file is the order of elements within some of the connections. I replaced the original connections with these:
```yaml
connections:
  # hosszú szár normál #1
  -
    - ['+12V out 1', 'GND out 1', 'XLiNE out 1']
    - '500 mm H hosszú szára #1': [1-3]
    - ['+12V in 1', 'GND in 1', 'XLiNE in 1']

  # hosszú szár normál #2
  -
    - ['+12V in 2', 'GND in 2', 'XLiNE in 2']
    - '500 mm H hosszú szára #2': [1-3]
    - ['+12V out 2', 'GND out 2', 'XLiNE out 2']

  # átkötő
  -
    - ['+12V out 1', 'GND out 1']
    - '300 mm átkötő': [1-2]
    - ['+12V out 2', 'GND out 2']
```
- I have simplified the source by joining all wires of a cable in the same connection entry, and dropped the redundant pin 1 specifier for simple 1-pin connectors, but none of that change the output in any way.
- I have reversed the order of connection entries for cables `hosszú szára #1` and `átkötő` to obtain the zig-zag cable pattern as in Tutorial 7.

martonmiklos, 2022-09-16:
Many thanks for the hint @kvid !

I was not aware of the ordering importance!

Do you know if the orientation of cable exit ports on the boxes could be set somehow?

The cable in the question is rather a H cable than a Z in reality, and it could have added value to visualize that way.

kvid, 2022-09-16:
1. The left or right exit port is determined by the order of connection entries: For each wire spline between two connection entries, it goes from the right side of one entry to the left side of the next entry.

2. When you describe the harness as an H, does that mean the middle cable is actually connected between the middle points of the two other cables, and not the connectors in either end? If so, then you might consider including those middle points in the harness as 3-way wire junctions, but all cable boxes in the diagram will have a left-right orientation, so a true H with both horizontal and vertical cables might not be so easy to obtain.

nindroz, 2024-05-15:
For CAN wiring purposes I was looking to genuinely force wires out of one side, does anyone know if its doable?

kvid, 2024-05-15:
@nindroz wrote:
> For CAN wiring purposes I was looking to genuinely force wires out of one side, does anyone know if its doable?

I know what a CAN bus is, but I don't really understand what you're trying to achieve. Can you show here what you've tried, and then explain what you would like to be different. If your problem has very little in common with what was asked or suggested here, you should consider creating a new issue.

martonmiklos, 2024-05-15:
> @nindroz wrote:
> 
> > For CAN wiring purposes I was looking to genuinely force wires out of one side, does anyone know if its doable?
> 
> I know what a CAN bus is, but I don't really understand what you're trying to achieve. 

If I guess well he is wiring the CAN bus to a node back and forth (by connecting two wires to a same pin). To describe what is done physically the wires needs to be connected on the same side to the connector not the opposite.

nindroz, 2024-05-16:
Essentially, I need the green and yellow wires going into the left side of the TT_FR connector. It needs to be on the left of both connectors as they recieve the other 2 wires (power and ground from a connector the far left)
[image]

kvid, 2024-05-16:
@nindroz wrote:
> Essentially, I need the green and yellow wires going into the left side of the TT_FR connector. It needs to be on the left of both connectors as they recieve the other 2 wires (power and ground from a connector the far left)

After you've decided the left-right order of your cables and connectors, then specify connections ordered from left to right. See also point 1 in https://github.com/wireviz/WireViz/issues/293#issuecomment-1249843359.

nindroz, 2024-05-17:
For the CAN connection, there is no left and right for the connectors. They esssentially need to be parallell.  I specifically need the green and yellow wire in the picture to be on the left of the TT_FR connector and the left of the steering connector.

kvid, 2024-05-18:
Sorry, but I don't understand which end of the TT_FRC bundle that is connected to the TT_FR connector. This connector seems to be connected to both ends, and that doesn't normally make sense.

tobiasfalk, 2024-05-19:
@nindroz could you give the wirvize code? this may allow experimenting to look what you want.

tobiasfalk, 2024-05-22:
@nindroz are any of these three that what you meant?
1:
[image]

2:
[image]

3:
[image]

nindroz, 2024-05-22:
The last one!

EloiVilalta, 2024-06-05:
Hello, exactly what I was looking for!
Right now we can only do "right to left" connections following the Connector-Cable-Connector method. But @tobiasfalk , in the "2" you got "left" TT_FRC to "left" TT_FR and in "3" you got "left" TT_FRC to "left" _X_3. How did you do it?

The [daisy chain example](https://github.com/wireviz/WireViz/tree/master/tutorial#07---daisy-chaining-ii) is changing the connectors from left to right, but I think that @nindroz and me need the connectors always on the same side. 
This is what I'm trying to achieve (in vertical position):
[image]
[Here](https://res.cloudinary.com/engineering-com/image/upload/v1541085989/tips/RS485_vqi7tu.png) is another example of what I mean (in horizontal position, but we can't make vertical connections, right?).

I understand that we'll have to place the cable ("TT_FRC" in this case) between the connectors (in my case the "TT_FR" and "_X_3" are named "Y/T splice.1" and "Y/T splice.2"), but at some point we need the "left to left" connection as you did.

Let us know how you did it please! :D

Thanks!
Eloi

PD: vertical connections would be awesome, I did not checked if it's already possible or not.

tobiasfalk, 2024-06-05:
> Hello, exactly what I was looking for!
> Right now we can only do "right to left" connections following the Connector-Cable-Connector method. But @tobiasfalk , in the "2" you got "left" TT_FRC to "left" TT_FR and in "3" you got "left" TT_FRC to "left" _X_3. How did you do it?

I did this by editing the .gv output to give a better visualisation of what @nindroz meant.

EloiVilalta, 2024-06-05:
@nindroz , I finally got it. Let me explain how I did it so everyone can solve this issue with a super basic example:

"file_name".yml:
```
connectors:
  C1:
    style: simple
    type: Connector
  C2:
    style: simple
    type: Connector      
cables:  
  W1:
    wirecount: 1
    show_name: false
    show_wirecount: false
    colors: [RD]

connections:
  -
    - C1: [1]
    - W1: [1]
    - C2: [1]
``` 
This is what we get:
[image]

As .gv file [is not generated nowadays](https://github.com/wireviz/WireViz/blob/master/docs/CHANGELOG.md#backward-incompatible-changes) by default we need to add `-f g` after `wireviz "file_name".yml` to produce as output the .gv file, so `wireviz "file_name".yml -f g`.

With the .gv open, you have to search and modify the second connection:
```diff
 > fillcolor="#FFFFFF" shape=box style=filled]
  edge [color="#000000:#ff0000:#000000"]
  C1:e -- W1:w1:w
-  W1:w1:e -- C2:w
+  W1:w1:w -- C2:w
  W1 [label=<
``` 
The "e" means "right", and the "w" means "left".
Save it.
Use `dot -O -Tpng "new_file_name.gv"` to generate the image again.
Now we have this new .png image:
[image]

Try to imagine it as:
[image]

Hope it helps!

tobiasfalk, 2024-06-05:
Yes that is basically what I did in the .gv output

EloiVilalta, 2024-06-05:
Sure @tobiasfalk , I previously asked how to do it, but ended up learning it and sharing how to do it ;) Thanks!

For everybody,

In the example above I changed a **cable** connection, but in case you want to change a **connector** (Splice), keep in mind there is more "info" in the .gv that has to be changed. I only could go from this:
[image]
To this:
[image]

I had to change, as explained before, the "e" to "w":
```diff
 > fillcolor="#FFFFFF" shape=box style=filled]
  edge [color="#000000:#ff0000:#000000"]
-  JB3:p1r:e -- W3:w1:w
+  JB3:p1l:w -- W3:w1:w
  W3:w1:e -- JB4:p1l:w
  edge [color="#000000:#8000ff:#000000"]
-  JB3:p2r:e -- W3:w2:w
+  JB3:p2l:w -- W3:w2:w
  W3:w2:e -- JB4:p2l:w
  edge [color="#000000:#ffffff:#000000"]
-  JB3:p3r:e -- W3:w3:w
+  JB3:p3l:w -- W3:w3:w
  W3:w3:e -- JB4:p3l:w
  edge [color="#000000:#000000:#000000"]
-  JB3:p4r:e -- W3:w4:w
+  JB3:p4l:w -- W3:w4:w
  W3:w4:e -- JB4:p4l:w
  edge [color="#000000"]
-  JB3:p5r:e -- W3:ws:w
+  JB3:p5l:w -- W3:ws:w
  W3:ws:e -- JB4:p5l:w
  W3 [label=<
``` 
And another time for the second cable in my code.

Keep in mind to also change the "r" to "l", as it refers to the "numbered box", if you don't change the "r" to "l" you will get the "left" side of the "left" box:
[image]

I would suggest to "redistribute/regroup" the cables and connectors prior the modification of the .gv file using [this tweak](https://github.com/wireviz/WireViz/issues/174#issuecomment-707372543).

Anyone know how to tweak (override/append) the "left/right" issue inside the .yml file? 
Thanks!

kvid, 2024-06-05:
@EloiVilalta wrote:
> [...]
> The [daisy chain example](https://github.com/wireviz/WireViz/tree/master/tutorial#07---daisy-chaining-ii) is changing the connectors from left to right, but I think that @nindroz and me need the connectors always on the same side. This is what I'm trying to achieve (in vertical position):  

If what you really need, is daisy-chaining bus wires between a series of connectors, why do you require all wires connecting from the same side?

> [Here](https://res.cloudinary.com/engineering-com/image/upload/v1541085989/tips/RS485_vqi7tu.png) is another example of what I mean (in horizontal position, but we can't make vertical connections, right?).

The RS485 use case you link to, can easily be implemented (similarly to tutorial04) in WireVis like this:
[image]
```yaml
connectors:
  X:
    pinlabels: [GND, B-, A+]
    color: GN
  J:
    type: Junction
    style: simple
cables:
  W:
    colors: [BN, YE, GN]
    shield: true
connections:
  -
    - X.: [1-3]
    - W.W1: [1-3]
    - X.: [1-3]
    - W.W2: [1-3]
    - X.: [1-3]
  -
    - J.
    - W1: [s]
    - J.
    - W2: [s]
    - J.
```
The number of connector can easily be expanded, but when the number of connector increases, this diagram gets very wide. Then I would rather recommend zig-zag daisy-chaining like in tutorial07 to obtain a more compact diagram. Electrically, it doesn't matter which of these drawing techniques are used. The connections are the same, just drawn in a different layout and order to help readability.

Forcing wires around to the opposite side, like in some of the examples above, will only harm readability, IMHO.

EloiVilalta, 2024-06-05:
Thank you @kvid , my problem is that from the " bus connections" there is custom cables going to the devices, with different pinout, different colors... 
I'm designing a redundant bus with differents Y splices that needs to be resin moulded, some with two cables inside, some with just one, some with parallel cables, some with cables that joins to the "bus connection".

Anyway, I've been checking all day what GraphViz is capable of doing and you [can only choose ONE direction](https://graphviz.org/docs/attrs/rankdir/). 
Unless we can do a subgraph with different direction settings (I tried it and could not make it because it gets overwritten), we cannot do it... This guy was [trying the same](https://stackoverflow.com/questions/7777722/top-down-subgraphs-left-right-inside-subgraphs) on stackoverflow.

> why do you require all wires connecting from the same side?

The schematics had been like this since the beginning of the project, and change it now it's pretty difficult. I wanted to place the detailed cables on top of the actual schematic.

Another problem I faced is that the cables will choose to go over or under the "box" depending if they start over the center or under it. Unless I put the "picture" on top and all the cables goes at the bottom, it's impossible to make it look fancy, here is the example:
[image]
I could not figure it out any attribute to force to go over or under the node. It looks like (not working on DOT), there is [this option](https://graphviz.org/docs/attr-types/splineType/) that you can edit the "splines"... 

I'll have to make it Zig-Zag. But I'm really happy about what I've learnt today! And I'm pretty sure I'll use GraphViz more often!

BTW, I'm using http://magjac.com/graphviz-visual-editor/ to edit the .gv and see the changes in real time.

martinrieder, 2024-06-05:
FYI: Graphviz comes with a tool called `gvpr` that can be used to post-process the graph. Just a hint that it is worth taking a look...
https://graphviz.org/pdf/gvpr.1.pdf

akikinho, 2024-06-20:
Hello Wireviz Team,
I am trying to understand the usage of dot notation in Wireviz but unable to replicate as it throws an exception 'First item not found anywhere.' 

```
connectors:
  X:
    pinlabels: [GND,B-,A+]
    color: GN
    
cables:
  W:
    colors: [BN,YE,GN]
    shield: true
    
connections:
  -
    - X.: [1-3]
    - W.W1: [1-3]
    - X.: [1-3]  
```

emileakbarzadeh, 2025-04-09:
To add on to the discussion here, I have found the need to specify the position of the wires to a connector extremely important for wire splices. The diagram needs to show the correct direction for wire splices for the assembler.

The order of connections can solve this problem most of the time, but I occasionally need to describe splices where the wire is on the left of both connectors.