wireviz/WireViz #294: [feature] Do not graph unused library connectors included with --prepend-file

alecraymond, 2022-09-23:
I am diagramming a fairly large system (~50 cable assemblies). I've put all of my connectors in a connectors.yml file that I include using the command line --prepend-file option. It would be nice to be able to graph various combinations of connectors in cables/connections.yml files each using a subset of the connectors in connectors.yml. It seems that all connectors are graphed by default. This results in each of my cables/connections.yml files graphing all of the intended cable assemblies, plus all of the unused connectors from connectors.yml.

kvid, 2022-09-24:
Thank you for this suggested feature. It would need to be an optional feature. Othewise, it would be impossible to document single connectors without any connections.

However, you might also consider using the existing template feature as shown in [Tutorial 3](https://github.com/formatc1702/WireViz/tree/master/tutorial#03---pinouts-shielding-templates-i). Then you can

kvid, 2022-09-24:
I'm sorry that I managed to close this issue by a mistake while writing my answer, and my answer was also saved before it was finished. That was not my intention. I will edit and complete my answer text shortly.

alecraymond, 2022-09-26:
@kvid Thanks for the quick and detailed response! I agree that it would need to be an optional feature, preserving the ability to document unused connectors.

Using inheritance as you suggest does result in the diagrams that I expect, and the use of templates reduces maintenance to zero as long as the count/names of templates reused within each child.yml do not need to be changed as the cables and connections are changed. It would be convenient not to be required to copy all of the relevant connector templates into each child.yml, but I will do it this way.

kvid, 2024-05-12:
@alecraymond - v0.4 was released today, and it includes new functionality that I believe resolves this issue. Please try it and report back here your findings.

alecraymond, 2024-05-17:
@kvid Yes, this resolves this issue perfectly. I can include a single connector template file to generate diagrams that display only the cable assemblies with connections. I also like that connectors and cables that are not referenced are displayed as a warning in the output.

closed 2024-05-17 by alecraymond