wireviz/WireViz #299: Generated output images (all formats) width and height

tobiokanobi, 2022-12-12:
Great piece of work! (But) When using it under Windows Subsystem for Linux (WSL), the generated output files seem to have width and height reversed?

Attached to images, the first is the tutorial01, unchanged, the second is a copy and new created image (tobi_1).

[image]
[image]

Kind regards,

Tobias

kvid, 2022-12-12:
Which release or branch of WireViz are you using? This should not happen with the latest release.

This seems to be a duplicate of #258 that was fixed by PR #261 and then merged into master in [v0.3.2](https://github.com/formatc1702/WireViz/releases/tag/v0.3.2), but not currently included in dev, latest, or any of the PR branches. If you need this fix together with such a branch, then I suggest you create your own local branch from the dev or PR branch and cherry-pick the fix https://github.com/formatc1702/WireViz/commit/9af0cb8ab0314bc6754b153769f7562b5b38ec10 on top of it.

tobiokanobi, 2022-12-12:
The branch seems to be *dev ; the version is 0.4dev0; 

I checked the code (found the missing new lines) and tried to cherry pick, but just got some funny errors after that. I may have done some *idots errors*, by changing just the code, saving it and then trying to run it again.

tobiokanobi, 2022-12-12:
Strange, uninstalled it, installed latest version again (pip3 install wireviz), suddenly the version is 0.3.2 and it is working :-?

kvid, 2022-12-12:
Maybe you have earlier cloned the git repository and installed the dev branch. Now, when you have installed the latest release, it should work without needing the cherry-pick.

tobiokanobi, 2022-12-12:
Guess you are right! Thanks for help.

closed 2022-12-12 by tobiokanobi

Tensdale, 2023-03-31:
I am currently also having this exact problem after cloning the current repository:

!pip3 install wireviz
!pip3 install --upgrade wireviz

The example codes gets saved like this:

[image]

tobiokanobi, 2023-03-31:
Check which branch you cloned (master or dev.) . This was my problem.