wireviz/WireViz #300: [feauture] ignore additional components in bom

ask6483, 2023-01-05:
Is it possible to also ignore additional components in bom generation?
For example **ignore_in_bom** will hide connector housing but not crimp terminals:
```
  J1:
    pincount: 2
    type: PicoBlade 1.25 mm 
    subtype: Receptacle Crimp Housing
    manufacturer: Molex
    mpn: '510210200'
    pinlabels: [GND, VCC]
    notes: Camber Main PCB
    ignore_in_bom: true
    additional_components:
      -
        type: Female Crimp Terminal
        subtype: |
          PicoBlade 1.25 mm 
          Tin Plating, 26-28AWG
        qty_multiplier: populated
        manufacturer: Molex
        mpn: '500798001'
```
I want to create cable diagram but the part is available as pre-crimped wire so there is only one item in bom.

kvid, 2023-01-06:
Thank you for reporting your use case. This must be a simple bug, was my first impression, and I quickly located two code locations where I added one extra indentation level to make them part of the `if not ignore_in_bom:` block just above.

https://github.com/formatc1702/WireViz/blob/master/src/wireviz/wv_bom.py#L83-L84
https://github.com/formatc1702/WireViz/blob/master/src/wireviz/wv_bom.py#L114-L115

That simple fix described above does indeed what you ask for, but only if `options.mini_bom_mode` is False, and the default value is True. When `options.mini_bom_mode` is True, then any list of additional components in the diagram nodes are supposed to contain the BOM entry IDs as a reference for more information details. That code fails with an exception when there is no matching entry in the BOM. We need to decide how to handle such a conflict.

- One way could be to disallow hiding a connector/cable with additional components when `options.mini_bom_mode` is True.
- Another way could be to ignore `options.mini_bom_mode` for additional components that are hidden from the BOM.
- Please suggest other alternatives.

kvid, 2023-01-16:
@ask6483, do you also request controlling `ignore_in_bom` for each additional component? That would enable ignoring any combination of connector/cable entry and additional component entries in the BOM. However, the same conflicts as described above, still exist, and must be handled.