wireviz/WireViz #301: [Error] Cannot run WireViz after proper installation on Mac OS

ShimmyG, 2023-01-11:
Hi, I am a noob to all this coding and am still learning, so I may very well be overlooking something here. I followed all the instructions to install WireViz and am able to import it on python and the Apple terminal.

When I execute the run line `wireviz ~/path/to/file/mywire.yml`,
I am receiving this error: `Error: input file /Users/user/path/to/file/mywire.yml inaccessible or does not exist, check path`

I also cannot locate this file path or open it through the command line. I checked and yaml module is installed as well as graphviz. Any help would be greatly appreciated!

kvid, 2023-01-11:
Hi, I know a bit about using WireViz in Linux and Windows, but I don't have any machine running Mac OS. However, I can try explaining why you get these error messages.
1. In your first use case, you ask WireViz to process the input file `~/path/to/file/mywire.yml` and your operating system or shell translated the leading `~` into `/Users/user` which I assume is the correct home path of the current user at your machine. The error message you got tells me that WireViz tried to open  `/Users/user/path/to/file/mywire.yml` as an input file, but failed to do so. Probable causes for this might be:
  - You forgot to create an input file before executing WireViz.
  - The filename or path to the input file you created is misspelled.
  - The current user does not have read access to the input file.
2. In your second use case, you execute WireViz without any command line arguments. Due to this, WireViz does not know what input to process, and prints a simple help text to give you a hint about the command line argument alternatives it expects. The command line argument alternatives in brackets are optional, but a YAML input file argument is required to specify what input you want WireViz to process.
3. See the [documentation](https://github.com/formatc1702/WireViz/tree/master/docs) for examples and more information on how to create a proper input file for WireViz.
4. I hope this helps you a bit further to understand how to use WireViz, but don't be afraid to ask for more help if needed.

ShimmyG, 2023-01-11:
I appreciate the help and thanks to you I figured it out. I thought to myself the directory was to a specific folder, but rather it was just an example directory: `/path/to/file/mywire.yml`
I put a sample file in my document directory and the module was able to execute it correctly. Thanks!

closed 2023-01-11 by ShimmyG