wireviz/WireViz #302: Multiple suppliers per connector/cable?

nyq, 2023-01-17:
Is there a way to specify multiple suppliers for the same connector/cable? I tried so specify more than one pair of supplier/spn keys, but all it does is simply overrides all previous pair values and just uses the last pair specified.

This feature is often needed in production environments, as it is typical to maintain several suppliers for the same part on file to ensure the ability to purchase from alternative sources if there are stocking issues with the primary supplier.

kvid, 2023-02-11:
Thank you for your use case description. Currently, it's not possible to specify more than one mpn or more than one spn for any component. To enable something like that, it must be defined a clear and unambiguous way to display a component with many such values in a diagram, and in the BOM. Please suggest.

Your suggested input syntax is not valig YAML, as all keys in a mapping must be unique. However, maybe a variation like this, might be possible:
```yaml
connectors:
  X1:
    manufacturer:
      Lolex:
        mpn: 12345
        spn:
          MousiKey: MK-12345
          ArroArk: AA-12345
      ET Connection:
        mpn: 54321
        spn:
            MousiKey: MK-54321
            ArroArk: AA-54321
```

kvid, 2023-09-02:
I've not yet seen any suggestion describing how such multiple manufacturers and suppliers could be listed in the BOM. How is this done in other systems or documents?

nyq, 2023-09-04:
We are currently using a semi-automated BOM based on protected Word document with VBA code to handle addition/removal of the parts as well as their distribution to procedure documents. Given such limited non-database platform we decided that implementing arbitrarily long list of suppliers for a given part is not realistic and opted for two fixed fields, Vendor1 and Vendor2, which gives us the ability to specify at least one alternative supplier. Not the prettiest solution, but it is better than having no alternative suppliers at all. As far as alternative manufacturers are concerned, we have not implemented that yet as there was no pressing need. But in principle it could be done in form of fixed fields too, basically Factory1, Factory1.Vendor1, Factory1.Vendor2, Factory2, Factory2.Vendor1, Factory2.Vendor2. Not pretty, but better than nothing.

A true arbitrary length list of vendors would require a database (of any form, including YAML) and rendering in tabular form will never be great, as these multiple values will need to be crammed into the same column, but such is life. I think the rendering sample that you have provided above is the best I have seen so far.