wireviz/WireViz #306: Tolerance in cable lengths

ct2irw, 2023-02-03:
Hi there,

My name is Diogo. So often manufacturers ask to include tolerances in the cable lengths. I think it would make sense to have a key/value for that. The way I've implemented that was by mostly cloning the length key... The only change I did in that was to inherit the units from the length if none is provided instead of assuming 'm' like length does. Ideally, it would process different tolerances like -10mm and +15mm, but I did not have time to mess with that.

Let me know if you want any help implementing that.

Best regards,
DIogo

formatc1702, 2023-05-15:
It would be nice if this didn't require an additional key/value in the YAML, but if it could be parsed within the `length` attribute itself. Internally, it probably makes sense to have it as a separate variable for sure.

Some suggestions for YAML input:
```yaml
length: 100  # assume default unit
length: 200 mm  # explicit unit
length: 300 ±5 mm   # place unit before or after tolerance? allow both?
length: 400 +-5 mm  # allow +- instead of ± for easier input
length: 500 +5 -0  # allow asymmetric tolerance
length: 600 ±10%  # allow percentages?
                  # show as percentage in output or do the math?
                  # maybe as a second step later on
length: 700~750  # allow length range?
```

kvid, 2023-08-27:
How should similar entries with such lengths be summed up in the BOM?
- Sum nominal lengths only?
- Sum nominal lengths, positive tolerances, negative tolerances (all in the same length unit) and present
  as `{sum_nominals} +{sum_positives} -{sum_negatives} {unit}`
- Sum minimum and maximum lengths of each length range and present
  as `{sum_minimums}~{sum_maximums} {unit}`

If specifying length range is allowed as input - hence no nominal specified, then only the latter alternative above might be possible.

If different lengths in #268 will be implemented, then care must be taken how to present that combined with different length tolerances for each length.

formatc1702, 2023-09-16:
BOM should probably show a sum of nominal lengths.

The last possibility (specifying a length range without a singular nominal length, e.g. `length: 700~750`) should be discarded so that there always is a nominal length available.

Tolerances (sum of positives, sum of negatives) could become their own column in the BOM, but it might be overkill to actually display them?

kvid, 2023-09-26:
> BOM should probably show a sum of nominal lengths.

Yes, I guess that's what most users would expect.

> 
> The last possibility (specifying a length range without a singular nominal length, e.g. `length: 700~750`) should be discarded so that there always is a nominal length available.

It's also possible to define a length range input to be regarded as equal to a nominal length = range middle +- tolerance = half range length.

> 
> Tolerances (sum of positives, sum of negatives) could become their own column in the BOM, but it might be overkill to actually display them?

kvid, 2024-09-10:
Maybe `ufloat` from the [uncertainties](https://pypi.org/project/uncertainties/) library might be used to store and accumulate such cable lengths? In that case, it might be wise to consider the input syntax of [ufloat_fromstr()](https://uncertainties.readthedocs.io/en/latest/tech_guide.html#uncertainties.ufloat_fromstr). However asymmetric values might not be supported: https://github.com/lmfit/uncertainties/issues/25

formatc1702, 2024-10-05:
I'm torn on adding a library for this.. sounds a bit overcomplicated? or at least should be split as a separate task, after implementing "simple" tolerances, that show up in each cable's node, but have no effect on BOM (which only uses nominal lengths), and getting feedback on that.

kvid, 2024-10-05:
@formatc1702 - You are probably right it might be overkill using such a library for a simple sum of tolerances, but maybe it can serve as an inspiration, by picking a few elements we also need...

I do like the idea to have a class representing a float with it's tolerances, but how to fit this with other requirements depend on several questions:
- Do we require tolerance values to be in the same unit as the nominal value (or percentage)? Probably yes.
- If different lengths in #268 also will be implemented, do we require all length elements to use the same unit? However, such length elements probably might need different tolerances.