wireviz/WireViz #307: [feature] Use the yaml new document --- to create multiple outputs with a single input.

EvilMustacheTwirl, 2023-02-07:
It would be great to be able to create multiple outputs (especially for larger/messy setups) that can be easily self referenced. 

Using yaml's `---` to tell wireviz that this is a new output but still connected and maybe adding a full document Title, Note, Legend, Filename, etc. tag similar to the metadata options would make it easier to produce cleaner looking outputs. If no filename tag is added wireviz could add an increment-er to the inputs file name.

kvid, 2023-03-30:
Thank you for suggesting this new feature. I've not worked much with the parsing part of WireViz, myself, but the pyyaml library should be able to parse `---`. However, I don't know how much coding is needed to obtain what you ask for. Are you motivated to give it a try creating some code for such a feature yourself? See [Submitting a new Pull Request](https://github.com/formatc1702/WireViz/blob/master/docs/CONTRIBUTING.md#submitting-a-new-pull-request). I assume there will be some questions about how such a feature should handle certain edge cases, and this issue is the proper place to ask questions like that...

martinrieder, 2024-05-30:
@kvid wrote:
> An alternative use case for such a syntax (leading or trailing arrow) could be to display an arrow indicating the wire continues somewhere else (in a large diagram or in another diagram). It would probably make sense to support a text label like this:
> ```yaml
> # single wire that continues somewhere else
> -
>  - <-- Common signal ground
>  - WIRE
>  - --> Diagram B
> ```

I wrote:
> Great idea! We should implement this in addition to the above. If a text is supplied, then label it, otherwise take it from the connected wire. 
_in https://github.com/wireviz/WireViz/issues/270#issuecomment-2123548853_